rockowitz/ddcutil #2: Cannot set vcp value

serkazak, 2016-06-11:
i get the following error whenever i try to set a vcp value :

Setting value failed. rc=-3015: DDCRC_RETRIES(-3015): maximum retries exceeded

for instance i send command such as this one
ddctool --adl=0.0 setvcp 10 20

i tried this command maybe a hundred times and it did not work once so i assume maximum retries is not the problem here. 
i use debian stable and use fglrx driver. i am using philips 234E5 display. i got no issues building from source and functions such as detect, capabilities, getvcp work fine. am i missing something here? does the return code 3015 refer to anything? here is the commands and features of this display ;
[ddctool.txt](https://github.com/rockowitz/ddctool/files/310440/ddctool.txt)

i am quite interested in this tool because i change brightness during day and night manually every day. it would be great if i could get this work. also i think tools such as redshift, flux etc could potentially use this tool.

rockowitz, 2016-06-12:
Thank you for the error report!

First off, DDCRC_RETRIES is indeed a valid status code.  3015 is its
numeric value.   The I2C bus is inherently unreliable.  It is the
application's responsibility to perform retry if an error is detected.  
In my experience, most monitors are pretty clean.   Some, such as Dell
P2014H's (I've tested several) have very high error rates.  
DDCRC_RETRIES indicates that the maximum number of retries has been
exceeded. 

That said, there's something peculiar going on here.   "ddctool
capabilities" succeeds.  It is what I call a "multi-exchange" command,
requiring multiple I2C writes and reads, each of which must succeed.
This, more than anywhere, is where I would expect to see retries. So my
best guess is that I'm treating some fatal error as retryable. 

I'm going to suggest several things:

1) Add the  "--stats" option on all command invocations.

2) Add  "--trace  adl --trace ddc" to the setvcp command and send me the
output. 

3) Execute "ddctool environment".   The output may give you a clue if
something needs to be changed in your environment.
You can also use the "--verbose" option on this command to dump more
information.

4) Does the problem persist if you run ddctool as root?

5) Execute the command "ddctool interrogate" and send me the output. 

Regards,
Sanford

serkazak, 2016-06-12:
i think the problem is with write only mode
here is the setvcp command with trace and stats parameters
[setvcp.txt](https://github.com/rockowitz/ddctool/files/310715/setvcp.txt)

and here is the interrogate
[interrogate.txt](https://github.com/rockowitz/ddctool/files/310716/interrogate.txt)

by the way i get "(get_hiddev_device_names) scandir() error: No such file or directory" when i issue a command without adl parameter but the rest of the output is the same with and without adl. so i assume it is irrelevant.

i cannot run it as root because it cannot find a display :
"No X display found by ADL. Apparently running in console environment. (ADL_ERR_NO_XDISPLAY)
(get_hiddev_device_names) scandir() error: No such file or directory
No active displays found"

rockowitz, 2016-06-12:
>
> i think the problem is with write only mode
> here is the setvcp command with trace and stats parameters
> setvcp.txt <https://github.com/rockowitz/ddctool/files/310715/setvcp.txt>

Yep.  Error -2001 is ADL error -1 (ADL_ERR). 
The explanation is "Generic Error.  Most likely one or more of the
Escape calls to the driver failed!"

This is in the "something weird is going on" category, since the status
code appears to be coming from the core function
(call_ADL_Display_DDCBlockAccess_Get) that's used for all writes and
reads to the monitor, and every read (e.g. on getvcp) requires a prior
write.)
>
> and here is the interrogate
> interrogate.txt

Looks like you're using the integrated Radeon graphics on an ASRock
motherboard.   The monitor is connected using a VGA cable.   (xrandr
reports device CRT1 connected, devices DFP1 and DFP2 disconnected.)

> by the way i get "(get_hiddev_device_names) scandir() error: No such
> file or directory" when i issue a command without adl parameter but
> the rest of the output is the same with and without adl. so i assume
> it is irrelevant.

It's a benign error message that needs to be suppressed.  When you don't
specify an ADL device, ddctool looks for all monitors.  In particular,
some monitors, such as Apple Cinema displays and (more interestingly)
professional level Eizo and NEC monitors communicate settings over the
the USB bus.  The error message is just saying that there are no USB
devices of interest.
>
> i cannot run it as root because it cannot find a display :
> "No X display found by ADL. Apparently running in console environment.
> (ADL_ERR_NO_XDISPLAY)
> (get_hiddev_device_names) scandir() error: No such file or directory
> No active displays found"

Were you running in a windowed terminal on the GUI screen or had you
switched to a virtual terminal (e.g. ALT-CTL-F1)?  I would expect the
former to work, the latter to fail with ADL_ERR_NO_XDISPLAY).   If the
former, what terminal program are you using (e.g. gnome-terminal,
mate-terminal, kconsole)?  Did you use "sudo ddctool" to run the program
or did you run as root some other way?

My best hunch is that there's something about your monitor that ADL
doesn't like.  Can you try connecting it using an HDMI cable? 
(According to the specs I found, the monitor has only VGA and HDMI
connections.  I can't tell what connections the motherboard has.)   Can
you try connecting a different monitor?

I'll study things further on my ADL test system and probably push out
some code with increased diagnostics.

Sanford

rockowitz, 2016-06-12:
I'm able to reproduce the problem on my fglrx test system.    Will keep
you posted.

Sanford

serkazak, 2016-06-12:
i switched to radeon driver and now i am able to send vcp commands to the same monitor so it seems to be related with the fglrx driver

rockowitz, 2016-06-12:
I've applied a fix (file ddc_packet_io.c) that works here and pushed it
out to github.    Let me know.  You'll probably also see some diagnostic
output that needs to be cleaned up.

Sanford

rockowitz, 2016-06-13:
Short answer: as expected. 

Long answer:  There are 3 paths for communicating Monitor Control
Command Set (MCCS) instructions with the monitor.  

1) In the normal case, the video driver surfaces raw I2C communication
as devices named /dev/i2c-*.  ddctool is responsible for all the gory
details of managing the I2C protocol using writes and reads to the
/dev/i2c-* devices.   All the open source drivers (nouveau, radeon,
intel) do this, as does Nvidia's proprietary driver.

2) The AMD fglrx driver wraps I2C communication in an API.  So at a
certain level, ddctool calls fglrx functions instead of doing the low
level I2C I/O itself.

3) Some monitors use USB to communicate the Monitor Control Command Set
(MCCS).

The error, corrected in the patch I uploaded  few hours ago, was in
packet adjustment for case (2). 

Sanford

serkazak, 2016-06-13:
reinstalled fglrx driver but when i was at make stage i got this;
"util/hiddev_util.c:31:21: fatal error: libudev.h: No such file or directory
 #include <libudev.h>
                     ^
compilation terminated."
so i installed libudev-dev but now i get the following 

util/ddctool-udev_util.o: In function `report_udev_device':
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:45: undefined reference to `udev_device_get_devpath'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:46: undefined reference to `udev_device_get_subsystem'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:47: undefined reference to `udev_device_get_devtype'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:48: undefined reference to `udev_device_get_syspath'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:49: undefined reference to `udev_device_get_sysname'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:50: undefined reference to `udev_device_get_sysnum'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:51: undefined reference to `udev_device_get_devnode'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:57: undefined reference to `udev_device_get_properties_list_entry'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:58: undefined reference to `udev_device_get_sysattr_list_entry'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:62: undefined reference to `udev_list_entry_get_name'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:63: undefined reference to `udev_list_entry_get_value'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:64: undefined reference to `udev_device_get_property_value'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:61: undefined reference to `udev_list_entry_get_next'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:71: undefined reference to `udev_list_entry_get_name'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:72: undefined reference to `udev_list_entry_get_value'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:73: undefined reference to `udev_device_get_sysattr_value'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:70: undefined reference to `udev_list_entry_get_next'
util/ddctool-udev_util.o: In function `query_udev_subsystem':
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:89: undefined reference to `udev_new'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:96: undefined reference to `udev_enumerate_new'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:97: undefined reference to `udev_enumerate_add_match_subsystem'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:98: undefined reference to `udev_enumerate_scan_devices'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:99: undefined reference to `udev_enumerate_get_list_entry'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:145: undefined reference to `udev_device_get_sysattr_value'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:145: undefined reference to `udev_device_get_sysattr_value'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:148: undefined reference to `udev_device_get_sysattr_value'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:148: undefined reference to `udev_device_get_sysattr_value'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:151: undefined reference to `udev_device_get_sysattr_value'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:157: undefined reference to `udev_device_unref'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:105: undefined reference to `udev_list_entry_get_next'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:111: undefined reference to `udev_list_entry_get_name'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:113: undefined reference to `udev_device_new_from_syspath'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:117: undefined reference to `udev_device_get_devnode'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:127: undefined reference to `udev_device_get_parent_with_subsystem_devtype'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:160: undefined reference to `udev_enumerate_unref'
/home/serkan/Downloads/ddctool-master/src/util/udev_util.c:162: undefined reference to `udev_unref'
./.libs/libcommon.a(libcommon_la-hiddev_util.o): In function `get_hiddev_device_names_using_udev':
/home/serkan/Downloads/ddctool-master/src/util/hiddev_util.c:139: undefined reference to `udev_new'
/home/serkan/Downloads/ddctool-master/src/util/hiddev_util.c:147: undefined reference to `udev_enumerate_new'
/home/serkan/Downloads/ddctool-master/src/util/hiddev_util.c:148: undefined reference to `udev_enumerate_add_match_subsystem'
/home/serkan/Downloads/ddctool-master/src/util/hiddev_util.c:149: undefined reference to `udev_enumerate_scan_devices'
/home/serkan/Downloads/ddctool-master/src/util/hiddev_util.c:150: undefined reference to `udev_enumerate_get_list_entry'
/home/serkan/Downloads/ddctool-master/src/util/hiddev_util.c:167: undefined reference to `udev_device_unref'
/home/serkan/Downloads/ddctool-master/src/util/hiddev_util.c:155: undefined reference to `udev_list_entry_get_next'
/home/serkan/Downloads/ddctool-master/src/util/hiddev_util.c:160: undefined reference to `udev_list_entry_get_name'
[... cut ...]
/home/serkan/Downloads/ddctool-master/src/util/hiddev_util.c:189: undefined reference to `udev_unref'
collect2: error: ld returned 1 exit status
Makefile:1086: recipe for target 'ddctool' failed
make[2]: *** [ddctool] Error 1
make[2]: Leaving directory '/home/serkan/Downloads/ddctool-master/src'
Makefile:429: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/serkan/Downloads/ddctool-master'
Makefile:358: recipe for target 'all' failed
make: *** [all] Error 2

am i missing something here?

serkazak, 2016-06-13:
now i got it working with the updated version with fglrx. stats show no retries and it works as expected thank you.

closed 2016-06-13 by serkazak

rockowitz, 2016-06-13:
I take it I can disregard your post re libudev.h issues.

rockowitz, 2016-06-13:
Are you saying that updating fglrx resolved the retries issue?   If so,
what versions of fglrx did you upgrade from/to?

rockowitz, 2016-06-13:
I've added some diagnostic code to try to understand why ddctool does
not detect the username on your system. 

When you have chance, please rebuild ddctool with the current github
contents, issue the command "ddctool environment" and send me the output.  

Thanks,
Sanford