rockowitz/ddcutil #3: libxrandr-dev on Ubuntu 16.04

diegocerdan, 2016-07-13:
I also needed to install libxrandr-dev on Ubuntu 16.04 to be able to compile it.

I was getting the following error when running _./configure_

> checking for XRANDR... no
> cconfigure: error: Package requirements (xrandr) were not met:

> No package 'xrandr' found

> Consider adjusting the PKG_CONFIG_PATH environment variable if you installed software in a non-standard prefix.

> Alternatively, you may set the environment variables XRANDR_CFLAGS and XRANDR_LIBS to avoid the need to call pkg-config.
> See the pkg-config man page for more details.

The issue is fixed after installing libxrandr-dev.

rockowitz, 2016-07-14:
Thank you for the information.  I'll take a look to see if I can make
the package requirements clearer. 

Sanford

rockowitz, 2016-09-09:
I've added an entry to the FAQ (www.ddctool.com/faq) pointing out that error messages from configure regarding missing packages typically mean that the associated "-dev" package is missing, and added similar information to the page on building ddctool (www.ddctool.com/building).

closed 2016-10-08 by diegocerdan