rockowitz/ddcutil #4: Can you put releases in the releases tab?

dracwyrm, 2016-09-02:
Hi,

Currently there are no releases in the releases tab, but there are tarballs in the main repo. It would be better if the downloads were in the releases tab, which I think you can tell github to do automatically(?) and remove the tarballs from the main repository.

Thanks.

rockowitz, 2016-09-02:
Jonathan,

Thanks for pointing this out.   Having tarballs in the main repo was a
temporary expedient which is due for replacement.   I'll look at putting
them in the releases tab. 

Sanford

dracwyrm, 2016-09-06:
Thanks for responding. This is the github guide:
https://help.github.com/articles/creating-releases/

The part about including binaries is for anything that is not in the repo. It should create the zips and tarballs for you.

rockowitz, 2016-09-09:
Jonathan,

Thanks for directing me to the documentation.

ddctool has just had its first formal release, 0.5.0.  github now
creates the tarball as part of the release, and it's no longer in the
mail repo.   The packaging is still a bit ragged, but progress.

Regards,
Sanford

dracwyrm, 2016-09-09:
Sanford,

No problem. :) Keep up the great work on this tool!

Jon

closed 2016-10-01 by dracwyrm