rockowitz/ddcutil #6: ddctool v0.5.2: No command specified error

dracwyrm, 2016-09-24:
I went from v0.5.0 to v0.5.2 and when I run a command, I get this:
```
# ddctool detect
No command specified
```

When before on v0.5.2, I would get this:
```
# ddctool detect
Display 1
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           ACI
      Model:            ASUS VN247
      Serial number:    {snip}
      Manufacture year: 2014
      EDID version:     1.3
   VCP version:         2.1
```

This happens with both user and root. v0.5.1 failed to compile with this error:
```main.c:(.text.startup+0xc0): undefined reference to `is_module_loaded_using_sysfs'```

Cheers.

rockowitz, 2016-09-24:
Jonathan,

I just released v0.5.3 with a fix for No Command Specified.

The compilation failure was the bug fixed earlier today by the ill-fated
v0.5.2.

Thanks for the report. 

Sanford

dracwyrm, 2016-09-24:
It's working now. Thanks for replying so quickly and providing the fix as well.

closed 2016-10-01 by rockowitz