rockowitz/ddcutil #7: i2c device or resource busy

diggit, 2016-11-12:
I tried to detect ddc devices and it ended with error...
```
sudo ddcutil interrogate                                                                                                                                                                                                                               Sat 12 Nov 2016 02:24:36 PM CET
Setting output level verbose...
Setting maximum retries...
Forcing --stats...
This command will take a while to run...

System information (uname):
   Linux version 4.8.6-1-ARCH (builduser@tobias) (gcc version 6.2.1 20160830 (GCC) ) #1 SMP PREEMPT Mon Oct 31 18:51:30 CET 2016

*** Primary Check 1: Identify video card and driver ***
Obtaining card and driver information from /sys...

Determining driver name and possibly version...
   Driver name:    i915
   Unable to determine driver version

Video card identification:
   Vendor:              8086       Intel Corporation
   Device:              0166       3rd Gen Core processor Graphics Controller
   Subvendor/Subdevice: 17aa/21f3  Lenovo

*** Primary Check 2: Check that /dev/i2c-* exist and writable ***

Checking /dev/i2c-* devices...

Unless the system is using the AMD proprietary driver fglrx, devices /dev/i2c-*
must exist and the logged on user must have read/write permission for those
devices (or at least those devices associated with monitors).
Typically, this access is enabled by:
  - setting the group for /dev/i2c-* to i2c
  - setting group RW permissions for /dev/i2c-*
  - making the current user a member of group i2c
Alternatively, this could be enabled by just giving everyone RW permission
The following tests probe for these conditions.

Checking for /dev/i2c-* devices...
   crw------- 1 root root 89, 0 Nov  5 20:59 /dev/i2c-0
   crw------- 1 root root 89, 1 Nov  5 20:59 /dev/i2c-1
   crw------- 1 root root 89, 2 Nov  5 20:59 /dev/i2c-2
   crw------- 1 root root 89, 3 Nov  5 20:59 /dev/i2c-3
   crw------- 1 root root 89, 4 Nov  5 20:59 /dev/i2c-4
   crw------- 1 root root 89, 5 Nov  5 20:59 /dev/i2c-5
   crw------- 1 root root 89, 6 Nov  5 20:59 /dev/i2c-6
   crw------- 1 root root 89, 7 Nov  5 20:59 /dev/i2c-7
   crw------- 1 root root 89, 8 Nov  5 20:59 /dev/i2c-8
   crw------- 1 root root 89, 9 Nov  5 20:59 /dev/i2c-9

Current user: root (0)

ioctl error in function i2c_set_addr at line 159 in file i2c/i2c_bus_core.c: errno=EBUSY(16): Device or resource busy
Aborting. Internal status code = -3026
```

My version is:
```
ddcutil -V                                                                                                                                                                                                                                        Sat 12 Nov 2016 02:27:20 PM CET
ddcutil 0.6.0
Built Nov  5 2016 at 13:26:09
Built without support for AMD Display Library (AMD proprietary driver).
Built with support for USB connected displays.

Copyright (C) 2015-2016 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

ddccontrol is capable to communicate with my monitor, so device is accessible.

What can I do to help you debug this?

rockowitz, 2016-11-13:
> I tried to detect ddc devices and it ended with error...I

Patrik,

Thank you for the report.  Looks like you're running Arch Linux on a
Lenovo laptop (Thinkpad?).   External monitor?  The error is happening
on an ioctl call never seen to fail.    I'll need to add recovery code
to figure out what's going on.  Are you comfortable building ddcutil
from a git branch?

Sanford

rockowitz, 2016-11-14:
Patrik,

I have pushed a new branch, named "set_slave_error" to github.   It
contains extensive changes to percolate the return code from the failed
ioctl call up the call stack.   This is not a general release, as there
are a lot of debugging trace messages turned on, but it should help to
(a) identify the system conflict causing the failure and (b) test the
recovery.     Command "ddcutil interrogate" should now run to completion.

As pure speculation, any chance you were running ddccontrol/gddccontrol
at the same time you encountered the EBUSY error?

Let me know how it goes. 

Sanford

diggit, 2016-11-19:
Sorry for such late reply. Busy weeks for me.
I tried ddccontrol before ddcutil, but definitely not at the same time.
Now I tried `ddcutil interrogate` again and everything works. IDK what caused that problem. I might be some other package bug which got fixed. Arch rolls out updates quite often.
Hard to guess what caused it, my uptimes are commonly over 10 days. Not sure if something bad happened during this time affecting i2c bus system.

Writing over ddc works well.

Hope you didn't spend much time on debug coding.

Patrik

rockowitz, 2016-11-19:
Hi Patrik,

Thanks for the update.  I've incorporated the improved error handling
code into the next point release that will roll out in a few days.  
Please ignore the git set_slave_error branch, which is now dated.
Hopefully, if you encounter the problem again, the upcoming release will
provide more informative diagnostics.  

I'd be very appreciative if you could send me the output of "ddcutil
interrogate".   Monitors differ, often significantly, in their
interpretation and implementation of the DDC related standards.   The
more monitors I see the more robust I can make ddcutil.  Thanks. 

Sanford

diggit, 2016-11-19:
Sure, here it is: https://gist.github.com/diggit/c12bb97ad9ab166c0d3e61ce230e7348
I have connected internal LVDS display and external [LCD display](http://www.benq.com/product/monitor/bl2420pt/) through DisplayPort.

I have decoded functions and ranges of 99% Manufacturer specific commands. Unfortunately dcccontrol looks dead and ddcutil does not use external display scpecifications(afaik).

As you wrote earlier, I use Thinkpad T430 with integrated Intel graphics.

rockowitz, 2016-11-20:
Patrik,

Thank you for the execution log.   There's always some new quirks.

You're correct that, unlike ddccontrol, ddcutil does not use external
display descriptors, but instead relies on the DDC specs for
interpreting monitor communication.  This will, I believe, make ddcutil
more robust in the long run.  Coming up with a definition for each
monitor is a herculean task - consider how out of date ddccontrol's
monitor database is.  ddcutil instead puts its effort into understanding
the DDC/MCCS specs and the variation among versions.  

That said, I have considered allowing the use of an external control
file to describe the manufacturer specific commands.  I would not,
however, want to get into the business of maintaining a monitor spec
database. 

Sanford

rockowitz, 2016-11-21:
Patrik,

Release 0.6.1 of ddcutil, which was uploaded earlier today, contains
code to recover from and diagnose the transient EBUSY error you
encountered.  In particular, it should indicate whether the I2C device
exhibiting the error is actually a monitor.

Sanford

closed 2017-01-03 by rockowitz