rockowitz/ddcutil #8: ThinkPad dock not working

daradib, 2016-11-15:
I have a Lenovo ThinkPad T440s laptop, ThinkPad Ultra dock, and Dell U2415 monitor. I'm using Intel i915 graphics (haswell).

When I connect the monitor directly to the laptop, I can set the brightness with `ddcutil setvcp 10`. However, when the monitor is connected through the dock, `ddcutil detect` reports "Invalid display" for the monitor and a broken display 1 (see below) and `ddcutil setvcp` reports "Cannot communicate DDC".

The dock seems to expose monitors as multi-stream DisplayPort (DP2-1, DP2-2, DP2-3, though I'm only using the first). So I'm curious if DDC works over a dual-link DisplayPort connection. Otherwise this could also be an issue with I2C or the dock not passing through DDC.

     $ ./src/ddcutil detect
      [...]
     Invalid display
        I2C bus:             /dev/i2c-9
        EDID synopsis:
           Mfg id:           DEL
           Model:            DELL U2415
           Serial number:    [redacted]
           Manufacture year: 2016
           EDID version:     1.4

     Display 1
        I2C bus:             /dev/i2c-10
           No EDID
     ddcutil: ddc/ddc_packet_io.c:166: ddc_open_display: Assertion `pDispHandle->pedid' failed.
     Aborted

    $ cat /sys/bus/i2c/devices/i2c-9/name
    DPDDC-C

    $ cat /sys/bus/i2c/devices/i2c-10/name 
    DPMST

rockowitz, 2016-11-16:
Dara,

Thanks for the report.  I'm afraid I don't have an immediate
solution.    DisplayPort implements ("emulates" may be a better word)
I2C by carrying it over the DisplayPort AUX channel.  (You can search
for "I2C-over-AUX".)  This is entirely the responsibility of the video
driver, which ddcutil relies on, and is hidden from userland.  So
there's not much that ddcutil can do. 

That said, some further comments. 

My laptop is a T410.  Normally it runs Windows, but I can boot it into
Linux.   It has a docking station, and when it's on my desk it's
connected to an HP ZR2740w (2560x1440) using a DisplayPort cable.  Even
with Windows, the monitor does not work properly when connected to the
dock.  The monitor has to be plugged into the laptop itself.  So I
suspect the behavior you're seeing reflects a limitation of the dock. 

Dual-link is a DVI concept.  More recent DisplayPort revisions have
increasing bandwidth, but even DisplayPort 1.0 (and single link DVI)
have sufficient bandwidth for a U2415 (1920x1600 @60hz).  If you want to
convert a DisplayPort signal to dual-link DVI, a  special adapter is
required, and its conceivable that such an adapter might not convert the
I2C signal correctly.  I do have single link DVI to DisplayPort
adapters, and I have not seen a problem with the I2C signal.

My desktop has multiple monitors.  The largest is a Dell U3011
(2560x1600), which at various times has been connected using either
dual-link DVI or DisplayPort.  I2C communication has worked in either
case.  So again, I would say that the problem is in the dock.

What I can do is improve ddcutil's error handling, to respond more
gracefully and hopefully provide more useful diagnostics. Assertions are
for situations that should never arise.

Finally, if you can run "ddcutil interrogate" when the monitor is
connected directly to the laptop I'd be very appreciative.    Recent
Ultrasharps have some interesting capabilities, and I'm curious to see
what the interrogate command shows.

Sanford

closed 2017-01-03 by rockowitz

diggit, 2017-01-03:
Didn't notice this issue earlier. I have T430 with Intel graphics docked and my DP monitor Benq BL2420PT connected to this dock (dock version with 2DPs and 2DVIs, but older version meant for T420 with eSATA instead of USB3.0). ddcutil works well. DP1 is DP on T430, DP2 and DP3 are located on dock.

rockowitz, 2017-01-04:
Useful to know.  More data points are helpful.  Thanks.

Sanford