rockowitz/ddcutil #9: Missing test for libudev in configure

moschlar, 2016-12-14:
On Debian Jessie, I had to additionally install `libudev-dev`, but that only occured when running make.

rockowitz, 2016-12-14:
The test for libudev-dev in configure is a bit ugly, since (a) udev
support may or may not be enabled (options --enable-usb/--disable-usb),
and depending on platform and release, udev support may or may not be
part of systemd. 

Did configure complain that libudev-dev was missing, or did you have to
figure this out on your own?  Note that the output from pkg-config can
be a bit misleading.   A complaint, for example, that libudev is missing
typically means that the corresponding -dev package is missing.

moschlar, 2016-12-16:
No, configure didn't complain at all.
Make errored because there was no udev.h

rockowitz, 2016-12-20:
configure has been revised to complain if libudev.h is not found and
explain that the necessary package can vary by Linux distribution and
release.  This change is part of a general rework of configure for the
next ddcutil release.  Thanks again for the report.

Sanford

closed 2017-01-03 by rockowitz