rockowitz/ddcutil #11: ddcutil not working through lenovo dock

thmcmahon, 2017-01-12:
Hi

I built ddcutil 0.7.0 on Ubuntu 16.10. I have two external displays connected to my Lenovo x260 via a Lenovo Ultra Dock. The displays are Dell U2412Ms.

Here is the output of `ddcutil detect`:

```
sudo ddcutil detect
Display 1
   I2C bus:             /dev/i2c-3
   Supports DDC:        true
   EDID synopsis:
      Mfg id:           AUO
      Model:            Unspecified
      Serial number:    Unspecified
      Manufacture year: 2015
      EDID version:     1.4
   VCP version:         Detection failed

Invalid display
   I2C bus:             /dev/i2c-5
   Supports DDC:        false
   EDID synopsis:
      Mfg id:           DEL
      Model:            DELL U2412M
      Serial number:    Unspecified
      Manufacture year: 2015
      EDID version:     1.4

Display 2
   I2C bus:             /dev/i2c-6
   Supports DDC:        true
(ddc_open_display) No EDID for device on bus /dev/i2c-6
   Error opening display bus /dev/i2c-6, error = -3016 (DDCRC_EDID)
(ddc_close_display) close_i2c_bus returned 9

Display 3
   I2C bus:             /dev/i2c-7
   Supports DDC:        true
(ddc_open_display) No EDID for device on bus /dev/i2c-7
   Error opening display bus /dev/i2c-7, error = -3016 (DDCRC_EDID)
(ddc_close_display) close_i2c_bus returned 9

```

However when I try and detect monitor capabilities it doesn't work:
```
tom@x260:~$ sudo ddcutil capabilities --display 1
Unable to get capabilities for monitor on Display_Handle[i2c: fh=3, busno=3].  Maximum DDC retries exceeded.
```

This seems like it may be related to [issue number 8](https://github.com/rockowitz/ddcutil/issues/8) - I'm not really all that stressed about not getting this to work, but thought this might be helpful.

rockowitz, 2017-01-12:
Thomas,

Your report is very useful.   I've been working on trying to better
handle the corner case where the pseudo-I2C bus for the laptop's display
claims to support DDC when it fact it doesn't. 

Please run the following command and send me the output.
  ddcutil environment --verbose

Are you installing ddcutil from git, the tarball, or one of the rpms or
dpkgs?  I ask because it might be helpful if you could install a
development version of ddcutil.  

Thanks,
Sanford

thmcmahon, 2017-01-12:
Sanford

I installed it from the tarball on the website. Happy to try a dev version if that would help.

Here's the output of `sudo ddcutil environment --verbose`

```
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
The following tests probe the runtime environment using multiple overlapping methods.

System information (uname):
   Linux version 4.8.0-32-generic (buildd@lcy01-34) (gcc version 6.2.0 20161005 (Ubuntu 6.2.0-5ubuntu12) ) #34-Ubuntu SMP Tue Dec 13 14:30:43 UTC 2016

*** Primary Check 1: Identify video card and driver ***
Obtaining card and driver information from /sys...

Determining driver name and possibly version...
   Driver name:    i915
   Unable to determine driver version

Video card identification:
   Vendor:              8086       Intel Corporation
   Device:              1916       HD Graphics 520
   Subvendor/Subdevice: 17aa/504a  Lenovo

*** Primary Check 2: Check that /dev/i2c-* exist and writable ***

Checking /dev/i2c-* devices...

Unless the system is using the AMD proprietary driver fglrx, devices /dev/i2c-*
must exist and the logged on user must have read/write permission for those
devices (or at least those devices associated with monitors).
Typically, this access is enabled by:
  - setting the group for /dev/i2c-* to i2c
  - setting group RW permissions for /dev/i2c-*
  - making the current user a member of group i2c
Alternatively, this could be enabled by just giving everyone RW permission
The following tests probe for these conditions.

Checking for /dev/i2c-* devices...
   crw-rw---- 1 root i2c 89, 0 Jan 12 15:44 /dev/i2c-0
   crw-rw---- 1 root i2c 89, 1 Jan 12 15:44 /dev/i2c-1
   crw-rw---- 1 root i2c 89, 2 Jan 12 15:44 /dev/i2c-2
   crw-rw---- 1 root i2c 89, 3 Jan 12 15:44 /dev/i2c-3
   crw-rw---- 1 root i2c 89, 4 Jan 12 15:44 /dev/i2c-4
   crw-rw---- 1 root i2c 89, 5 Jan 12 15:44 /dev/i2c-5
   crw-rw---- 1 root i2c 89, 6 Jan 12 15:44 /dev/i2c-6
   crw-rw---- 1 root i2c 89, 7 Jan 12 15:44 /dev/i2c-7

Current user: root (0)

Current user (root) has RW access to all /dev/i2c-* devices.

Checking for group i2c...
   Group i2c exists
   WARNING: Current user root is NOT a member of group i2c

Looking for udev nodes files that reference i2c:
   grep: /etc/udev/makedev.d/*: No such file or directory

Looking for udev rules files that reference i2c:
   /lib/udev/rules.d/40-i2c-tools.rules:KERNEL=="i2c-[0-9]*", GROUP="i2c", MODE="0660"
   grep: /run/udev/rules.d/*rules: No such file or directory
   grep: /etc/udev/rules.d/*rules: No such file or directory

*** Primary Check 3: Check that module i2c_dev is loaded ***

Checking for module i2c_dev...
[... 531 lines cut ...]
   path: /sys/devices/pci0000:00/0000:00:02.0/i2c-7/i2c-dev/i2c-7
   Device Node Path: /dev/i2c-7
      struct udev_device at: 0x55b07261fc70
         devpath:   /devices/pci0000:00/0000:00:02.0/i2c-7/i2c-dev/i2c-7
         subsystem: i2c-dev
         devtype:   (null)
         syspath:   /sys/devices/pci0000:00/0000:00:02.0/i2c-7/i2c-dev/i2c-7
         sysname:   i2c-7
         sysnum:    7
         devnode:   /dev/i2c-7
         Properties:
            DEVNAME -> /dev/i2c-7
            DEVPATH -> /devices/pci0000:00/0000:00:02.0/i2c-7/i2c-dev/i2c-7
            MAJOR -> 89
            MINOR -> 7
            SUBSYSTEM -> i2c-dev
         Sysattrs:
            dev -> 89:7
            device -> (null)
            name -> DPMST
            subsystem -> i2c-dev
            uevent -> MAJOR=89
                      MINOR=7
                      DEVNAME=i2c-7
```

rockowitz, 2017-01-13:
Thomas,

Well, this is seriously weird.  

First off, I'm going to ask you to send me the full, gory, diagnostics,
using command "sudo ddcutil interrogate".   This will take a while to run. 

Before I get into the weeds, a bit of background on how ddcutil
(currently) detects monitors on the I2C bus.   It scans each of the
/dev/i2c-n devices, checking if slave addresses x50 (EDID) and x37 (DDC
communication) are active.  If both are, it concludes it's found a monitor.

There are 2 things happening on your system.  The first is indeed
related to issue #8 <https://github.com/rockowitz/ddcutil/issues/8>.  
The driver is reports that x37 is active on bus /dev/i2c-3, which is the
laptop display.   But in fact DDC communication fails (it's a laptop
display),  which is why you get the errors on "ddcutil capabilities". 
The current development tree performs an additional "sanity" test of DDC
communication before concluding it's found a DDC capable monitor.

The second thing that's going on with the external monitors is truly
weird.   "ddcutil environment' probes the DDC environment in multiple
ways.   The system command i2cdetect examines I2C buses, using
essentially the same algorithm as ddcutil.   Like ddcutil, it finds that
/dev/i2c-3, 5, 6, and 7 have both slave address active.   So these would
appear to be  monitors.  /dev/i3c-3 is the laptop display, which we know
about. 

On the other hand, udev combined with xrandr reports displays on i2c-3
(eDP-1), i2c-4 (DP-1) and i2c-5 (DP2).    Again, i2c-3 (eDP-1) is the
laptop display.  So i2cdetect and udev are in disagreement, which hints
at the failures you're seeing.  I suspect a video driver issue.   This
will require some thought, and probably additional code to test your
environment further. 

Are the external monitors connected to the dock individually, or are you
using DisplayPort chaining? 

What happens if you use the DisplayPort connection on the laptop instead
of the dock?

Thanks for your help. 

Sanford

thmcmahon, 2017-01-17:
I'll answer the last questions first as this command dump is long.

1. Monitors connected to the dock individually, one with display port the other with DVI.
2. Don't know - laptop has mini display port. I think I've got one at home so I'll test that.

Here's the dump of `sudo ddcutil interrogate` - it segfaulted so let m eknow if you want me to run again.

```
Setting output level verbose...
Setting maximum retries...
Forcing --stats...
This command will take a while to run...

System information (uname):
   Linux version 4.8.0-32-generic (buildd@lcy01-34) (gcc version 6.2.0 20161005 (Ubuntu 6.2.0-5ubuntu12) ) #34-Ubuntu SMP Tue Dec 13 14:30:43 UTC 2016

*** Primary Check 1: Identify video card and driver ***
Obtaining card and driver information from /sys...

Determining driver name and possibly version...
   Driver name:    i915
   Unable to determine driver version

Video card identification:
   Vendor:              8086       Intel Corporation
   Device:              1916       HD Graphics 520
   Subvendor/Subdevice: 17aa/504a  Lenovo

*** Primary Check 2: Check that /dev/i2c-* exist and writable ***

Checking /dev/i2c-* devices...

Unless the system is using the AMD proprietary driver fglrx, devices /dev/i2c-*
must exist and the logged on user must have read/write permission for those
devices (or at least those devices associated with monitors).
Typically, this access is enabled by:
  - setting the group for /dev/i2c-* to i2c
  - setting group RW permissions for /dev/i2c-*
  - making the current user a member of group i2c
Alternatively, this could be enabled by just giving everyone RW permission
The following tests probe for these conditions.

Checking for /dev/i2c-* devices...
   crw-rw---- 1 root i2c 89, 0 Jan 12 15:44 /dev/i2c-0
   crw-rw---- 1 root i2c 89, 1 Jan 12 15:44 /dev/i2c-1
   crw-rw---- 1 root i2c 89, 2 Jan 12 15:44 /dev/i2c-2
   crw-rw---- 1 root i2c 89, 3 Jan 12 15:44 /dev/i2c-3
   crw-rw---- 1 root i2c 89, 4 Jan 12 15:44 /dev/i2c-4
   crw-rw---- 1 root i2c 89, 5 Jan 12 15:44 /dev/i2c-5
   crw-rw---- 1 root i2c 89, 6 Jan 17 11:18 /dev/i2c-6
   crw-rw---- 1 root i2c 89, 7 Jan 17 11:18 /dev/i2c-7

Current user: root (0)

Current user (root) has RW access to all /dev/i2c-* devices.

Checking for group i2c...
   Group i2c exists
   WARNING: Current user root is NOT a member of group i2c

Looking for udev nodes files that reference i2c:
   grep: /etc/udev/makedev.d/*: No such file or directory

Looking for udev rules files that reference i2c:
   /lib/udev/rules.d/40-i2c-tools.rules:KERNEL=="i2c-[0-9]*", GROUP="i2c", MODE="0660"
   grep: /run/udev/rules.d/*rules: No such file or directory
   grep: /etc/udev/rules.d/*rules: No such file or directory

*** Primary Check 3: Check that module i2c_dev is loaded ***

[... 2334 lines cut ...]
Getting data for table VCP code 0xfd - Manufacturer Specific:
VCP code 0xfd (Manufacturer Specific         ): Maximum retries exceeded

Getting data for non-table VCP code 0xfe - Manufacturer Specific:
VCP code 0xfe (Manufacturer Specific         ): Maximum retries exceeded

Getting data for table VCP code 0xfe - Manufacturer Specific:
VCP code 0xfe (Manufacturer Specific         ): Maximum retries exceeded

Getting data for non-table VCP code 0xff - Manufacturer Specific:
VCP code 0xff (Manufacturer Specific         ): Maximum retries exceeded

Getting data for table VCP code 0xff - Manufacturer Specific:
VCP code 0xff (Manufacturer Specific         ): Maximum retries exceeded

Unable to read or parse capabilities.
Skipping comparison of declared capabilities to observed features

Unable to calculate color temperature from VCP features x0B and x0C

Probing display 2
(ddc_open_display) No EDID for device on bus /dev/i2c-6
Unable to open display bus /dev/i2c-6, status code -3016 (DDCRC_EDID)
Pr
```

rockowitz, 2017-01-19:
Hi Thomas,

Thanks for the detailed output.  It really does look like something odd
is going on with the I2C buses on the dock.  I strongly suspect the
video driver which creates the I2C buses.   It would be interesting to
see what happens when a monitor is plugged directly into the laptop,
both when it's docked and undocked.  Another possibility, though not
likely, is the monitor.  Some monitors are just ill-behaved.  I have a
Dell P2411 that fails at DDC communication, and testing a second one
showed similar problems.   So if you have a different model monitor that
you can easily plug in, that would be interesting to check as well.

The following commands would give me most of the information I need,
without running the full interrogate:

  ddcutil detect --verbose
  ddcutil environment --verbose

(Please send any extensive output as an attachment.  It's easier for me
to work with.)

It would be interesting to see how other programs behave.    The
simplest to try get-edid and parse-edid programs in package read-edid.

   for i in {0..7}; do get-edid -b $i | parse-edid; done

If your laptop dual boots into Windows, there's are very helpful Windows
program for exercising DDC at http://entechtaiwan.com/lib/softmccs.shtm

I'll try adding some more diagnostics in the next point release, but I
suspect to really diagnose the issue I'll need to put my hands on a
similar system.

Thanks again for your help.  I realize that I'm asking you to plug and
unplug equipment, which is inconvenient.  Whatever you can do would be
useful. 

Sanford

rockowitz, 2017-01-27:
Thomas

ddcutil version 0.7.1 has been released.  There's a new option
--force-slave-address, which may address your problems of monitor
detection.  I say "may" because I don't have a test environment that
exhibits the behavior you're seeing. 

The segfault should also be fixed.

Please run the following commands and send me the output:

  ddcutil detect --verbose --force-slave_address --stats
  ddcutil environment --verbose --force-slave-address --stats

These commands will produce the subset of "ddcutil interrogate"
information that is relevant to monitor detection. 

Please lease send the output as attachments.  It's easier to work with.  

Thanks,
Sanford

On 01/16/2017 09:00 PM, Thomas McMahon wrote:
> I'll answer the last questions first as this command dump is long.
>
>  1. Monitors connected to the dock individually, one with display port
>     the other with DVI.
>  2. Don't know - laptop has mini [...]

thmcmahon, 2017-02-02:
OK I've install 0.7.2

`sudo ddcutil detect --verbose --force-slave-address --stats > ddcutil1.txt`

[ddcutil1.txt](https://github.com/rockowitz/ddcutil/files/746876/ddcutil1.txt)

`sudo ddcutil environment --verbose --force-slave-address --stats > ddcutil2.txt`

[ddcutil2.txt](https://github.com/rockowitz/ddcutil/files/746877/ddcutil2.txt)

rockowitz, 2017-02-02:
Thomas,

Thanks for taking the trouble to create the reports.  

First, to state the obvious, --force-slave-address has no effect on your
situation.   If the failure mode addressed by --force-slave-address had
occurred, there would have been traces in the logs.

The enhanced environment code in 0.7.2 is consistent with earlier
conjectures.  The problem is most likely in one of these places:

- The intel video driver generally
- The dock hardware
- How the video driver handles the dock
- The monitor model

Interestingly, the (enhanced) environment log shows an errno status code
on /dev/i2c-5 never before seen: EREMOTEIO.   This appears to be coming
from the i2c-over-aux code in the driver.  (file gpu/drm/drm_dp_helper.c
or gpu/drm/drm_dp_mst_topology.c)

So I'm afraid all I can suggest is testing if hardware changes have any
effect:
- plug monitor directly into laptop
- try a different monitor

There's one other peculiarity I see in the logs, though I don't think
it's related to your problem.  The monitor EDIDs for the U2142Ms read
directly by ddcutil over I2c do not quite match those reported by
xrandr.    In particular, the former do not contain a string serial
number, the latter do.  I have a few conjectures as to why this might be
the case, and have added code to the environment command that will be
included in the next point release. 

Regards,
Sanford

mixmastamyk, 2017-05-04:
I have a new Dell TB16 (thunderbolt) dock that seems to be interfering with DDC to my monitor as well.  

I used to use a script with ddccontrol in it to change the brightness on my directly-connected monitor.  With the dock in between, it no longer works.  Read about ddcutil, then installed it, but it doesn't seem to be able to talk to it either through the dock.

`ddcutil environment --verbose > ddc_environment.txt`

[ddc_environment.txt](https://github.com/rockowitz/ddcutil/files/974988/ddc_environment.txt)

`ddcutil detect --verbose > ddc_detect.txt`
[ddc_detect.txt](https://github.com/rockowitz/ddcutil/files/974989/ddc_detect.txt)

Perhaps these can shed some light, however I suspect the problem is with the dock, firmware, or drivers themselves.

rockowitz, 2017-05-04:
Thanks for the report.  Yes, it does seem to be another case of I2C
passthrough on a dock not working.  All the other reports have been for
Lenovo Thinkpad docks, so it looks like the problem is more general. 
Common to all these cases is the i915 Intel video driver.

Looks like you're connecting the monitor to the dock via DisplayPort. 
Have you tried DVI, HDMI, or (gasp) VGA?  One person was able to work
around the problem by having both IIRC DisplayPort and VGA connections
to the monitor. He sets the monitor input to the DisplayPort
connection.  Surprisingly, the monitor is nonetheless response to
I2C/DDC communication on the VGA connection.

Sanford

mixmastamyk, 2017-05-04:
Good to know, thanks.

The laptop has an HDMI port so I could use it, but I don't have a cable handy, and not sure it supports 4k @ 60hz.  Also, the reason I bought this dock in the first place was to get rid of all these cables crossing my desk, and replacing it with one.  Connecting directly defeats 50% of the reason for the purchase, sigh.

rockowitz, 2017-05-04:
I looked a the specs for the dock and it seemed to have all the
connections.

Sanford

mixmastamyk, 2017-05-04:
Hah, just remembered my daughter's kano computer came with an unused HDMI cable, so plugged it in and it works as an extra channel to communicate!  Big, bright yellow Fischer-Price cable sticking out of my "professional" black laptop, haha.

Still curious if a video driver could interfere with I2C/DDC?

rockowitz, 2017-05-04:
As as been discussed, there seems to be a pattern of I2C communication
failing with docking stations and the i915 driver.  It's not
particularly useful for me to file a vague bug report with the i2c
folks, but if I can identify a pattern, .e.g. the problem is associated
with certain recent processors, that may be more useful.   Please run
"ddcutil interrogate" as root and send me the output as a file.

Sanford

mixmastamyk, 2017-05-04:
Ok. [ddc_interrogate.txt](https://github.com/rockowitz/ddcutil/files/977270/ddc_interrogate.txt)

rockowitz, 2017-05-04:
Was that the whole output?  The file ends abruptly at line 2338 in the
middle of a report.  You might try sending it as an attachment.

Also, what is your processor.  The video VID/PID is 8086/591B, which
appears to be in the Intel graphics range, but isn't in the PID
database, and I can't find the number anywhere online.  Must be very new.

Sanford

mixmastamyk, 2017-05-05:
I think so.  I did run it again and got this:

    Job 2, “ddcutil interrogate > ddc_inter…” terminated by signal SIGSEGV (Address boundary error)

Laptop is brand new, but has been around since January I think.

    Intel(R) Core(TM) i5-7440HQ CPU @ 2.80GHz

https://ark.intel.com/products/97459/Intel-Core-i5-7440HQ-Processor-6M-Cache-up-to-3_80-GHz

rockowitz, 2017-05-05:
Can you send what was written to ddc_inter_?  Where it ends would narrow
down where the fault is. 

Even better, can you run it under valgrind?  i.e. "valgrind ddcutil
interrogate"

Alternatively, "ddcutil interrogate" is essentially a combination of the
following commands.  Which of the these fail?  And where?

  ddcutil detect --verbose
  ddcutil environment --verbose
  ddcutil usbenvironment --verbose
  and, for each valid display number N:
  ddcutil probe --display N probe
 
Thanks.  I regard this as serious.  Configuration issues go with the
territory for ddcutil. It's in the nature of I2C.  But segfaults are my
doing.

Sanford

mixmastamyk, 2017-05-05:
See bug #21 .

rockowitz, 2017-05-05:
Re the original problem before we got distracted by the segfault, here's
a (still incomplete)l exchange today with someone on the i915 team:

https://lists.freedesktop.org/archives/intel-gfx/2017-May/127424.html

Bottom line, the DDC/I2C failures when using a display connected to a
recent docking station appear to be in the DisplayPort driver code,
which affects the DVI and HDMI connectors, not just the DisplayPort
connector.  Until that's fixed, there's not much ddcutil can do.  

But here's the positive way to think about it.  Your bug report helped
me see a pattern, which led to identifying a driver bug.  So hopefully
this will get fixed, though it will be a while before the fix makes its
way into released code. 

Sanford

rockowitz, 2017-05-07:
Closing this issue since it has been determined to be a problem in the DisplayPort Multi-Stream I2C driver code.

For a detailed description of this issue, see this [discussion thread](https://lists.freedesktop.org/archives/intel-gfx/2017-May/127424.html). 

Updates will be posted on the [ddcutil FAQ](http://www.ddcutil.com/faq/#docking).

closed 2017-05-07 by rockowitz

mixmastamyk, 2017-05-08:
Nice, so it looks like this may be fixed sometime in the future.  Tracked down the bug at freedesktop.org:

- https://bugs.freedesktop.org/show_bug.cgi?id=100954

(Looks like the developer did not get CC'd on the bug as he asked, unless he is on the mailing list, which is likely.)

rockowitz, 2017-05-08:
Could not CC him within the bug, unrecognized email address.  Sent him
email directly.

bhack, 2018-12-05:
@rockowitz Can we do something to create more priority on the upstream bugs? Seems anything interesting in the last month.

rockowitz, 2018-12-06:
I suggest you add yourself to the CC list for the following bug: i2c-dev 
failures with recent chips and docking stations 
<https://bugs.freedesktop.org/show_bug.cgi?id=100954>

Sanford

bhack, 2018-12-14:
/cc @libthinkpad

saveman71, 2019-01-07:
@rockowitz I have the same issue with a lenovo dock. Can I contribute in anyway?

I added myself as a CC to the issue linked above.

I confirm plugging in the HDMI cable directly to the laptop works as expected. The monitor is a Dell P2418D.

<details>
<summary>Connected to DP via dock</summary>

```
Setting output level verbose...
Setting maximum retries...
Forcing --stats...
Forcing --force-slave-address..
This command will take a while to run...

   Current time (local): 2019-01-07 20:22:29 CET
   Current time (UTC):   2019-01-07 19:22:29
   Seconds since boot:   450431

*** Basic System Information ***

ddcutil version: 0.9.4

/proc/version:
   Linux version 4.20.0-arch1-1-ARCH (builduser@heftig-29859) (gcc version 8.2.1 20181127 (GCC)) #1 SMP PREEMPT Mon Dec 24 03:00:40 UTC 2018

Architecture:     x86_64
Distributor id:   Arch
Release:          rolling
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=/vmlinuz-linux root=/dev/mapper/MZVLW256HEHP-root rw cryptdevice=UUID=d9b2f01c-1b0f-4059-bb92-308c5c729572:cryptolvm:allow-discards resume=/dev/mapper/MZVLW256HEHP-swap

Compiler information:
   C standard: 201710
   gcc compatible compiler:
      Compiler version: 8.2.1

Processor information as reported by lscpu:
   Architecture:        x86_64
   CPU op-mode(s):      32-bit, 64-bit
   Byte Order:          Little Endian
   Address sizes:       39 bits physical, 48 bits virtual
   CPU(s):              4
   On-line CPU(s) list: 0-3
   Thread(s) per core:  2
   Core(s) per socket:  2
   Socket(s):           1
   NUMA node(s):        1
   Vendor ID:           GenuineIntel
   CPU family:          6
   Model:               142
   Model name:          Intel(R) Core(TM) i7-7500U CPU @ 2.70GHz
   Stepping:            9
   CPU MHz:             799.267
   CPU max MHz:         3500.0000
   CPU min MHz:         400.0000
   BogoMIPS:            5810.00
   Virtualization:      VT-x
   L1d cache:           32K
   L1i cache:           32K
   L2 cache:            256K
   L3 cache:            4096K
   NUMA node0 CPU(s):   0-3
   Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp

DMI Information from /sys/class/dmi/id:
   Motherboard vendor:       LENOVO
[... 3699 lines cut ...]

Examining system logs...
Current timestamps:
   Current time (local): 2019-01-07 20:26:34 CET
   Current time (UTC):   2019-01-07 19:26:34
   Seconds since boot:   450675

   Scanning dmesg output for I2C related entries...
   Executing command: dmesg
      [91292.007266] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
      [104333.055168] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
      [104343.141274] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
      [104642.028340] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
      [106353.293987] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
      [109117.003905] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
      [119755.590998] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
      [122358.883869] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
      [134597.647222] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
      [134598.182814] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
      [134631.193116] drm_dp_i2c_do_msg: 2 callbacks suppressed
      [134694.132608] drm_dp_i2c_do_msg: 2 callbacks suppressed
      [134704.665076] drm_dp_i2c_do_msg: 2 callbacks suppressed
      [134712.819451] drm_dp_i2c_do_msg: 2 callbacks suppressed
      [134885.106197] drm_dp_i2c_do_msg: 2 callbacks suppressed
      [135124.303862

rockowitz, 2019-01-08:
On 1/7/19 2:31 PM, Antoine Bolvy wrote:
>
> @rockowitz <https://github.com/rockowitz> I have the same issue with a 
> lenovo dock. Can I contribute in anyway?
>
> I added myself as a CC to the issue linked above
>
Adding yourself to the CC list for the freedesktop.org bug is the most 
useful thing to do in my estimation, as it indicates the significance of 
the problem to the i915 driver developers.  If and when they address the 
problem, then it would be useful to build their latest driver and test 
if it works.

> I confirm plugging in the HDMI cable directly to the laptop works as 
> expected. The monitor is a Dell P2418D.
>
> Connected to DP via dock
> |Setting output level verbose... Setting maximum retries... Forcing 
> --stats... Forcing --force-slave-address.. This command will take a 
> while to run... Current time (local): 2019-01-07 20:22:29 CET Current 
> time (UTC): 2019-01-07 19:22:29 Seconds since boot: 450431 *** Basic 
> System Information *** ddcutil version: 0.9.4 /proc/version: Linux 
> version 4.20.0-arch1-1-ARCH (builduser@heftig-29859) (gcc version 
> 8.2.1 20181127 (GCC)) #1 SMP PREEMPT Mon Dec 24 03:00:40 UTC 2018 
> Architecture: x86_64 Distributor id: Arch Release: rolling Found a 
> known architecture /proc/cmdline: BOOT_IMAGE=/vmlinuz-linux 
> root=/dev/mapper/MZVLW256HEHP-root rw 
> cryptdevice=UUID=d9b2f01c-1b0f-4059-bb92-308c5c729572:cryptolvm:allow-discards 
> resume=/dev/mapper/MZVLW256HEHP-swap Compiler information: C standard: 
> 201710 gcc compatible compiler: Compiler version: 8.2.1 Processor 
> information as reported by lscpu: Architecture: x86_64 CPU op-mode(s): 
> 32-bit, 64-bit Byte Order: Little Endian Address sizes: 39 bits 
> physical, 48 bits virtual CPU(s): 4 On-line CPU(s) list: 0-3 Thread(s) 
> per core: 2 Core(s) per socket: 2 Socket(s): 1 NUMA node(s): 1 Vendor 
> ID: GenuineIntel CPU family: 6 Model: 142 Model name: Intel(R) 
> Core(TM) i7-7500U CPU @ 2.70GHz Stepping: 9 CPU MHz: 799.267 CPU max 
> MHz: 3500.0000 CPU min MHz: 400.0000 BogoMIPS: 5810.00 Virtualization: 
> VT-x L1d cache: 32K L1i cache: 32K L2 cache: 256K L3 cache: 4096K NUMA 
> node0 CPU(s): 0-3 Flags: fpu vme de pse tsc msr pae mce cx8 apic sep 
> mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht 
> tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs 
> bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf 
> tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 
> sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt 
> tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch 
> cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept 
> vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx 
> rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves 
> dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp DMI 
> Information from /sys/class/dmi/id: Motherboard vendor: LENOVO 
> Motherboard product name: 20HFCTO1WW System vendor: LENOVO System 
> product name: 20HFCTO1WW Chassis vendor: LENOVO Chassis type: 10 - 
> Notebook Byte order checks: Is big endian (local test): false 
> WORDS_BIGENDIAN macro (autoconf): not defined __BYTE_ORDER__ macro 
> (gcc): __ORDER_LITTLE_ENDIAN__ *** Primary Check 1: Identify video 
> card and driver *** Obtaining card and driver information from /sys... 
> Primary video controller at PCI address 0000:00:02.0 (boot_vga flag is 
> set) Device class: x030000 VGA compatible controller Vendor: x8086 
> Intel Corporation Device: x5916 HD Graphics 620 Subvendor/Subdevice: 
> 17aa/224b Lenovo Driver name: i915 Driver version: Unable to determine 
> I2C device: i2c-3 name: i915 gmbus dpd I2C device: i2c-1 name: i915 
> gmbus dpc I2C device: i2c-2 name: i915 gmbus dpb I2C device: i2c-7 
> name: DPMST *** Primary Check 2: Check that /dev/i2c-* exist and 
> writable *** Current user: root (0) Checking /dev/i2c-* devices... 
> Unless the system is using the AMD proprietary driver fglrx, devices 
> /dev/i2c-* must exist and the logged on user must have read/write 
> permission for those devices (or at least those devices associated 
> with monitors). Typically, this access is enabled by: - setting the 
> group for /dev/i2c-* to i2c - setting group RW permissions for 
> /dev/i2c-* - making the current user a member of group i2c 
[... 861 lines cut ...]
> Description 15 EREMOTEIO ( -121) Remote I/O error 1 DDCRC_RETRIES 
> (-3007) maximum retries exceeded (derived) Total errors: 16 Errors 
> Wrapped in Retry: Count Status Code Description 15 EREMOTEIO ( -121) 
> Remote I/O error Total errors: 15 Sleep Strategy Stats: Total IO 
> events: 130 IO error count: 16 Total sleep events: 1 Sleep Event type 
> Count SE_WRITE_TO_READ 0 SE_POST_OPEN 1 SE_POST_WRITE 0 SE_POST_READ 0 
> SE_DDC_NULL 0 SE_POST_SAVE_SETTINGS 0 Call Stats: Type Count Millisec 
> ( Nanosec) write calls (IE_WRITE) 55 826 ( 826610062) read calls 
> (IE_READ) 7 192 ( 192755604) open file calls (IE_OPEN) 16 0 ( 712025) 
> close file calls (IE_CLOSE) 16 0 ( 205127) other I/O calls (IE_OTHER) 
> 36 0 ( 163503) Totals: 130 1020 ( 1020446321) Sleep Call Stats: Total 
> sleep calls: 15 Requested sleep time milliseconds : 750 Actual sleep 
> milliseconds (nanosec): 752 ( 752549529) Total elapsed milliseconds 
> (nanoseconds): 20802 ( 20802920252) *** Detected Displays *** Invalid 
> display I2C bus: /dev/i2c-4 I2C address 0x50 (EDID) present: true Is 
[... cut ...]
> Manufacture year: 2016 EDID version: 1.4 Product code: 1313 Extra 
> descriptor: LP140WF6-SPB6 Video input definition: 0x95 - Digital Input 
> (DisplayPort) Supported features: DPMS standby DPMS suspend DPMS 
> active-off Digital display type: RGB 4:4:4 Standard sRGB color space: 
> False White x,y: 0.313, 0.329 Red x,y: 0.580, 0.360 Green x,y: 0.335, 
> 0.560 Blue x,y: 0.160, 0.120 Extension blocks: 0 EDID source: EDID hex 
> dump: +0 +4 +8 +c 0 4 8 c +0000 00 ff ff ff ff ff ff 00 30 e4 21 05 00 
> 00 00 00 ........0.!..... +0010

mattcph, 2019-10-23:
For what it's worth, I have 2 DP Dell U2717D monitors via the lenovo ultra dock, and a x270, and they work great. Unfortunately, ddcutil loses it's ability to control brightness when connected directly to 1 monitor, as per this description.

Seems like the gory thread was useful at the least for pinpointing the actual cause.

Very keen to assist in testing or other such things to solve for the docking peeps.

saveman71, 2019-10-23:
@mattcph I saw your comment on the freedesktop.org bug, you say you filed a bug with lenovo, is it public? Any link to this report?

mattcph, 2019-10-23:
https://forums.lenovo.com/t5/ThinkPad-X-Series-Laptops/Ultra-Dock-90W-eu-thinkpads-brightness-control/td-p/4556548

jbbr, 2019-10-25:
I have an Dell TB16 connected to a XPS 9560. Monitor is connected via DP to the dock.
On Windows I tried multiple DDC utils like clickmonitorddc and they all work just fine.
On Linux ddcutils & ddccontrol both work only if monitors are directly connected to the laptop. 

So at least for my specific setup I can say that this is a non-issue when using Windows.

mixmastamyk, 2019-10-25:
Same here, Dell set up, I get around the problem (literally) by having a second cable from my laptop to the monitor.  :-/  

Kinda defeats the purpose of the docking station but I use it for other things as well.

evgeni, 2020-06-28:
FWIW, the upstream bug got migrated to https://gitlab.freedesktop.org/drm/intel/-/issues/37 but still no fix :(

mixmastamyk, 2020-12-14:
This should help, said to be in Linux kernel 5.10, in distros in 2021:

https://lkml.org/lkml/2020/9/1/300

rockowitz, 2020-12-15:
I have checked.  The changes are in the just released 5.10 kernel.

evgeni, 2020-12-16:
I've tested this on 5.10.1 and it works fine, the only interesting thing is, my monitor shows up twice now:

```
% sudo ddcutil detect    
[sudo] password for evgeni: 
Invalid display
   I2C bus:             /dev/i2c-4
   EDID synopsis:
      Mfg id:           BOE
      Model:            
      Serial number:    
      Manufacture year: 2017
      EDID version:     1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:             /dev/i2c-5
   EDID synopsis:
      Mfg id:           AOC
      Model:            U2790B
      Serial number:    
      Manufacture year: 2020
      EDID version:     1.4
   DDC communication failed

Display 1
   I2C bus:             /dev/i2c-7
   EDID synopsis:
      Mfg id:           AOC
      Model:            U2790B
      Serial number:    
      Manufacture year: 2020
      EDID version:     1.4
   VCP version:         2.2
```

Probably not really `ddcutil` related, but worth documenting?

rockowitz, 2020-12-16:
First, thank you @thmcmahon for reporting the success with  the 5.10 kernel.   Many people are interested in this problem. 

Multiple /dev/i2c devices for the same monitor is a known problem.  See, the [FAQ](https://www.ddcutil.com/faq/#duplicate_displayport).   Clearly a driver problem, and I've been adding more and more code to the **environment** command to analyse this situation.  Can I ask you to run **ddcutil environment --verbose** as root and submit the output as an attachment?

I've created a new issue, #160, specifically for this problem.  Let's continue the discussion there.

suuuehgi, 2021-01-08:
A Christmas miracle! I had already stopped hoping that this would work someday.

So, YES, I can confirm it to be working too (kernel v 5.10.4) on a Lenovo ThinkPad Thunderbolt 3 Dock Gen 2.

What a nice surprise! What a relieve!