rockowitz/ddcutil #12: ddcutil 0.7.0 doesn't build when --enable-lib=no.

dracwyrm, 2017-01-17:
Hi,

The way I understand it, disabling shared lib will then disable the building of the C API. However, I receive the following error when trying to build without shared lib.

Also, please note that I have SWIG disabled, but it is finding python. If swig is not enabled, then it shouldn't even look for it.

```
>>> Install ddcutil-0.7.0 into /var/tmp/portage/app-misc/ddcutil-0.7.0/image/ category app-misc
make -j2 DESTDIR=/var/tmp/portage/app-misc/ddcutil-0.7.0/image/ install 
Making install in src
make[1]: Entering directory '/var/tmp/portage/app-misc/ddcutil-0.7.0/work/ddcutil-0.7.0/src'
Making install in .
make[2]: Entering directory '/var/tmp/portage/app-misc/ddcutil-0.7.0/work/ddcutil-0.7.0/src'
Executing hook all-local...
make[3]: Entering directory '/var/tmp/portage/app-misc/ddcutil-0.7.0/work/ddcutil-0.7.0/src'
 /bin/mkdir -p '/var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/include'
 /bin/mkdir -p '/var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/bin'
 /usr/lib/portage/python3.5/ebuild-helpers/xattr/install -c -m 644 public/ddcutil_types.h public/ddcutil_c_api.h '/var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/include'
  /bin/sh ../libtool   --mode=install /usr/lib/portage/python3.5/ebuild-helpers/xattr/install -c ddcutil '/var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/bin'
libtool: install: /usr/lib/portage/python3.5/ebuild-helpers/xattr/install -c ddcutil /var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/bin/ddcutil
make  install-exec-hook
make[4]: Entering directory '/var/tmp/portage/app-misc/ddcutil-0.7.0/work/ddcutil-0.7.0/src'
(install-exec-hook) Executing...
  pythondir       = /usr/lib64/python3.4/site-packages
  pyexecdir       = /usr/lib64/python3.4/site-packages
/var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64
ls -ld /var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64
ls: cannot access '/var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64': No such file or directory
make[4]: *** [Makefile:1980: install-exec-hook] Error 2
make[4]: Leaving directory '/var/tmp/portage/app-misc/ddcutil-0.7.0/work/ddcutil-0.7.0/src'
make[3]: *** [Makefile:1845: install-exec-am] Error 2
make[3]: Leaving directory '/var/tmp/portage/app-misc/ddcutil-0.7.0/work/ddcutil-0.7.0/src'
make[2]: *** [Makefile:1753: install-am] Error 2
make[2]: Leaving directory '/var/tmp/portage/app-misc/ddcutil-0.7.0/work/ddcutil-0.7.0/src'
make[1]: *** [Makefile:1586: install-recursive] Error 1
make[1]: Leaving directory '/var/tmp/portage/app-misc/ddcutil-0.7.0/work/ddcutil-0.7.0/src'
make: *** [Makefile:553: install-recursive] Error 1
```

With the same configure options, but with shared libs enabled, it installs just fine.

The config options:
```
>>> Configuring source in /var/tmp/portage/app-misc/ddcutil-0.7.0/work/ddcutil-0.7.0 ...
 * econf: updating ddcutil-0.7.0/config/config.guess with /usr/share/gnuconfig/config.guess
 * econf: updating ddcutil-0.7.0/config/config.sub with /usr/share/gnuconfig/config.sub
./configure --prefix=/usr --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/share --sysconfdir=/etc --localstatedir=/var/lib --disable-dependency-tracking --disable-silent-rules --docdir=/usr/share/doc/ddcutil-0.7.0 --htmldir=/usr/share/doc/ddcutil-0.7.0/html --libdir=/usr/lib64 --disable-lib --disable-usb --disable-swig
configure: debug messages disabled
checking for a BSD-compatible install... /usr/lib/portage/python3.5/ebuild-helpers/xattr/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for style of include used by make... GNU
checking for x86_64-pc-linux-gnu-gcc... x86_64-pc-linux-gnu-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether x86_64-pc-linux-gnu-gcc accepts -g... yes
checking for x86_64-pc-linux-gnu-gcc option to accept ISO C89... none needed
checking whether x86_64-pc-linux-gnu-gcc understands -c and -o together... yes
checking dependency style of x86_64-pc-linux-gnu-gcc... none
checking for x86_64-pc-linux-gnu-ar... x86_64-pc-linux-gnu-ar
checking the archiver (x86_64-pc-linux-gnu-ar) interface... ar
checking for x86_64-pc-linux-gnu-pkg-config... /usr/bin/x86_64-pc-linux-gnu-pkg-config
[... 179 lines cut ...]
config.status: executing depfiles commands
config.status: executing libtool commands

        ddcutil 0.7.0
        =============

        libtool version         0:0:0

        prefix:                 /usr
        datarootdir:            ${prefix}/share
        datadir:                /usr/share
        mandir:                 /usr/share/man

        adl_header_dir:         
        enable_lib:             no
        enable_swig:            no
        enable_usb:             no
        enable_doxygen          no

        compiler:               x86_64-pc-linux-gnu-gcc
        cflags:                 -march=native -O2 -w -pipe
        ldflags:                -Wl,-O1 -Wl,--as-needed

>>> Source configured.
```

rockowitz, 2017-01-19:
Jonathan,

Thanks for the detailed report.   Here's what I observe:
- You're building on Gentoo using portage.    
- "make install" is looking for  a shared library to install, even
though it doesn't exist.
- The following line contains double slashes.   This should work, but I
wonder if you commonly see this using portage

|ls -ld /var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64 |

|Did you try regenerating all the autotools files  using aclocal,
autoconf, automake before running configure? |
||
|Yes, the Python check is superfluous. It's benign if Python is not
found and you're not building the SWIG interface,
but it can be distracting. I'll make it conditional. |
||
|Sanford|
||

dracwyrm, 2017-01-19:
Hi,

The auto files are regenerated each time a build is run. Even re-extracting the files so it's a 100% clean build from scratch. Sometimes the // happens because the variable is defined with ending in a slash and the path is set to begin with a slash, so when the two are concatenated together, the double slash happens. The variable for the path to install dir (/var/tmp/portage/app-misc/ddcutil-0.7.0/image/) ends in the slash, and the way the install dir in the makefiles might begin with / for /usr/lib.

However, I think I found the issue:
In the compile output there is this:
```
(install-exec-hook) Executing...
  pythondir       = /usr/lib64/python3.5/site-packages
  pyexecdir       = /usr/lib64/python3.5/site-packages
/var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64
ls -ld /var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64
ls: cannot access '/var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64': No such file or directory
```
I looked for that in the make files. In "src/Makefile.am" there is this at the very bottom:
```
install-exec-hook:
	@echo "(install-exec-hook) Executing..." 
	@echo "  pythondir       = $(pythondir)"
	@echo "  pyexecdir       = $(pyexecdir)"
	@echo  $(DESTDIR)$(libdir)
	ls -ld $(DESTDIR)$(libdir)
	ls -l $(DESTDIR)$(libdir)/*la
	sed -i "/dependency_libs/ s/'.*'/''/" $(DESTDIR)$(libdir)/libddcutil.la
```
When there are no shared libs, I'm guess there are no libs installed to /usr/lib{32,64}, so this section should not be executed at all. It is trying to access a directory that doesn't exist, and then it will try to list files that don't exists and then it will try to sed a file that doesn't exist.

Hope this helps.

rockowitz, 2017-01-19:
Jonathan,

Yes, that's it.   Thank you. 

The sed script exists to clean up the .la file to avoid a lintian
warning.   Since you're on Gentoo, I believe you should be able to just
delete the
sed line with no ill effect.   Or you can replace it with:

    @if [ -f $(DESTDIR)$(libdir)/libddcutil.la ] ; then \
      echo "(install-exec-hook) Stripping dependency_libs from
libddcutil.la" ; \
      sed -i "/dependency_libs/ s/'.*'/''/"
$(DESTDIR)$(libdir)/libddcutil.la ; \
    fi

This change will go into the next point release.  

Sanford

dracwyrm, 2017-01-20:
Sanford,

The test needs to detect the directory  because it's also trying to use ls command in a directory that doesn't exist, which is the error:
```
ls: cannot access '/var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64': No such file or directory
```
because that directory doesn't exist at all when using --disable-lib
```
if [ -d $(DESTDIR)$(libdir) ]; then
	@echo  $(DESTDIR)$(libdir)
	ls -ld $(DESTDIR)$(libdir)
	ls -l $(DESTDIR)$(libdir)/*la
	sed -i "/dependency_libs/ s/'.*'/''/" $(DESTDIR)$(libdir)/libddcutil.la
fi
```

Cheers.

dracwyrm, 2017-01-21:
I did a bit more testing. I was able to compile ddcutil with lib disabled. However, there are some things that are installed that shouldn't be.

ddcutil.pc file is installed pointing to include and libs, but the libs are not installed:
```
ddcutil.pc:
prefix=/usr
exec_prefix=${prefix}
libdir=/usr/lib64
includedir=${prefix}/include

Name: ddcutil
Description: Control display settings
URL: http://www.ddcutil.com
Version: 0.7.0
# Requires:  ??
# Libs and Cflags not needed since using default locations
Libs: -L${libdir} 
Cflags: -I${includedir}
```

Also, the header files are installed: /usr/include/{ddcutil_c_api.h,ddcutil_types.h} which mention that it's for the C API, however, the C API needed the shared libs that are not installed. Thus, the header files should not be installed as well.

There must be a better way of doing the patch as when I run compile, this is outputted:
```
(install-exec-hook) Executing...
  pythondir       = /usr/lib64/python3.5/site-packages
  pyexecdir       = /usr/lib64/python3.5/site-packages
if [[ -d /var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64 ]]; then \
        @echo  /var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64; \
        ls -ld /var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64; \
        ls -l /var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64/*la; \
        sed -i "/dependency_libs/ s/'.*'/''/" /var/tmp/portage/app-misc/ddcutil-0.7.0/image//usr/lib64/libddcutil.la; \
fi
```
So it doesn't make a very clean build log.

This is the patch I used:
```
diff -purN a/src/Makefile.am b/src/Makefile.am
--- a/src/Makefile.am	2017-01-02 21:42:04.000000000 +0000
+++ b/src/Makefile.am	2017-01-20 12:28:35.192559062 +0000
@@ -366,10 +366,12 @@ install-exec-hook:
 	@echo "(install-exec-hook) Executing..." 
 	@echo "  pythondir       = $(pythondir)"
 	@echo "  pyexecdir       = $(pyexecdir)"
-	@echo  $(DESTDIR)$(libdir)
-	ls -ld $(DESTDIR)$(libdir)
-	ls -l $(DESTDIR)$(libdir)/*la
-	sed -i "/dependency_libs/ s/'.*'/''/" $(DESTDIR)$(libdir)/libddcutil.la
+	if [[ -d $(DESTDIR)$(libdir) ]]; then \
+		@echo  $(DESTDIR)$(libdir); \
+		ls -ld $(DESTDIR)$(libdir); \
+		ls -l $(DESTDIR)$(libdir)/*la; \
+		sed -i "/dependency_libs/ s/'.*'/''/" $(DESTDIR)$(libdir)/libddcutil.la; \
+	fi
 
 # objdump -p $(DESTDIR)$(libdir)/libddcutil.so | sed -n -e's/^[[:space:]]*SONAME[[:space:]]*//p' |  sed -r -e's/([0-9])\.so\./\1-/; s/\.so(\.|$)//; y/_/-/; s/(.*)/\L&/'
 

```

Hope this helps.

rockowitz, 2017-01-21:
More for the TODO list.    I really appreciate your reports.  Getting
build code to work everywhere exposes issue after issue. 

I'm reworking the Python configure.ac code right now.   Making it to
work reasonably is a bloody mess.  In particular, there doesn't seem to
be any universally right way to set pyexecdir.  From my notes:

Posts on the problems:
  
https://tschoonj.github.io/blog/2013/09/04/building-libtool-modules-python-bindings/
   https://gist.github.com/tschoonj/6441999
   https://lists.ubuntu.com/archives/ubuntu-devel/2009-February/027439.html
   https://bugzilla.redhat.com/show_bug.cgi?id=533920

Some of the issues:
- Python 3 is a distinct language from Python 2, yet AM_PYTHON treats it
as an upwardly compatible version Python 2.
- How to build for multiple version fo Python 2 an Python 3?
- Debian (and derivatives such as Ubuntu) have modified the directory
scheme.
  Directory names end with dist-packages instead of site-packages.  
  But this is incompletely implemented.   sysconfig.get_path() returns
names  ending with site-package on python 3
- sys.path is built using calls to site.get_site_packages().  But this
function  hardcodes the paths. 
 There is no reference to sysconfig.get_path() or 
distutils.sysconfig_get_python_lib(). 
 So code that relies on the latter functions can return a library that
is not in sys.path
- Where do local extensions to the distribution version of Python belong?
  In a directory under /usr/local or under /usr?
- pkg-config names can vary.
  Fedora 25:  python and python2 refer to current Python2
                   python3, python-3.5, python-3.5m refer to current
Python 3
  openSUSE 42.2: 
          python3 not defined, python-3.4 refers to current Python 3
 So we need to know the version number of Python 3 to construct a name
to pass to pkg-config

dracwyrm, 2017-01-21:
Ah. The python trap.

Gentoo is nice because there are variables that store everything for building. It doesn't matter what the system has, as in, /usr/bin/python pointing to python35 or python27. When we build something, we give the configure script all the paths. For example, a user can want the program for python 2.7 even though 3.5 is default, so the path to 2.7 and the site-directory, includes... so on are passed through to the configure script and it installs everything for the right version of python and the right locations. :) (https://devmanual.gentoo.org/eclass-reference/python-utils-r1.eclass/index.html If you want to see how we sort it all).

All you need are variables that hold python version, site directories, library, includes, so on, set sane defaults and if they are different, then they can be overridden by package maintainers. That will save you a load of headaches as you only need to know your system, and set the defaults to that.

Cheers.

rockowitz, 2017-01-23:
Changes will appear in the next point release.

>
> I did a bit more testing. I was able to compile ddcutil with lib
> disabled. However, there are some things that are installed that
> shouldn't be.
ddcutil.pc and the include files will only be installed if --enable-lib
is set

> ddcutil.pc file is installed pointing to include and libs, but the
> libs are not installed:
>
> |ddcutil.pc: prefix=/usr exec_prefix=${prefix} libdir=/usr/lib64
> includedir=${prefix}/include Name: [...]
The informational lines have been removed, as they were just for
debugging. 
File libddcutil.la must explicitly be checked for using the -f option
before calling sed.  In general, the existence of the lib64 directory
does not imply that libddcutil.la exists.

dracwyrm, 2017-01-23:
Thanks for your hard work on this. :)

dracwyrm, 2017-01-28:
0.7.1 compiles perfectly with or without shared libs enabled. Thanks for your hard work on fixing this.

Jon

closed 2017-01-28 by dracwyrm