rockowitz/ddcutil #15: curly bracket missing for c++ use

d-vogel, 2017-03-16:
Hi,
Opening curly bracket missing  [here](https://github.com/rockowitz/ddcutil/blob/master/src/public/ddcutil_c_api.h#L39)

rockowitz, 2017-03-16:
Thanks for catching this.  The fix will go into the next point release.

If you're using the ddcutil API, please let me know how it goes.  It's a
work in progress, and feedback would be helpful.   Are there features
that you need.  There are still facilities in the command line version
that aren't exposed. 

Sanford

d-vogel, 2017-03-29:
Hey, here is what I could quickly do [QMLddcutil](https://github.com/d-vogel/QMLddcutil).
Any comment appreciated :smile: 
The only this I found somewhat weird is that `DDCA_Capabilities` does not contain the descriptions for possible values of VCPs. [One has to get them from a `Version_Feature_Info` structure.](https://github.com/d-vogel/QMLddcutil/blob/master/ddcutilController.cpp#L104).

rockowitz, 2017-03-29:
>
> Hey, here is what I could quickly do QMLddcutil
> Any comment appreciated 😄

Will build it and take a look. 
>
> The only this I found somewhat weird is that |DDCA_Capabilities| does
> not contain the descriptions for possible values of VCPs. One has to
> get them from a |Version_Feature_Info| structure.

From a use case perspective, I see what you're saying.  Command "ddcutil
capabilities" does perform the lookup.   The API was designed as it is
for a couple reasons:

- Minimize the complexity of data structures returned by a single API call.
- There are subtle problems with interpreting capabilities values, and I
didn't want to prejudge how an app might handle them.  (Though of
course, at this point, I probably understand them better than anyone.) 
In particular, interpretation can vary by VCP version, and I've seen at
least one monitor where the VCP version reported in the capabilities
string differs from that reported by VCP feature DF.  So the
interpretation is a judgement call.

I'll give it some more thought.

Sanford

rockowitz, 2017-03-29:
>
> Hey, here is what I could quickly do QMLddcutil
Build instructions?  I don't see Makefile, CMakeLists.txt, or any precursor.

rockowitz, 2017-05-07:
This missing curly bracket was added in Relase 0.8.0.

closed 2017-05-07 by rockowitz