rockowitz/ddcutil #16: FindDDCUtil.cmake module ?

d-vogel, 2017-03-22:
May I suggest integration of a module for cmake ? For now I added it in my project source tree.

[https://gist.github.com/d-vogel/44defe784d49617f86fee6073ba78ff1](https://gist.github.com/d-vogel/44defe784d49617f86fee6073ba78ff1)

cmake modules are supposed to be in ` /usr/share/cmake/Modules/` as per [https://cmake.org/Wiki/CMake:How_To_Find_Libraries](https://cmake.org/Wiki/CMake:How_To_Find_Libraries)

Cheers,
Dorian

rockowitz, 2017-03-23:
Dorian,

First off, let me say that my knowledge of cmake is superficial.   So I
trust that your script is a sufficiently general way to provide cmake
integration.  (It looks like you're basically wrappering the pkg-config
file that is provided.)

Is there a convention for a subdirectory in which the script would be
located in the source tree?  ddcutil.pc.in is currently in the top level
directory.

Would you be willing to handle any questions regarding use of the
script?   As I said, my knowledge of cmake is superficial.

Are you comfortable changing the license to GPL2 for consistency with
the rest of ddcutil?

Additional changes needed:
- make install: install the script to /usr/share/cmake/Modules or
/usr/local/share/cmake/Modules
- rpm and dkpg installation scripts would also have to change.

This all seems simple enough that I could do it myself.  Or you can
create a pull request.

Sanford

d-vogel, 2017-03-29:
Feel free to use this simple module as you wish !

Dorian

rockowitz, 2017-03-29:
>
> Feel free to use this simple module as you wish !

Thanks you.

rockowitz, 2017-05-07:
File **FindDDCUtil.cmake** is in tarball directory data/usr/share/cmake/Modules. 
It is installed to directory /usr/share/ddcutil/data or /usr/local/share/ddcutil/data. 
It is left to the user to copy this file to the appropriate directory on their system (typically /usr/share/cmake/Modules).

closed 2017-05-07 by rockowitz