rockowitz/ddcutil #17: ddcutil detect --> I2C address 0x30 (EDID block#)  present: false + DDC communication failed

krugman53, 2017-04-01:
Hello I couldn't find anything related to my error message so I opened an issue. My setup is pretty complicated, with four displays (2x Asus MX279 + 2x Asus VX239) hooked to Nvidia GTX 1080 via 1x HDMI->HDMI, 2x DP->HDMI, and 1x DVI->HDMI. DDC/CI is enabled in every display's OSD, yet all seem to output the same pattern:

(ddc_open_display) No EDID for device on bus /dev/i2c-2
Invalid display
   I2C bus:             /dev/i2c-2
   Supports DDC:        true
      I2C address 0x30 (EDID block#)  present: false
      I2C address 0x37 (DDC)          present: true 
      I2C address 0x50 (EDID)         present: true 
      /sys/bus/i2c/devices/i2c-2/name: NVIDIA i2c adapter 4 at 1:00.0
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?

I have attached the outputs you requested in similar cases like [#14](https://github.com/rockowitz/ddcutil/issues/14), and would greatly appreciate pointers toward fixing the problem. Thank you!

[ddcutil detect --verbose --ddc --force-slave-address --stats.txt](https://github.com/rockowitz/ddcutil/files/887597/ddcutil.detect.--verbose.--ddc.--force-slave-address.--stats.txt)
[ddcutil environment --verbose --force-slave-address --stats.txt](https://github.com/rockowitz/ddcutil/files/887598/ddcutil.environment.--verbose.--force-slave-address.--stats.txt)

rockowitz, 2017-04-02:
First, some observations:

- You have 2 video cards, and GTX750Ti running vfio-pci (presumably for
VMs) and a GTX 1080 running the Nvidia proprietary driver.

- You're at least trying to enable the modesetting features of the
Nvidia driver (nvidia-drm.modeset=1).  (I say trying because, if you
look at the end of the environment output, the API for accessing the
modesetting features features is not working.)

- xorg.conf includes the special settings that needed for I2C to work on
at least some Nvidia cards.

- The messages about I2C address x30 not being present is informational
only.  It's the mechanism by which secondary 128 byte EDID blocks (if
any) are read.  ddcutil uses only the first EDID block.

My best guess is that the problem is lack of support for I2C on the
latest Nvidia cards in the Nvidia driver.   To confirm this would entail
some invasive changes, I'm afraid.

- try putting the monitors on the older 750Ti
- try running with just one video card
- try replacing the Nvidia driver with nouveau

Sanford

krugman53, 2017-04-03:
Having learned about this wonderful technology just recently, I'm relieved to hear my setup wasn't an obvious showstopper. I saw threads about similar problems on the Nvidia dev forums. I don't think I can take apart my setup for the testing right now, so I'll take my chances with the proprietary driver being the culprit. I can chime in if an update eventually fixes the issue.

Now, after some further testing, I was pleasantly shocked to find an interesting workaround: I was able to achieve my desired functionality through a hackintosh VM on my (indeed vfio-pci-separated) GTX 750 Ti, using ddcctl, which I can of course script from Linux. What an amazingly flexible technology this is...

Thank you again for taking the time to look into this. If my setup can provide you with some (not super invasive) testing information, please let me know.

closed 2017-04-07 by krugman53