rockowitz/ddcutil #18: DisplayLink DL-4xxx support

SicVolo, 2017-04-03:
ddcutil is not finding USB3 DL monitor, even though the monitor is working fine on Ubuntu 16.04. According to the [official note](http://support.displaylink.com/knowledgebase/articles/1096210-usb-monitor-brightness-and-contrast-control-via-sc) DDC/CI is supported, and works with ScreenBright. 
usbenv's only relevant output is
"(possible_monitor_config_descriptor) Examining only interface 0 for device 2:1, vid=0x17e9, pid=0x4301  DisplayLink (null)", but nothing else

When running ddcutil with the --usb 2.1 I get "Unable to find monitor with the specified USB bus and device numbers"

--trace all is no help

Where's the culprit? How do troubleshoot it further?

rockowitz, 2017-04-04:
Sicilian,

What is the manufacturer and model of the monitor?  What version of
DisplayLink?

ddcutil supports monitors that communicate the Monitor Control Command
Set (MCCS) in one of 2 different ways:
- Using the DDC/CI protocol over an I2C bus connection to the monitor. 
On Linux, the video driver normally exposes the I2C bus as devices
/dev/ic2-*.
- The monitor uses USB to communicate MCCS, as per the USB Monitor
Control Class Specification
(http://www.usb.org/developers/hidpage/usbmon10.pdf).

Since DisplayLInk claims to support DDC/CI,  it would appears that the
DisplayLink video driver, at least on Windows, emulates the I2C bus.  
The question is whether the Ubuntu driver provides this support.   Just
because DisplayLink runs over USB does not mean that there is a USB MCCS
compliant interface.

Please run the following commands and send the output (as attachments,
not inline):

ddcutil detect --verbose
ddcutil environment --verbose
ddcutil usbenvironment --verbose

Sanford

SicVolo, 2017-04-04:
I2C is not connected and there is no MCCS. It's the difference between the Windows and the Linux driver, thanks for the hint. I am stuck until DL gets their act together.

closed 2017-04-04 by SicVolo