rockowitz/ddcutil #20: ddcutil always coredumps

GoodMirek, 2017-04-28:
I run Fedora 25 64bit, with latest updates, on Dell Inspiron N5110, CPU: Intel i3-2350M, with integrated Intel graphics.
```
uname -a
Linux localhost.localdomain 4.10.11-200.fc25.x86_64 #1 SMP Tue Apr 18 17:25:01 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```
Monitor attached via HDMI is LG 29UB67 Ultrawide. Internal laptop monitor is disabled via Gnome settings.

Whatever command I try, I get `Segmentation fault (core dumped)`.
I run ddcutil as root. Commands I tried:
```
ddcutil detect
ddcutil  --force-slave-address detect
ddcutil environment
ddcutil --force environment
```

I have installed ddcutil with commands:
```
dnf config-manager --add-repo http://download.opensuse.org/repositories/home:rockowitz/Fedora_25/home:rockowitz.repo
dnf install ddcutil
```
which resulted in successful installation of following packages:
```
i2c-tools-3.1.0-17.fc25.x86_64.rpm
ddcutil-0.7.4-1.1.fc25.x86_64.rpm
```

i2c_dev kernel module is loaded, /dev/i2c-n devices are present.
```
lsmod | grep i2c
i2c_dev                16384  0
i2c_i801               24576  0
i2c_algo_bit           16384  1 i915

ls -l /dev/i2c-*
crw-------. 1 root root 89, 0 Apr 28 14:48 /dev/i2c-0
crw-------. 1 root root 89, 1 Apr 28 14:48 /dev/i2c-1
crw-------. 1 root root 89, 2 Apr 28 14:48 /dev/i2c-2
crw-------. 1 root root 89, 3 Apr 28 14:48 /dev/i2c-3
crw-------. 1 root root 89, 4 Apr 28 14:48 /dev/i2c-4
crw-------. 1 root root 89, 5 Apr 28 14:48 /dev/i2c-5
crw-------. 1 root root 89, 6 Apr 28 14:48 /dev/i2c-6
crw-------. 1 root root 89, 7 Apr 28 14:48 /dev/i2c-7
```
Do you have any idea how to troubleshoot?
Will coredump help you?

GoodMirek, 2017-04-28:
I tried also to switch selinux off, no change.

rockowitz, 2017-04-28:
There are 2 projects at opensuse, ddcutil and ddcutil-test.   Somehow
the test files got into the production version.  (Production is v0.7.3,
the current test version is 0.7.4, which you installed.)   There was a
segfault problem in the development version yesterday, which is probably
what you hit.   Will get back to you once the problem is resolved. 
Thanks for reporting it.

Sanford

rockowitz, 2017-04-28:
The packages at https://software.opensuse.org/package/ddcutil have been
corrected to be the production version 0.7.3.  Not sure how it happened,
but I apologize for letting the development version leak out. 

Sanford

GoodMirek, 2017-04-28:
Thanks for quick response. I have installed 0.7.3 and got much farther. However, while trying to set brightness, `ddcutil` again crashes.
I have captured the output to the attached file. If I do not add command line options `--force-slave-address -f --verify` it makes no difference, the result is the same. Disabling selinux makes no difference.
[ddcutil-error.txt](https://github.com/rockowitz/ddcutil/files/965668/ddcutil-error.txt)

rockowitz, 2017-04-29:
The versioning issue is getting strange.   The bug you hit was added to
the code base on 3/9 (after the 0.7.3 release) to plug a minor memory
leak, and fixed on 4/14.  The  --verify option was added on 4/1.  So
with a code base before 4/1 you'd have gotten a parse error for
--verify, after 4/14 the bug is fixed.  Looks like something odd
happened when I tried to fix up project ddcutil on OBS earlier today. 

Here's the workaround.

The bug is triggered by using the 0X prefix on a feature code.   Feature
codes are always referred to in VCP documentation by their hex values,
not their decimal values, so using the 0x prefix on feature codes is
optional in ddcutil.  Try the following command:

ddcutil -setvcp 10 20

Notes on the command options you've used.  They're unnecessary.

--verify:   read VCP value after setting it, intended to address a rare
condition where update seems to work but value not
             changed in the monitor.   Will be the default in 0.7.4.

-force-slave-address:  addresses a rare condition of contention on the
I2C bus, not normally needed

-f (--force):  Ignores internal errors and forges ahead.  Likely to
result in a crash. Not for normal use

--display 1:    The default

Sanford

GoodMirek, 2017-04-29:
Thanks again for quick help!
Omitting 0x indeed helped. It is much more convenient to set backlight via keyboard than using LG's joystick to change settings buried deep in the OSD menu.

Btw maybe you could find following patch proposal interesting:
https://patchwork.kernel.org/patch/9530565/
If that ever gets implemented, it will bring convenience to yet another level.

closed 2017-04-29 by GoodMirek