rockowitz/ddcutil #21: segfaults

mixmastamyk, 2017-05-05:
Looks like it is crashing when it can't access some device files:

```
⏵ apt-cache policy ddcutil
ddcutil:
  Installed: 0.8.0-1

⏵ ddcutil detect --verbose > ddc_detect.txt
Failure getting EDID for /dev/i2c-6: status code=EIO(-5): Input/output error
ddcutil: ddc/ddc_displays.c:401: ddc_report_display_by_dref: Assertion `dref->flags & DREF_DDC_COMMUNICATION_CHECKED' failed.
fish: “ddcutil detect --verbose > ddc_…” terminated by signal SIGABRT (Abort)

⏵ ddcutil environment --verbose > ddc_env.txt
Failure getting EDID for /dev/i2c-6: status code=EIO(-5): Input/output error

⏵ ddcutil usbenvironment --verbose  > ddc_usb.txt
Failure getting EDID for /dev/i2c-6: status code=EIO(-5): Input/output error
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/002: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
fish: “ddcutil usbenvironment --verbos…” terminated by signal SIGSEGV (Address boundary error)

⏵ sudo ddcutil usbenvironment --verbose  > ddc_usb.txt
Failure getting EDID for /dev/i2c-6: status code=EIO(-5): Input/output error

```

mixmastamyk, 2017-05-05:
[ddc_interrogate_valgrind_output.txt](https://github.com/rockowitz/ddcutil/files/979751/ddc_interrogate_valgrind_output.txt)

Got farther when I ran with sudo:

[ddc_interrogate_sudo.txt](https://github.com/rockowitz/ddcutil/files/979756/ddc_interrogate_sudo.txt)

rockowitz, 2017-05-05:
Can you send me file ddc_interrogate.txt, the file that was written to
in log file ddc_interrogate_valgrind_output.txt?

Alternatively, run "ddcutil usbenv -v" in a terminal without redirecting
the output, then screen scrape the last couple hundred lines or so -
enough to include the normal output before the failure as well as the
valgrind output. 

Thanks,
Sanford

mixmastamyk, 2017-05-05:
I think it got clobbered, here is a new one:
[ddc_interrogate.txt](https://github.com/rockowitz/ddcutil/files/980119/ddc_interrogate.txt)

rockowitz, 2017-05-05:
Does the ddc_interrogate.txt file you sent yesterday correspond to the
ddc_interrogate_valgind_output_txt file you sent today?  If so, no need
to resend.  But tthe screen scrape I requested in the previous message
would still be helpful.   Thanks.

Sanford

mixmastamyk, 2017-05-05:
No, I redid them:

[ddc_valgrind_output.txt](https://github.com/rockowitz/ddcutil/files/980147/ddc_valgrind_output.txt)
[ddc_interrogate.txt](https://github.com/rockowitz/ddcutil/files/980148/ddc_interrogate.txt)

mixmastamyk, 2017-05-05:
[ddc_usbenv.txt](https://github.com/rockowitz/ddcutil/files/980155/ddc_usbenv.txt)

rockowitz, 2017-05-06:
Thanks.  That file pinpointed the problem.  The segfault occurs when an
unexpected condition occurs forming a message in the error recovery code
executed when open() fails for a /dev/usb/hiddev device that might
implement the USB Monitor Control Class Spec.  Why might a failure in
open() occur?  Probably because of permissions.   A fix for for this
situation, which can occur in the mainline code for USB connected
monitors as well as the diagnostic code, will go into the next point
release. 

If you're building from the git source, the best workaround is to build
ddcutil without its USB support using configure option --disable-usb

Alternatively, only run interrogate as root.

Regards,
Sanford

rockowitz, 2017-05-07:
This segfault is fixed in Release 0.8.1. 

It might occur when creating an extended error message after **open()** fails for a **/dev/usb/hiddev** device.

closed 2017-05-07 by rockowitz