rockowitz/ddcutil #22: ddcutil cores - dref->flags & DREF_DDC_COMMUNICATION_CHECKED failed

dangill14, 2017-05-24:
We are working on an ubuntu 16.04 system with two monitors attached; one is a standard Dell, the other is a touchscreen with custom EDID.  The touchscreen is flashed and EDID is programmed successfully.  

When executing 'ddcutil detect', we get an abort / core for our touchscreen (/dev/i2c-4).

Attached debug output.  

Any help is appreciated.

Thanks,

Dan.

[ddcutil_detect.txt](https://github.com/rockowitz/ddcutil/files/1027456/ddcutil_detect.txt)
[ddcutil_environment.txt](https://github.com/rockowitz/ddcutil/files/1027457/ddcutil_environment.txt)
[misc.txt](https://github.com/rockowitz/ddcutil/files/1027458/misc.txt)

rockowitz, 2017-05-25:
Dan,

Thank you for the detailed report.   There are several things going on
here.   Assuming that you're building from source, I'm including some
changes you can make.

1) I2C bus address x37 is not active for /dev/i2c-4, the touchscreen
device, so even if there were no other problem, ddcutil would regard it
as an invalid display.

2) In the diagnostic routines, ddcutil gets the EDID for the touchscreen
using both X11 and drm services.  In those cases it parses the EDID
successfully.  So the EDID would appear valid.  In the mainline code
ddcutil reads the EDID directly off I2C bus address x50.  I suspect this
is the point of failure.   In file src/i2c/i2c_bus.core.c at line 506,
change "debug = false" to "debug = true" to turn on diagnostic output. 
You can also try changing "conservative = true" to "conservative =
false" at line 509.   Alternatively, you can turn on diagnostic output
by executing ddcutil with option "--trace i2c", but you'll have to pick
through an overwhelming amount of output.   Whichever way you do it,
please send me a copy of the output.

3) The segfault is happening at a point in the code that checks that bus
address x50 (the EDID address) is active, but fails to check that a
valid EDID was read.  To fix this, at line 917 in function
ddc_detect_all_displays() in file src/ddc/ddc_displays.c, change the if
test:

  if ( (businfo->flags & I2C_BUS_ADDR_0X50) ) {

to:

  if ( (businfo->flags & I2C_BUS_ADDR_0X50) && dref->pedid ) {

This fix will go into the next point release.

4) The final item is more a curiosity than a problem, as it's not
tripping up ddcutil.  There are I2C devices whose names don't conform to
the /dev/i2c-N format, e.g. the /sys file system shows:

   /sys/bus/i2c/devices/i2c-INTCF1C:00/name: INTCF1C:00

Do these devices show up  among the /dev/i2c* devices?

rockowitz, 2017-07-22:
Fix applied to file ddc_displays.c in release 0,8.4

closed 2017-07-22 by rockowitz