rockowitz/ddcutil #23: relative input for setvcp

diggit, 2017-05-29:
Hi,
I want to set speaker volume of my LCD monitor from PC, but setvcp accepts absolute volume only. Would it be possible to implement relative value input? Like `setvcp -10`. ddcutil would in such case read current value from device a write modified value back.

One more question, is it possible to somehow speedup ddcutil?
`ddcutil --noverify --nodetect --bus=9 setvcp 0x62 [volume]`
Takes 450ms and that is still too much.

rockowitz, 2017-05-30:
Patrik,

Specifying a relative value on setvcp is an interesting idea.  It would 
apply only to simple continuous values.  I'll give it some more thought.

Re performance, most of ddcutil's elapsed time is spent in sleep's 
mandated by the DDC protocol.  The problem is compounded if DDC retries 
occur.  To see how ddcutil is spending its time, and the protocol errors 
encountered, you can add the --stats parm.  If you'd like me to look 
further, please send the output of your setvcp command with the --stats 
option, and also the output of ddcutil interrogate run as root. Note 
that the latter takes a long time to run.   Please send the files as an 
attachment.

Sanford

diggit, 2017-06-04:
Yeah, you are right. Relative values would be useful only for few continuous properties.

From stats, most of the time, ddcutil is waiting. If it means, that relative value requiring read and write would take twice as much, it probably does not worth integrating to ddcutil. I'd rather use some local file to store last set value and then use it for relative to absolute calculation.

Here are both logs.
[interrogate.txt](https://github.com/rockowitz/ddcutil/files/1050492/interrogate.txt)
[setvcp_vol20.txt](https://github.com/rockowitz/ddcutil/files/1050493/setvcp_vol20.txt)

rockowitz, 2017-06-04:
Hi Patrik,

As you can see, ddcutil's execution time is dominated by sleeps mandated 
by the DDC spec to give the monitor time to respond. There is an 
undocumented (and subject to change) parm that allows users to play with 
the sleep time.   I don't have the code in front of me, but IIRC "-y 1" 
doubles the mandated sleep time, and "-y 2" halves it.  You might get 
lucky with shorter sleeps.  Let me know if it works for you.

Sandy

diggit, 2017-06-04:
Hi Sandy,
thanks you for all ideas!

With `-y 1` I can get down to 360ms. That is fastest option. It should be ok with custom script.

Do you have any suggestion how to use ddcutil without root? It is inconvenient  to use sudo to set volume on my monitor.

rockowitz, 2017-06-05:
>
> Hi Sandy,
> thanks you for all ideas!
>
> With |-y 1| I can get down to 360ms. That is fastest option. It should 
> be ok with custom script.
>
> Do you have any suggestion how to use ddcutil without root? It is 
> inconvenient to use sudo to set volume on my monitor.

See http://www.ddcutil.com/i2c_permissions.  If i2c-tools has not 
created group i2c, you should do that. The bottom line is that your user 
needs RW permission to the /dev/i2c-n devices. (Note: just caught a typo 
on the page, at one point "udev" is misspelled "udef".)

closed 2017-07-22 by rockowitz