rockowitz/ddcutil #25: Useful return codes

tobimensch, 2017-06-23:
Hello,

so you want feedback. It's a great little tool and I thank you for it.

I have one issue though.

I want ddcutil to retry doing stuff until it reaches success.

Unfortunately ddcutil doesn't exit with return code 0 when it
does something (such as lowering brightness) with success.

Only way I've found to is to use --verify option and to catch stderr
output. If there is output I assume that something failed, if there's no
output I assume it worked.

However when using this logic in some bash code, you get something
like this:

function my_dim_screen
{
  RET=$(ddcutil -d $1 setvcp 10 $2 --maxtries "15,15,15" --verify 2>&1)
  [[ "${#DDRET}" -eq 0 ]]
}

my_dim_screen returns the return codes I would expect from a well behaving program.

But my solution is bad for two reasons:
1) I have to use a subshell which is less efficient
2) ugliness

I hope you can fix this.

rockowitz, 2017-06-24:
Tobi,

Thanks for your input.  Your point is well taken.  I am traveling and 
will consider the question more fully when I return home in a couple weeks.

Sanford

rockowitz, 2017-07-07:
Tobi,

I apologize for taking so long to reply to your email.  I was traveling 
and only urgent ddcutil matters were being attended to.

I've reviewed the code base and performed some tests.  Things should be 
working as you would like.   The command line version of ddcutil has 2 
status code values that are returned to the shell: 0 for success or 1 
for failure. This is confirmed by checking shell variable "$?".   There 
are a lot of grey areas as to whether a command has succeeded, but for 
setvcp it is pretty clear.  If setting the value for any feature fails, 
the entire command fails.  (It is possible to specify multiple feature 
id/values on a single line.)   The one wrinkle for setvcp is that the 
DDC/CI protocol for setting a feature is that the monitor does not reply 
whether it has successfully executed the command.   The only way to know 
is by reading the value after writing it, which is what --verify does.  
So unless you use --verify, it is possible that setvcp has no effect but 
ddcutil does know there has been a failure.

If ddcutil is not returning 0 for success and 1 for failure, I'd very 
much appreciate an example.

Regards.
Sanford

closed 2017-07-22 by rockowitz