rockowitz/ddcutil #26: Doesn't work anymore after switching to laptop

tobimensch, 2017-07-14:
I used a desktop system until today, and now I switched to a laptop
as a desktop replacement.

The same monitor that used to work up until now over HDMI and which
I used to be able to control with ddcutil, is now connected over HDMI again,
however I can't control it through ddcutil anymore.

For example:

` ddcutil --sn YE8J023664 setvcp 10 100
ioctl error in function i2c_set_addr at line 214 in file i2c/i2c_bus_core.c: errno=EBUSY(16): Device or resource busy
(i2c_set_addr) addr = 0x50. Returning -16
Failure detecting bus addresses for /dev/i2c-5: status code=EBUSY(-16): Device or resource busy
ioctl error in function i2c_set_addr at line 214 in file i2c/i2c_bus_core.c: errno=EBUSY(16): Device or resource busy
(i2c_set_addr) addr = 0x50. Returning -16
Failure detecting bus addresses for /dev/i2c-18: status code=EBUSY(-16): Device or resource busy
Display not found
`
The other monitor is connected with VGA and can be controlled with ddcutil perfectly fine.

tobimensch, 2017-07-14:
After some more digging I found out with dmesg that the nouveau driver is spitting out error messages such as this one:

> [ 3345.132682] nouveau 0000:01:00.0: i2c: aux 000a: begin idle timeout ffffffff
[ 3345.133788] nouveau 0000:01:00.0: i2c: aux 000b: begin idle timeout ffffffff
[ 3345.134904] nouveau 0000:01:00.0: i2c: aux 000b: begin idle timeout ffffffff
[ 3345.136050] nouveau 0000:01:00.0: i2c: aux 000b: begin idle timeout ffffffff
[ 3345.137179] nouveau 0000:01:00.0: i2c: aux 000c: begin idle timeout ffffffff
[ 3345.138287] nouveau 0000:01:00.0: i2c: aux 000c: begin idle timeout ffffffff
[ 3345.139423] nouveau 0000:01:00.0: i2c: aux 000c: begin idle timeout ffffffff
[ 3345.140583] nouveau 0000:01:00.0: i2c: aux 000d: begin idle timeout ffffffff
[ 3345.141736] nouveau 0000:01:00.0: i2c: aux 000d: begin idle timeout ffffffff
[ 3345.142880] nouveau 0000:01:00.0: i2c: aux 000d: begin idle timeout ffffffff

So maybe this is a bug in nouveau, but I can't say for sure.

rockowitz, 2017-07-15:
Try using the "--force-slave-address" option.  This can sometimes fix 
EBUSY errors.  See the discussion in http://www.ddcutil.com/tuning.

Did you do a new installation on the laptop or did you just copy 
operating system from the desktop?   Most likely your video driver, 
which provides I2C, has changed.

If the problem persists, please run the following commands and submit 
the output, as attachments:

   # ddcutil detect --verbose
   # ddcutil environment --verbose

Regards,
Sanford

closed 2018-05-20 by rockowitz