rockowitz/ddcutil #27: I2C workaround Nvidia drivers for dual monitors?

AndreasBackx, 2017-07-15:
Previously I was using ddcutil without any problems on AMD hardware. Because of the mining craze I got a good deal on my used AMD hardware and a good deal on a GTX 1070. Now unfortunately I cannot get ddcutil to work on one of my monitors. It seems to only detect 1 monitor, but the environment says there is a second one, it just cannot find the udev information. I've got 2 Dell U2515H connected (1 landscape over HDMI and 1 portrait over DP), [here is some more information of the following commands](https://gist.github.com/AndreasBackx/aface790b59f3dd5b3c143fa0e1244ee).

```shell
sudo ddcutil detect --verbose
```

```shell
sudo ddcutil environment --verbose
```

```shell
sudo ddcutil interrogate --verbose
```

```shell
ddcutil interrogate --verbose
```

I'm not the greatest when it comes to X11 config, but here is my config regardless.

_/etc/X11/xorg.conf_

```
Section "ServerLayout"
    Identifier     "Layout0"
    Screen      0  "Screen0"
    InputDevice    "Keyboard0" "CoreKeyboard"
    InputDevice    "Mouse0" "CorePointer"
EndSection

Section "Files"
EndSection

Section "InputDevice"
    # generated from default
    Identifier     "Mouse0"
    Driver         "mouse"
    Option         "Protocol" "auto"
    Option         "Device" "/dev/psaux"
    Option         "Emulate3Buttons" "no"
    Option         "ZAxisMapping" "4 5"
EndSection

Section "InputDevice"
    # generated from default
    Identifier     "Keyboard0"
    Driver         "kbd"
EndSection

Section "Monitor"
    Identifier     "Monitor0"
    HorizSync       28.0 - 33.0
    VertRefresh     43.0 - 72.0
    Option         "DPMS"
EndSection

Section "Monitor"
    Identifier     "Monitor1"
    HorizSync       28.0 - 33.0
    VertRefresh     43.0 - 72.0
    Option         "DPMS"
EndSection

Section "Device"
    Identifier     "Device0"
    Driver         "nvidia"
    VendorName     "NVIDIA Corporation"
    Option	   "RegistryDwords"  "RMUseSwI2c=0x01; RMI2cSpeed=100"
EndSection

Section "Device"
    Identifier     "Device1"
    Driver         "nvidia"
    VendorName     "NVIDIA Corporation"
    Option	   "RegistryDwords"  "RMUseSwI2c=0x01; RMI2cSpeed=100"
EndSection

Section "Screen"
    Identifier     "Screen0"
    Device         "Device0"
    Monitor        "Monitor0"
    DefaultDepth   24
    Option         "DPI" "96 x 96"
    Option	   "metamodes" "nvidia-auto-select +0+0 { ForceFullCompositionPipeline = On }"
    SubSection     "Display"
        Depth       24
    EndSubSection
EndSection

Section "Screen"
    Identifier     "Screen1"
    Device         "Device1"
    Monitor        "Monitor1"
    DefaultDepth   24
    Option         "DPI" "96 x 96"
    Option 	   "metamodes" "nvidia-auto-select +0+0 { ForceFullCompositionPipeline = On }"
    SubSection     "Display"
        Depth       24
    EndSubSection
EndSection
```

rockowitz, 2017-07-17:
Andreas,

First, some lesser comments:
1) interrogate just packages up detect, environment, usbenvironment, and 
probe of each detected monitor, all with --verbose and a few other 
tweaks.  So for diagnosing your problem detect and environment give us 
all the relevant information.
2) The failure at the end of the "sudo ddcutil interrogate --verbose" 
report is due to an unexpected library return code in the usbenvironment 
section of interrogate.  Since you don't have a USB connected monitor, 
this is unrelated to your problem.  The output of detect and environment 
is all that matters.
3) Looking at the output of the environment portion of the interrogate 
run (without sudo), we see that user andreas does not have RW permission 
on the /dev/ii2c devices.  You'll need to fix this if you want to run 
ddcutil other than as root.  This is just a configuration issue.
4) You mention that you used to run ddcutil with AMD hardware. Were you 
using the proprietary AMD driver or the open source radeon driver?  I'm 
particularly interested if it's the former case.
5) In the future, please send send output reports as multiple 
attachments, with the output of 1 command per file.  It's more 
convenient for me
6) Re X11 configuration, I too find it painful.  I've had good success 
not specifying any xorg.conf file, letting X11 figure things out. 
Overriding values, such as the special RegistryDwords settings, are in 
files in /etc/X11/xorg.conf.d.  I've used the proprietary Nvidia driver 
on a GTX650ti with 4 monitors this way.
7) The special RegistryDwords settings may be necessary for successfully 
reading and writing DDC commands on I2C slave address x37.  However, 
they have no effect on whether that address is detected, only on whether 
reads and writes are not corrupted.

With all that out of the way, we're down to the real problem: why isn't 
the proprietary Nvidia driver allocating a /dev/i2c device for the 
second monitor.  I'm afraid I haven't a clear answer.  The nvidia I2C 
support for recent cards has been problematic.   Here are a couple 
things you can try:
- Use the DVI connection instead of DisplayPort.  You'll probably need a 
dual-link DVI cable.  Or use a second HDMI connection. With DisplayPort, 
I2C signals are transmitted over the AUX channel instead of dedicated 
pins.  (This is referred to as I2C-Over-AUX.)  While DisplayPort has 
always worked for me, user reports suggest there can be problems.
- Try eliminating file /etc/X11/xorg.conf, and let X11 figure out the 
configuration.

Let me know how it goes.

Sanford

AndreasBackx, 2017-07-19:
@rockowitz I was working using AMD's open source drivers. I will simply upload the files next time. When I remove the `xorg.conf` file and only use `xorg.conf.d/90-nvidia_i2c.conf` file seen below, then the second display also does not show up.

I do not have a DVI input available unfortunately and cannot test that proposed solution.

```
Section "Device"
    Identifier     "Dev0"
    Driver         "nvidia"
    VendorName     "NVIDIA Corporation"
    Option	   "RegistryDwords"  "RMUseSwI2c=0x01; RMI2cSpeed=100"
EndSection

Section "Screen"
    Identifier     "Screen0"
    Device         "Dev0"
    Monitor        "Monitor0"
    DefaultDepth   24
    Option         "DPI" "96 x 96"
    Option	   "metamodes" "nvidia-auto-select +0+0 { ForceFullCompositionPipeline = On }"
    SubSection     "Display"
        Depth       24
    EndSubSection
EndSection
```

rockowitz, 2017-07-19:
Andreas,

Thanks for the update.

I thought that all GTX1070s had a DVI connector.  For our purposes a 
second HDMI connector works just as well.  The U2515h implements 
DisplayPort MST (Multi-Stream-Technology).  I know there are problems 
with DisplayPort I2C-Over-Aux and the i915 driver related to MST (see: 
https://bugs.freedesktop.org/show_bug.cgi?id=100954), which is why I 
suggest you try connecting the monitors without using DisplayPort.  
Other than that, I'm afraid I don't have any suggestions.

Sanford

AndreasBackx, 2017-07-19:
The GTX 1070 does hav a DVI connector, but my monitors do not. My GTX 1070 also only has 1 HDMI output which is already connected to the monitor. I just tried to disable DP 1.2 on the monitor as that is required for MST, but it doesn't seem to detect the monitor still. MST is not being used btw, both monitors are connected with separate cables, one with HDMI and the other with DP so MST is not being used.

anupam-git, 2018-04-29:
I was facing issues with Nvidia driver for my single monitor setup using HDMI with GT-1030 Pascal Series GPU.

This helped me fix the issue : http://www.ddcutil.com/nvidia/

You can try this once..

AndreasBackx, 2018-04-29:
@anupam-git that does not work for DisplayPort. I'm just going to get a second HDMI cable and hope it's fixed.

toshi2k2, 2021-08-04:
I have a similar problem. ```ddcutil detect``` fails to detect second monitor which is connected by HDMI. It only detects the primary monitor connected via displayport. (The second monitor ddc input switch works from my Mac though)
```
ddcutil version:            1.1.0
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  true
   Dynamic sleep adjustment enabled:            false

Configuration file:         (none)
Display 1
   I2C bus:  /dev/i2c-6
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-6/name:   NVIDIA i2c adapter 6 at 2d:00.0
   EDID synopsis:
      Mfg id:               PHL
      Model:                PHL 276E8V
      Product code:         49551
      Serial number:        
      Binary serial number: 1919 (0x0000077f)
      Manufacture year:     2020,  Week: 27
   VCP version:         2.2
   Controller mfg:      Mstar
   Firmware version:    1.0
   Monitor returns DDC Null Response for unsupported features: false

```

closed 2025-02-09 by rockowitz