rockowitz/ddcutil #29: Feature request : incrementing value

lord-re, 2017-08-03:
I have a quite simple usage of ddcutil. I use it only to change the brightness of my monitor.
Being able to increment/decrement brightness could be quite handy :
Something like "ddcutil setvcp 10 +10" would read the current value and add 10 to that value.

rockowitz, 2017-08-03:
Thank you for the suggestion.  You're not the first to request such 
feature; your suggestion gives the feature request more weight.   The 
behavior would apply only to continuous VCP features, which makes the 
implementation not quite as simple as it might otherwise seem.  I do 
wonder about the syntax.  Documenting that "10 is an absolute value, but 
+10 a relative one, seems a bit arcane.

Sanford

rockowitz, 2017-08-03:
Should have added to my reply that comments from others on the 
desirability of this feature and its syntax would be most welcome.

Sanford

diggit, 2017-08-04:
Some discussion was here #23. I abandoned that feature request, because write operation takes on my HW about 360ms and reading out value to modify would make it double.

Volume control on my monitor does not have linear step so i switched to solution without feedback from monitor itself.

w-biggs, 2018-02-15:
I'd love this a lot too - it'd enable me to use keyboard shortcuts to increase/decrease my monitor's brightness, which would be really nice to be able to do.

rockowitz, 2018-02-15:
Wilson,

Thanks for the input, with a use case.

I remain hesitant about the ambiguity of "setvcp 30" specifying an 
absolute value, but "setvcp +30" specifying a relative value. But I'll 
continue to ponder it.   It may be that specifying "+30" is so strange 
that the only people who would do it would know that it indicates a 
relative value.

Sanford

rockowitz, 2018-02-25:
Ok guys, you've won me over (or worn me down).  My current development 
version implements relative setvcp values for Continuous VCP features.

The command line syntax is less than optimal.  I'm not thrilled about 
documenting it.  Feedback and/or alternatives appreciated.

Examples:

    setvcp 10 + 5
    setvcp 10 -  5

Note the space after the + or -.   This is necessary because ddcutil's 
command line parser, implemented using glib's parser, recognizes "-5" as 
unrecognized option "-5".

Sanford

rockowitz, 2018-05-13:
Release 0.9.0 implements relative values on setvcp, but only for standard continuous features, i.e. not those with a reserved subrange of NC values. See the release announcement at [www.ddcutil.com](http://www.ddcutil.com).  Note the need to surround "+" or "-" with blanks, to indicate a relative rather than absolute value.   I'm not mad-keen on the syntax, but it's the best I could come with.

closed 2018-05-19 by rockowitz