rockowitz/ddcutil #31: Feature Request: sys class backlight driver

r0b0, 2017-08-21:
Apparently, a lot of software uses /sys/class/backlight/*/actual_brightness to try to control screen brightness. It would be cool if it was possible to control DDC enabled monitors using this interface, too.

diggit, 2017-08-21:
Hi, /sys/... is domain of kernel, quick googling revealed some [existing patch](https://patchwork.kernel.org/patch/9530565/). Not sure what stage it is in.

rockowitz, 2017-10-18:
Just notice that my email reply on 8/23 does not appear in the github 
wiki, so I'm resending, with an additional comment.

Robert,

At first glance, this seems like a natural integration, and Patrik has 
referenced a first stab at an implementation. However, there are 
significant issues, some of which are discussed in that thread 
(https://patchwork.kernel.org/patch/9530565/).

First, there's the question of proper layering. /sys/class/backlight is 
an ACPI interface, and its structure conforms to an ACPI interface.   It 
works with laptop displays, which have a fundamentally different 
interface from monitors implementing DDC/CI, which is an I2C interface.  
Is it appropriate to create a pseudo-ACPI interface for DDC/CI devices?  
My instinct says no, but I would defer to kernel developers.

Laptops displays are always flat panels, so there's a direct 
relationship between backlight level and brightness. DDC/CI can 
interface with a wide variety of monitors, whose "brightness" can 
reflect the settings of multiple features that vary from monitor to 
monitor, and also the underlying technology.  Shoehorning DDC/CI into 
the ACPI framework at the driver level would mean either ignoring all 
this complexity and just using the value of VCP feature x10 
(luminescence), or hard coding a complex set of rules that might or 
might not be appropriate for any given monitor.

DDC/CI is a slow and inherently unreliable protocol.  (ddcutil spends 
considerable effort managing retries.)   Is it appropriate to have such 
an interface in /sys?

What I think would be more appropriate is a utility that sits on top of 
/sys/class/backlight and DDC/CI to manage brightness in a standard 
way.   It could use the ddcutil API for DDC/CI access.  As a higher 
level utility, it would be appropriate to make heuristic decisions as to 
what VCP features to use to manage brightness on a particular monitor, 
perhaps with a configuration file to tell it how to manage more complex 
displays.

Sanford

Additional comment:  KDE Neon has experimental code that integrates 
/sys/class/backlight and ddcutil in the powerdevil power management 
control.

On 08/21/2017 07:09 PM, Patrik Bachan wrote:

closed 2018-05-19 by rockowitz