rockowitz/ddcutil #33: [gtx1300 / ubuntu 16] ddcutil can't find any monitors after possibly nvidia driver update

altV, 2017-10-03:
Hi Sanford.

I had this tool working on my Ubuntu for 1 month, and after restarting a computer it stopped seeing my monitor.

I tried installing nvidia-381, nvidia-378, and nvidia-384, restarting several times, and saying 'oh mighty computer', all of which didn't help. I remember I did something one time to fix it by accident (launching nvidia-settings? setting refresh rate by xrandr?), but I have no idea what should I try next.

Can you give an advice how would I approach to digging brightness control of my monitor through command line?
(I was trying to buy passive AMD graphics card, but there are none on the market for 4K monitors with no cooler where I live)

Patch for Nvidia cards is there copied to xorg.conf.d dir, but the ticket I opened and solved when it was a problem described in the man page did not have a message 'no monitors found', it was different.

> Setting output level verbose...
> Setting maximum retries...
> Forcing --stats...
> Forcing --force-slave-address..
> This command will take a while to run...
> 
> 
> *** Basic System Information ***
> 
> ddcutil version: 0.8.4
> 
> /proc/version:
>    Linux version 4.4.0-96-generic (buildd@lgw01-10) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #119-Ubuntu SMP Tue Sep 12 14:59:54 UTC 2017
> 
> /etc/os-release...
>    PRETTY_NAME="Ubuntu 16.04.3 LTS"
> 
> /proc/cmdline:
>    BOOT_IMAGE=/boot/vmlinuz-4.4.0-96-generic root=UUID=890c5053-2fab-497d-b889-c60e496b7562 ro quiet splash vt.handoff=7
> 
> Processor information as reported by lscpu:
>    Architecture:          x86_64
>    CPU op-mode(s):        32-bit, 64-bit
>    Byte Order:            Little Endian
>    CPU(s):                8
>    On-line CPU(s) list:   0-7
>    Thread(s) per core:    2
>    Core(s) per socket:    4
>    Socket(s):             1
>    NUMA node(s):          1
>    Vendor ID:             GenuineIntel
>    CPU family:            6
>    Model:                 26
>    Model name:            Intel(R) Core(TM) i7 CPU         920  @ 2.67GHz
>    Stepping:              5
>    CPU MHz:               2793.000
>    CPU max MHz:           2793,0000
>    CPU min MHz:           1596,0000
>    BogoMIPS:              5397.56
>    Virtualization:        VT-x
>    L1d cache:             32K
>    L1i cache:             32K
>    L2 cache:              256K
>    L3 cache:              8192K
>    NUMA node0 CPU(s):     0-7
>    Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 popcnt lahf_lm tpr_shadow vnmi flexpriority ept vpid dtherm ida
> 
> DMI Information from /sys/class/dmi/id:
>    Motherboard vendor:       Gigabyte Technology Co., Ltd.
>    Motherboard product name: EX58-UD5
>    System vendor:            Gigabyte Technology Co., Ltd.
>    System product name:      EX58-UD5
>    Chassis vendor:           Gigabyte Technology Co., Ltd.
>    Chassis type:             3 - Desktop
> 
> *** Primary Check 1: Identify video card and driver ***
> 
> Obtaining card and driver information from /sys...
> 
[... 564 lines cut ...]
>    IO error count:           0
>    Total sleep events:       0
> 
>    Sleep Event type      Count
>    SE_WRITE_TO_READ          0
>    SE_POST_OPEN              0
>    SE_POST_WRITE             0
>    SE_POST_READ              0
>    SE_DDC_NULL               0
>    SE_POST_SAVE_SETTINGS     0
> 
> Call Stats:
>    Type                                     Count Millisec  (   Nanosec)
>    Totals:                                      0        0  (         0)
> 
> Sleep Call Stats:
>    Total sleep calls:                                    2
>    Requested sleep time milliseconds :                 100
>    Actual sleep milliseconds (nanosec):                100  ( 100287151)
> 
> Elapsed milliseconds since last reset (nanoseconds):     0  (     17882)
> Total elapsed milliseconds (nanoseconds):              605  ( 605929535)
> Setting output level normal  Table features will be skipped...
> 
> Display scanning complete.

rockowitz, 2017-10-05:
Leonid,

I see you've closed the issue, but for future reference what's key in 
the diagnostic output is that i2c slave addresses x50 and x37 and not 
present on either of the 2 /dev/i2c-n devices.  It's also odd that there 
are only 2 - normally there are many.

I suggest you download and install i2c-tools, which provides command 
i2cdetect.  "ddcutil environment --verbose" will use i2cdetect if 
available.   More importantly, you can use i2cdetect to see if there are 
any i2c buses having both slave address x50 (EDID) and x37 (DDC) 
active.   If not, there's nothing that ddcutil can do.

Sanford

altV, 2017-10-07:
Thank you. Will do and reply back, but within 1-2 weeks.

Currently everything works with latest nvidia drivers, and HDMI cable. (So my best guess that problem was in bad nvidia update plus the whole system not working with displayport for some reason, because I remember I switched to displayport, and only then started to try different drivers.)

So what I am going to try is to switch back to displayport, and provide the log in case it stops working again.
(I presume brightness control (DDC communication?) should work regardless of cable I connect monitor with.)

rockowitz, 2017-10-07:
Again, I suggest that you use command i2cdetect from package 
i2c-tools.   Its a quick way to check if slave addresses x50 (EDID) and 
x37 (DDC) are present on an i2c bus.   Unless they are, ddcutil cannot 
communicate on the bus.  If DDC communication is working, then there's 
nothing special about VCP feature code x10 (brightness).   I've yet to 
see a monitor that doesn't support that feature code.

Sanford

altV, 2017-10-07:
(i2c-tools package installed)
just in case, I attach output of "ddcutil environment --verbose" for the current setup (HDMI cable, nvidia-387, where currently everything is working fine)

> sudo ddcutil environment --verbose
```
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Force I2C slave address:    false

The following tests probe the runtime environment using multiple overlapping methods.

*** Basic System Information ***

ddcutil version: 0.8.4

/proc/version:
   Linux version 4.4.0-96-generic (buildd@lgw01-10) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #119-Ubuntu SMP Tue Sep 12 14:59:54 UTC 2017

/etc/os-release...
   PRETTY_NAME="Ubuntu 16.04.3 LTS"

/proc/cmdline:
   BOOT_IMAGE=/boot/vmlinuz-4.4.0-96-generic root=UUID=890c5053-2fab-497d-b889-c60e496b7562 ro quiet splash vt.handoff=7

Processor information as reported by lscpu:
   Architecture:          x86_64
   CPU op-mode(s):        32-bit, 64-bit
   Byte Order:            Little Endian
   CPU(s):                8
   On-line CPU(s) list:   0-7
   Thread(s) per core:    2
   Core(s) per socket:    4
   Socket(s):             1
   NUMA node(s):          1
   Vendor ID:             GenuineIntel
   CPU family:            6
   Model:                 26
   Model name:            Intel(R) Core(TM) i7 CPU         920  @ 2.67GHz
   Stepping:              5
   CPU MHz:               1862.000
   CPU max MHz:           2793,0000
   CPU min MHz:           1596,0000
   BogoMIPS:              5397.20
   Virtualization:        VT-x
   L1d cache:             32K
   L1i cache:             32K
   L2 cache:              256K
   L3 cache:              8192K
   NUMA node0 CPU(s):     0-7
   Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 popcnt lahf_lm tpr_shadow vnmi flexpriority ept vpid dtherm ida

DMI Information from /sys/class/dmi/id:
   Motherboard vendor:       Gigabyte Technology Co., Ltd.
   Motherboard product name: EX58-UD5
   System vendor:            Gigabyte Technology Co., Ltd.
   System product name:      EX58-UD5
   Chassis vendor:           Gigabyte Technology Co., Ltd.
   Chassis type:             3 - Desktop

*** Primary Check 1: Identify video card and driver ***

Obtaining card and driver information from /sys...

Determining driver name and possibly version...
   Driver name:    nvidia
   Driver version: 387.12

Video card identification:
   Vendor:              10de       NVIDIA Corporation
[... 559 lines cut ...]
      [223624.955] (--) NVIDIA(GPU-0): 
      [223770.351] (--) NVIDIA(GPU-0): Philips PHL BDM4037U (DFP-2): connected
      [223770.351] (--) NVIDIA(GPU-0): Philips PHL BDM4037U (DFP-2): Internal TMDS
      [223770.351] (--) NVIDIA(GPU-0): Philips PHL BDM4037U (DFP-2): 600.0 MHz maximum pixel clock
      [223770.351] (--) NVIDIA(GPU-0): 
      [223770.402] (--) NVIDIA(GPU-0): Philips PHL BDM4037U (DFP-2): connected
      [223770.402] (--) NVIDIA(GPU-0): Philips PHL BDM4037U (DFP-2): Internal TMDS
      [223770.402] (--) NVIDIA(GPU-0): Philips PHL BDM4037U (DFP-2): 600.0 MHz maximum pixel clock
      [223770.402] (--) NVIDIA(GPU-0): 
      [238067.162] (--) NVIDIA(GPU-0): Philips PHL BDM4037U (DFP-2): connected
      [238067.162] (--) NVIDIA(GPU-0): Philips PHL BDM4037U (DFP-2): Internal TMDS
      [238067.162] (--) NVIDIA(GPU-0): Philips PHL BDM4037U (DFP-2): 600.0 MHz maximum pixel clock

Not built with libdrm support.  Skipping DRM related checks

Device Identifier Cross Reference Report

   EDID: ...550A016C  Mfg: PHL  Model: PHL BDM4037U   SN: Unspecified  
      Bus:           /dev/i2c-1
      XrandR output: HDMI-0
      DRM connector: (null)
      UDEV name:     NVIDIA i2c adapter 5 at 4:00.0
      UDEV syspath:  /devices/pci0000:00/0000:00:07.0/0000:04:00.0/i2c-1/i2c-dev/i2c-1

```

altV, 2017-10-07:
```
❯ sudo i2cdetect 1
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1.
I will probe address range 0x03-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- 37 -- -- 3a -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- 4a 4b -- -- -- -- 
50: 50 -- -- -- 54 -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
                                                                                                                                                                                                                    
~ 10s
❯ sudo i2cdetect 0
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-0.
I will probe address range 0x03-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
                                
```

altV, 2017-10-07:
(Will attach same output of all the commands when I switch back to displayport and thus probably reproduce the issue.)

altV, 2017-11-27:
ok, I restarted computer today since then, and the issue came back.

I copied a file to  /etc/udev/rules.d, one binary to read usb light sensor to /usr/sbin and a systemd service wrapper, then restarted system again - and ddcutil was still working fine.

I did one more (third) restart - now it says again `Failure getting EDID for /dev/i2c-1: status code=DDCRC_EDID(-3016): invalid EDID`

output of `sudo i2cdetect 1` and `sudo i2cdetect` is the very same as in the comment above - 1 month ago (did diff comparison with vim)

@rockowitz, can you please offer an advice in further debugging?
From a blackbox point of view, I can only guess that my system has something non-deterministic in the load order such that sometimes ddcutil works, and sometimes doesn't - because copying files and running this tiny systemd service (and also deleting /dev/floppy0 from my /etc/fstab) was literally the only change between these three restarts, and there was no changes other than running google chrome between second restart (ddcutil worked with no errors) and the third one (ddcutil gives an error).

altV, 2017-11-27:
edit: actually, I do not see I have posted original error message of ddcutil in this ticket, which is not good, so I cannot compare current error message of ddcutil to something above.

Redoing steps from man on nvidia drivers now, and modprobe commands from above, because it looks like #30 and not like something that was in this ticket. Sorry for confusion

altV, 2017-11-27:
I just did one more restart, doing no changes whatsoever, and it works again. The workaround for nvidia in xorg.conf.d was there all the time, of course.

So it's something to do with the load order or something else. (Something something dark side.)

rockowitz, 2017-11-28:
Leonid,

My best guess is that it's an issue with the proprietary Nvidia driver, 
which we know has problems, particularly for newer cards. If you can, 
try switching to nouveau, at least temporarily, and see if the problem 
persists.

Status code DDCRC_EDID indicates that the bytes read for the EDID do not 
represent a well-formed EDID, probably because
the initial marker bytes or the checksum are invalid.

If you're running ddcutil 0.8.5, you can execute ddcutil with options:
    --trcfunc i2c_get_raw_edid_by_fd  --trcfunc i2c_get_parsed_edid_by_fd

This will give you a look a the EDID bytes and may give a hint as to the 
nature of the EDID corruption.

Sanford

anupam-git, 2018-04-29:
If you are still facing issues try this : http://www.ddcutil.com/nvidia/

This worked for me.

closed 2021-08-17 by rockowitz