rockowitz/ddcutil #34: Build without X11

macpijan, 2017-10-18:
On many embedded platforms X11 is not present anymore. It could be nice to allow disabling X11 support via configuration switch. What do you think about that? Do you think X11 is necessary to run this app? For a quick test I removed all X11 and xrand from code and configuration checks, and it seems to be working fine.

rockowitz, 2017-10-18:
Interesting idea.   IIRC, the X11 related code exists only in the 
environment and interrogate commands, to assist in remote diagnosis of 
user problems.   So it should be possible to build ddcutil without it.  
Can you describe in more detail the platforms you're interested in?   
Hardware, processor (e.g. ARM, MIPS), use cases?  How are the monitors 
connected?

Sanford

macpijan, 2017-10-18:
Yes, I could compile and run it without x11 libs. I have ARM board (i.MX) and exploring ways of controlling various display for digital signage. Ddc works great with standard computer monitors. Ddc seems to be present in some digital signage LCD displays as well. It doesn't seem to be present in standard TVs, though.

rockowitz, 2017-10-18:
Which Linux distribution are you running?

Interesting that DDC/CI works for some digital signage.   I've yet to 
see a standard TV for which it works.

Please try running command "ddcutil interrogate" and send the output.   
Given that "ddcutil interrogate" is a convenience wrapper around 
"ddcutil environment" and other commands, and that the code you've 
(presumably) disabled is in the environment command, I wouldn't be 
surprised if it just blows up, but even that would be interesting.   
Please send the output as an attachment.

Sanford

rockowitz, 2018-01-23:
configure option --disable-x11 was added in release 0.8.5

closed 2018-01-23 by rockowitz