rockowitz/ddcutil #35: trouble switching input source

ratmice, 2017-10-19:
I was have two computers hooked up to one monitor (one hdmi, one dvi)
which I can switch between using the physical buttons on the monitor to switch the 
input channel.

I didn't seem to find a way to do the same using ddcutil:

$ ddcutil capabilities
  Feature: 60 (Input Source)
    Values:
       01: VGA-1
       03: DVI-1
       04: DVI-2
trying to setvcp 60 1 it will switch to vga and back,
but switching between 3, and 4 doesn't appear to do anything (never leaves the current input source)

I ran ddcutil in the background switched input source with physical buttons
$ while true; do ./ddcutil getvcp 60; sleep 1; done
VCP code 0x60 (Input Source                  ): DVI-1 (sl=0x03)
Display not found
VCP code 0x60 (Input Source                  ): VGA-2 (sl=0x02)
Display not found
Display not found
<repeats until i switched back>
VCP code 0x60 (Input Source                  ): DVI-1 (sl=0x03)

What is mildly interesting to note is that the using the physical buttons it temporarily switched to
vga-2 before switching to hdmi input source (It shows this on screen as well),
but vga-2 doesn't show up in the ddcutil capabilities output above.  The values there mysteriously jump from 1 to 3.

Will try and get ddcutil running on the other hardware connected to the monitor and see what it does, let me know if there is anything else that would be helpful

rockowitz, 2017-10-19:
Several comments:

1) The monitor's capabilities string should be regarded as no more than 
suggestive.  It is often wrong.  The "ddcutil capabilities" command 
reports it, but other than that ddcutil does not rely on it.

2) To see the full list of values defined in the spec, use the command:

   ddcutil vcpinfo 60 --verbose

No monitor (that I know of) implements all the values, and there are 
some monitors with exotic connections that implement additional 
values.   Note that there are separate values (x11 and x12) for HDMI 
connections, though it is not unheard of for monitors to report an HDMI 
connection as a DVI connection because of their electrical similarity.

In your case, the capabilities string is declaring that connections 
VGA-1, DVI-1, and DVI-2 are present.

3) To see the actual values used by VCP feature x60, issue command 
"ddcutil getvcp 60" from the currently connected computer.  The "Display 
not found" lines in your output reflect the fact that the monitor is 
switched away from the current computer.

4) Whether it is possible to issue setvcp commands to a monitor taking 
its input from a different computer seems to be monitor dependent.  Once 
you have figured out the actual values of feature x60 for your monitor, 
try issuing the setvcp command from the computer whose connection is 
currently active, i.e. from the computer you are trying to switch away from.

5) Please execute the following commands on each of the computers and 
send the output in attached files.  It's more information than needed, 
but avoids lots of back and forth with little questions.

  ddcutil detect --verbose
  ddcutil environment --verbose

Alternatively, you can use the following convenience command, though it 
will take a long time to run as it performs additional checks.

  ddcutil interrogate

Sanford

ratmice, 2017-10-19:
> 2) To see the full list of values defined in the spec, use the command:
>
>   ddcutil vcpinfo 60 --verbose

Thanks, this was helpful, the HDMI input source ended up being
0x0f: DisplayPort-1

The following worked perfectly.
` $ ddcutil setvcp 60 0x0f`

So I suppose closing would be appropriate

closed 2017-10-19 by ratmice

rockowitz, 2017-10-19:
Matt,

Please run command "ddcutil interrogate" and send the output as an 
attachment.  I'd like to add your monitor to the rogues gallery at 
http://www.ddcutil.com/monitor_notes/

Sanford