rockowitz/ddcutil #37: Samsung 971p

notgood, 2017-10-21:
Not an issue per-se, just a quick report, ddcutil seems to be working just fine on good old Samsung 971p monitor.
Which is helpful, since it only has one hardware key to play with.

ddcutil interrogate output attached. [971p.txt](https://github.com/rockowitz/ddcutil/files/1404059/971p.txt)

Also, display mode labels are wrong, but I guess these are sample value only, since there is no way to query monitor for actual labels.

  Feature: DC (Display Mode)
    Values:

       01: Productivity -> Text
       02: Mixed -> Internet
       03: Movie -> Game
       04: User defined -> Sport
       05: Games -> Movie
       06: Sports -> Custom
       f0: Dynamic contrast = Dynamic contrast

rockowitz, 2017-10-21:
Thank you for taking the trouble to send the report.   It's helpful to 
see the variations in how monitors interpret the specs.

I notice you're on buster.   Did you install from the Debian buster 
repo, or some other way?

The names associated with values are taken from the Monitor Control 
Command Spec (MCCS).  You're correct, there's no way to query the 
monitor as to what names it associates the values. ddcutil just has to 
follow the spec.

It's actually a bit complicated when the MCCS version is taken into 
account, since values can be added and names changed. ddcutil tries to 
do the best it can.

Your monitor reports its VCP (MCCS) version as 2.0.   According the the 
2.0 spec, the defined values are

   x01 Productivity (e.g. office applications)
   x02 Mixed (e.g. internet)
   x03 Entertainment (e.g. TV/movie)
   x04 User defined.

According to the latest 2.2 spec, the defined values are:
   x00 Standard/default mode
   x01 Productivity (e.g. office applications)
   x02 Mixed (e.g. internet with mix of text and images)
   x03 Movie
   x04 User defined
   x05 Games (e.g. games console/PC game)
   x06 Sports (e.g. fast action)
   x07 Professional (all signal processing disabled)
   x08 Standard/default mode with intermediate power consumption
   x09 Standard/default mode with low power consumption
   x0a Demonstration (used for high visual impact in retail, etc.)
   xf0 Dynamic contrast

As you can see, even though your declared VCP version is 2.0, ddcutil is 
using some names from a later spec.

I'm considering adding the ability for users override the names using a 
local file.  This might be particularly useful when there are values not 
defined in the spec, or for VCP codes in the manufacturer-specific range 
xe0..xff.

Finally, a warning.  The information reported by "ddcutil capabilities", 
which is obtained from the monitor's capabilities string, should be 
taken only as a hint.  In practice, it is often wrong, and ddcutil does 
not rely on it. See, for example, the comparison of declared 
capabilities to observed features at the end of the interrogate output.

Regards,
Sanford

notgood, 2017-10-21:
I'm running Unstable Sid, installed ddcutil straight from Debian repository.

Here is another report: old low-end 19'' Acer V193 monitor. [Acer_V193.txt](https://github.com/rockowitz/ddcutil/files/1404264/Acer_V193.txt)

closed 2018-09-25 by rockowitz