rockowitz/ddcutil #38: pkg-config script issue

FedeDP, 2017-11-14:
Hi, first of all let me thank you for your great work!
By the way, trying to build an example with gcc, i realized that  `pkg-config --libs --cflags ddcutil` does not set required "-lddcutil" (that seems only needed flag too).
On my system (archlinux) version 0.8.4 sets:
`-I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -lglib-2.0 -lXrandr -lX11`

But without "-lddcutil" i am not able to build.
I guess that adding:
`Libs: -L${libdir} -lddcutil`
to ddcutil.pc.in should do the trick (if i am correct).
At least, that is what x11.pc does ("Libs: -L${libdir} -lX11").

rockowitz, 2017-11-15:
Thank you for reporting the bug.   The Libs: line has been modified as 
per your
suggestion and the change will appear in the upcoming point release.

If you're writing to the C API please keep me updated on how you're 
doing.  The API is a work in progress.
I'd appreciate hearing what works well, what is needlessly difficult, 
and is obscure.

Regards,
Sanford

FedeDP, 2017-11-15:
I'm only using ddca_set_continuous_vcp_value (and main loop as per https://github.com/rockowitz/ddcutil/blob/master/src/sample_clients/clmain.c) for my project (https://github.com/FedeDP/Clightd/tree/ddcutil#ddcutil-support), so no issues for the api.

Again, thanks for this software!

closed 2017-11-17 by FedeDP