rockowitz/ddcutil #40: Segfault on interrogate

Despruk, 2017-11-26:
Here's the output when running `interrogate`, same thing with `environment`

    $ sudo ddcutil interrogate
    Setting output level verbose...
    Setting maximum retries...
    Forcing --stats...
    Forcing --force-slave-address..
    This command will take a while to run...
    
    
    *** Basic System Information ***
    
    ddcutil version: 0.8.5
    
    /proc/version:
      Linux version 4.12.8-2-ARCH (builduser@foutrelis) (gcc version 7.2.0 (GCC) ) #1 SMP PREEMPT Fri Aug 18 14:08:02 UTC 2017
    
    Architecture:     (null)
    Distributor id:   Arch
    Release:          rolling
    Unexpected architecture (null).  Please report.
    Segmentation fault

Despruk, 2017-11-26:
https://github.com/rockowitz/ddcutil/blob/63a977191ac09a18f1c74079705a3cb80ae26cbb/src/app_sysenv/query_sysenv.c#L155
Looks like there is no `arch` program on Arch linux. `uname -m` would work.

rockowitz, 2017-11-27:
Changed.  The fix will appear in the next point release.

Thanks for the report.

rockowitz, 2018-01-23:
segfault fixed in 0.8.5

closed 2018-01-23 by rockowitz