rockowitz/ddcutil #41: Dell U3818DW on NVIDIA 960GTX not detected

moll, 2017-12-04:
Hey,

I'm having trouble getting ddcutil v0.8.5 detect a Dell 3818DW monitor over DisplayPort (Haven't tried HDMI yet). The `detect` command comes up with nothing. I previously used a Dell U2711 monitor on the same card (though through DVI), and that worked. Running `ddcctl` on MacOS with the same monitor over HDMI also works. Any suggestions? Thanks in advance!

```
$ ddcutil detect --verbose              
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false

No displays found
```

I2C detect lists the following:
```
i2cdetect -l
i2c-3	smbus     	SMBus PIIX4 adapter port 4 at 0b00	SMBus adapter
i2c-1	smbus     	SMBus PIIX4 adapter port 2 at 0b00	SMBus adapter
i2c-8	i2c       	NVIDIA i2c adapter 8 at e:00.0  	I2C adapter
i2c-6	i2c       	NVIDIA i2c adapter 6 at e:00.0  	I2C adapter
i2c-4	i2c       	NVIDIA i2c adapter 0 at e:00.0  	I2C adapter
i2c-2	smbus     	SMBus PIIX4 adapter port 3 at 0b00	SMBus adapter
i2c-0	smbus     	SMBus PIIX4 adapter port 0 at 0b00	SMBus adapter
i2c-9	i2c       	NVIDIA i2c adapter 9 at e:00.0  	I2C adapter
i2c-7	i2c       	NVIDIA i2c adapter 7 at e:00.0  	I2C adapter
i2c-5	i2c       	NVIDIA i2c adapter 2 at e:00.0  	I2C adapter
```

The interrogation command crashes because of https://github.com/rockowitz/ddcutil/issues/40.

rockowitz, 2017-12-04:
To summarize, you're running Arch Linux, using the Nvidia proprietary 
driver, and trying to use DisplayPort.

Nvidia's proprietary driver and DisplayPort I2C-over-Aux both seem to be 
problematic.  Usually they work, sometimes
they don't, typically with more recent cards.  What video card are you 
using?  Suggest you try a DVI or HDMI connection instead of DisplayPort, 
and also see if switching to the nouveau driver (at least temporarily) 
makes a difference.

Re the segfault in interrogate, there's a fix in the next point release 
for the lack of the "arch" command on Arch Linux. However, having just 
pushed out a release, the next one is some weeks away.   In the 
meantime, you could:
- See if there's some package on Arch Linux that provides the "arch" command
- Write a trivial script named arch and install it somewhere in your PATH:

   #!/bin/sh
   uname -m

- If you are building from source, modify line 155 in file 
sysenv/query_sysenv.c, changing "arch" to "uname -m"

Sanford

moll, 2018-11-13:
I can confirm that at least as of Nov 13, 2018 along with NVreg tweaks (below) U3818DW on 960GTX works! Thanks for your help!

`/etc/modprobe.d/nvidia.conf`:
```
options nvidia NVreg_RegistryDwords=RMUseSwI2c=0x01;RMI2cSpeed=100
```

closed 2018-11-13 by moll