rockowitz/ddcutil #42: Multiple unknown type name ‘PyFileObject’; did you mean ‘PyCodeObject’? errors

dracwyrm, 2018-01-20:
Hi

In src/swig/ddc_swig.h and ddc_swig.cpp I'm getting these errors:
```
src/swig/ddc_swig.h:92:31: error: unknown type name ‘PyFileObject’; did you mean ‘PyCodeObject’? void save_current_python_fout(PyFileObject * pfy);
src/swig/ddc_swig.h:93:1: error: unknown type name ‘PyFileObject’; did you mean ‘PyCodeObject’?
 PyFileObject * get_current_python_fout();
ddc_swig.c:154:8: error: unknown type name ‘PyFileObject’ static PyFileObject * current_python_fout;
ddc_swig.c:156:31: error: unknown type name ‘PyFileObject’; did you mean ‘PyCodeObject’? void save_current_python_fout(PyFileObject * pfy) {
ddc_swig.c:161:1: error: unknown type name ‘PyFileObject’; did you mean ‘PyCodeObject’? PyFileObject * get_current_python_fout() {
```

Python: 3.5.4
Swig: 3.0.12
Cffi: 1.11.2
Cython: 0.26.1

rockowitz, 2018-01-20:
Hi Jonathan,

Run configure with --disable-swig, --disable-cython --disable-cffi 
--disable-gobject
(These are the default settings.)

The Python code in the source tree is all experimental proof of concept 
explorations, so didn't get full attention in pushing out the release.
None of it is meant for public use.

Nonetheless, thanks for pointing out the errors.  The handling of FILE 
object changed between Python 2 and Python 3.  Looks like
the testing for Py2 vs 3 isn't quite right.

Regards,
Sanford

dracwyrm, 2018-01-21:
@rockowitz I disabled those and it builds fine.

I did look at the code and noticed that the line that errors is surrounded by a version test, but for some reason that didn't work right to prevent it from executing. The comment said there was no function like that in Python 3, which just makes it harder. There's a couple of workarounds that I found, but don't know if they would work for you.

I'll right for you to declare python safe for a public API before trying to enable it in the Gentoo Ebuilds. :)
Thanks for responding to this report.

closed 2018-01-23 by rockowitz