rockowitz/ddcutil #43: Compilation error when using the musl libc

newbluemoon, 2018-01-21:
Hi,

I was trying to compile version 0.8.6 on Void Linux using the musl libc and ran into the following error which I would like to report:

```
query_sysenv_usb.c: In function 'probe_hiddev':
query_sysenv_usb.c:193:26: error: overflow in implicit constant conversion [-Werror=overflow]
           rc = ioctl(fd, HIDIOCGDEVINFO, &dev_info);
                          ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

I can make the error disappear by removing `-Wpedantic` from `src/app_sysenv/Makefile` (it isn’t set in 0.8.5), but I’d rather solve the underlying issue ;) which seems to be that glibc defines `ioctl()` as

    extern int ioctl (int __fd, unsigned long int __request, ...) __THROW;

but the musl libc as

    int ioctl (int, int, ...);

It compiles when casting to `int`:

    rc = ioctl(fd, (int) HIDIOCGDEVINFO, &dev_info);

but I don’t know enough about the whole matter to be sure that’s safe to do.

There are a couple of instances more where a warning is issued, but not treated as an error:

```
make[3]: Entering directory '/builddir/ddcutil-0.8.6/src/usb'
  CC       usb_base.lo
usb_base.c: In function 'hiddev_get_device_info':
usb_base.c:188:23: warning: overflow in implicit constant conversion [-Woverflow]
    int rc = ioctl(fd, HIDIOCGDEVINFO, dev_info);
                       ^~~~~~~~~~~~~~
usb_base.c: In function 'hiddev_get_report_info':
usb_base.c:208:23: warning: overflow in implicit constant conversion [-Woverflow]
    int rc = ioctl(fd, HIDIOCGREPORTINFO, rinfo);
                       ^~~~~~~~~~~~~~~~~
usb_base.c: In function 'hiddev_get_field_info':
usb_base.c:228:23: warning: overflow in implicit constant conversion [-Woverflow]
    int rc = ioctl(fd, HIDIOCGFIELDINFO, finfo);
                       ^~~~~~~~~~~~~~~~
usb_base.c: In function 'hiddev_get_usage_code':
usb_base.c:250:23: warning: overflow in implicit constant conversion [-Woverflow]
    int rc = ioctl(fd, HIDIOCGUCODE, uref);    // Fills in usage code
                       ^~~~~~~~~~~~
usb_base.c: In function 'hiddev_get_usage_value':
usb_base.c:267:23: warning: overflow in implicit constant conversion [-Woverflow]
    int rc = ioctl(fd, HIDIOCGUSAGE, uref);
                       ^~~~~~~~~~~~
usb_base.c: In function 'hiddev_get_report':
usb_base.c:283:23: warning: overflow in implicit constant conversion [-Woverflow]
    int rc = ioctl(fd, HIDIOCGUCODE, rinfo);
                       ^~~~~~~~~~~~

...

usb_vcp.c: In function 'usb_get_usage_value_by_report_type_and_ucode':
usb_vcp.c:133:19: warning: overflow in implicit constant conversion [-Woverflow]
    rc = ioctl(fd, HIDIOCGFIELDINFO, &finfo);  // Fills in usage value
                   ^~~~~~~~~~~~~~~~

```

rockowitz, 2018-01-21:
Thanks for the report.   I wasn't aware of Void Linux.  (You might want 
have it added to the distributions tracked at repology.org.)

Per ioctl.h, ioctl commands are encoded into 32 bits.  So assuming that 
your int size is at least 32 bits, the warning can be ignored.

The --pedantic option just enforces strict ISO C compatibility. It was 
introduced in a fruitless attempt to figure out what was triggering a 
known stdlib.h/floatn.h error regarding 128 bit floats (irrelevant to 
ddcutil) that Coverity scan was complaining about.   You can safely 
eliminate the option.

The reason that query_sysenv_usb.c failed to compile, but the usb_base.c 
just issued errors, is due to a difference in the Makefile for 
subdirectory app_sysenv vs that for subdirectory usb.  Files in the 
former are compiled with -Werror, so all warnings are treated as errors.

Sanford

newbluemoon, 2018-01-21:
Thank you for your reply! Yes, integer size is 32 bit and I’ll drop -Wpedandic.
Didn’t know about repology.org, will check it out.

So I guess this can be closed then. Thanks again! :)

closed 2018-01-21 by newbluemoon