rockowitz/ddcutil #44: Some ddcutil memleaks

FedeDP, 2018-01-27:
Hi!
I'm currently using ddcutil C api 0.8.6 in a project.
I noticed that valgrind points out some memleaks, or at least some memory that is not freed after program ends; in fact they're not proper memleaks (ie: consecutive calls to ddcutil api won't leak new memory).

This is the code i'm using: http://dpaste.com/1H4QE1N. (it is a dbus interface to change both laptop's internal and external monitors backlight/brightness).
And this is valgrind output with debug symbols enabled: http://dpaste.com/1JGV0S5.

Hope this helps :)

rockowitz, 2018-01-28:
Federico,

Thanks for the detailed report.    I believe I've plugged the leaks you 
caught.  There's also a new API function ddca_free_any_vcp_value() to 
free the DDCA_Any_Vcp_Value struct returned by 
ddca_get_any_vcp_value().   The changes will go into the next point 
release.  If you'd like I can push out the changes to a branch on github 
or build a private tarball for you to test.   The latter is preferable 
for me as I prefer to keep only actual release code in the github repo.

Regards,
Sanford

FedeDP, 2018-01-28:
If you prefer latter method, i'm all for that!
Waiting for your input to try and check if it is now fixed!

rockowitz, 2018-01-28:
A tarball with the memory fixes and other recent changes can be found at 
http://www.ddcutil.com/tarballs/ddcutil-0.8.7.tar.gz

Note that the naming is a misnomer.  It's called 0.8.7 but not actually 
a release.

Sanford

FedeDP, 2018-01-28:
Thanks!
I still see 2 errors: http://dpaste.com/2D9F4CJ.

rockowitz, 2018-01-28:
Updated tarball at http://www.ddcutil.com/tarballs/ddcutil-0.8.7.tar.gz 
(same name as before).   Thanks again for the testing.

Sanford

FedeDP, 2018-01-28:
Thank you for your support!
I still see 3 errors: http://dpaste.com/0NK2G8M.
Moreover, i see (only) this error if executed without modprobing i2c-dev module: http://dpaste.com/1S6ZPF.
I'm glad to be helpful :)

rockowitz, 2018-01-29:
Uploaded a http://www.ddcutil.com/tarballs/ddcutil-0.8.7.tar.gz

The 3 errors were actually 1.  The earlier fixes freed a data structure 
too soon.   Without modprobing i2c-dev, monitors were undetected, so the 
new bug wasn't hit.

Thanks again.

Sanford

FedeDP, 2018-01-29:
Still 2 of them: http://dpaste.com/1CTCV6Q :)

Btw thanks for your support! I am sorry i got no time to look into the code and submit a patch myself.

rockowitz, 2018-01-31:
New tarball at http://www.ddcutil.com/tarballs/ddcutil-0.8.7.tar.gz

I've fixed a lot of allocation errors uncovered by valgrind.  I'd been 
relying on Coverity scan to find memory leaks, but clearly static 
analysis can only go so far.  Thanks for your help finding these 
problems.  No need for patches - once valgrind identifies a leak the fix 
is usually pretty obvious.

Sanford

FedeDP, 2018-01-31:
And all memleaks are gone!
Thanks for your support, looking forward to 0.8.7 release ;)

closed 2018-01-31 by FedeDP