rockowitz/ddcutil #45: Several "ISO C99 requires at least one argument for the "..." in a variadic macro" errors.

dracwyrm, 2018-01-27:
On Gentoo, a number of users are reporting this error. Since build logs are very long, I'll post the links to them.

One log is here: https://645830.bugs.gentoo.org/attachment.cgi?id=516796
Another user has slightly different functions that exhibit this: https://bugs.gentoo.org/645782

Is there a dependency version or lack there of that needs consideration?

Thanks.

rockowitz, 2018-01-28:
Jonathan,

Thanks for the clear reports.

Simplest solution:  Delete or comment out the line "AM_CFLAGS += 
-Wpedantic" in file src/ddc/Makefile.am, rerun automake and configure 
and then build.

Background:  I had added the -Wpedantic flag trying to identify the 
source of a benign (to ddcutil) error report in Coverity scan regarding 
__Float128 being undefined.  The flag causes warnings to be issued for 
violations of strict ISO compatibility.   If -Werror is also set, this 
results in a fatal error.   In particular, variadic macros, 
PROGRAM_LOGIC_ERROR() in this case, without variadic arguments are 
disallowed.  The use of PROGRAM_LOGIC_ERROR() in this way was only 
encountered if "--disable-usb" was passed to configure.

The overzealous use of -Wpedantic will be fixed in the next point release.

Sanford

dracwyrm, 2018-01-28:
Thanks for the feedback. Do you know when the next point release will be?

rockowitz, 2018-01-28:
Unless there are urgent issues, the next actual point release is several 
weeks away.

Sanford

rockowitz, 2018-05-14:
The -Wpedantic compile option requiring strict C99 compliance is no longer used in Release 0.9.0, so this problem should be eliminated.

closed 2018-05-19 by rockowitz