rockowitz/ddcutil #46: Invalid Display, HP monitor sends only 80 80 be 6e 6e bytes over I2C

monkeydri, 2018-02-02:
I try to control brightness of my HP monitor HP 27es (codename HWP3325), pluged in HDMI port of Asus Laptop (UX310). I tried both HDMI ports on the monitor (it was suggested by gddccontrol utility, which did not work on the first port. ddcutil reports same output on both ports).

**DDC/IC enabled in OSD**.

ddcutil detect reports Invalid Display :

```
ddcutil detect --verbose --ddc --force-slave-address
Output level:               Verbose
Reporting DDC data errors:  true
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    true

DDC: Unexpected source address 0x80, should be 0x6e
DDC: i2c_response_bytes: 80 80 be 6e 6e 80 be 6e 80 be 00 6e 80 be 6e 80 be 6e 6e 80
DDC: Unexpected source address 0xbe, should be 0x6e
DDC: i2c_response_bytes: be 00 6e 80 be 6e 80 be 6e 80 be be 6e 6e 80 be 6e 80 be be
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 80 be 6e 80 be 6e 80 be 00 6e 80 be 6e 80 be 6e 6e 80
DDC: Unexpected source address 0xbe, should be 0x6e
DDC: i2c_response_bytes: be 00 6e 80 be 6e 80 be 6e 80 be be 6e 80 be 6e 80 be 00 6e
DDC: Unexpected source address 0x80, should be 0x6e
DDC: i2c_response_bytes: 80 80 be 6e 80 be 6e 80 be 6e 80 80 be 6e 80 be 6e 80 be 6e
DDC: Unexpected source address 0x80, should be 0x6e
DDC: i2c_response_bytes: 80 80 be 6e 6e 80 be 6e 6e 80 be be 6e 6e 80 be 6e 80 be be
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 80 be 6e 80 be 6e 80 be 00 6e 80 be 6e 80 be 6e 80 be
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 80 be 6e 6e 80 be 6e 80 be be 6e 6e 80 be 6e 80 be be
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 80 be 6e 80 be 6e 80 be 00 6e 80 be 6e 80 be 6e 80 be
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 80 be 6e 80 be 6e 80 be 00 6e 80 be 6e 80 be 6e 80 be
Invalid display
   I2C bus:             /dev/i2c-0
   Supports DDC:        true
      I2C address 0x30 (EDID block#)  present: false
      I2C address 0x37 (DDC)          present: true
      I2C address 0x50 (EDID)         present: true
      /sys/bus/i2c/devices/i2c-0/name: i915 gmbus dpc
   EDID synopsis:
      Mfg id:           HWP
      Model:            HP 27es
      Serial number:    3CM7240ZLG   
      Manufacture year: 2017
      EDID version:     1.3
      Product code:     0x3326 (13094)
      Extra descriptor: Unspecified
      Video input definition: 0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.312, 0.328
      Red   x,y:        0.652, 0.337
      Green x,y:        0.319, 0.611
      Blue  x,y:        0.149, 0.067
      Extension blocks: 1
   EDID source:
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 22 f0 26 33 01 00 00 00   ........".&3....
      +0010   18 1b 01 03 80 3c 22 78 2a 1e 50 a7 56 51 9c 26   .....<"x*.P.VQ.&
      +0020   11 50 54 a1 08 00 d1 c0 b3 00 a9 c0 95 00 81 80   .PT.............
      +0030   81 00 81 c0 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
      +0040   45 00 56 50 21 00 00 1e 00 00 00 fd 00 32 3c 1e   E.VP!........2<.
      +0050   50 11 00 0a 20 20 20 20 20 20 00 00 00 fc 00 48   P...      .....H
      +0060   50 20 32 37 65 73 0a 20 20 20 20 20 00 00 00 ff   P 27es.     ....
      +0070   00 33 43 4d 37 32 34 30 5a 4c 47 20 20 20 01 38   .3CM7240ZLG   .8
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?
```

ddcutil 0.8.6 on Ubuntu 17.10 (wayland) for what it worth.

[ddcutil-detect.txt](https://github.com/rockowitz/ddcutil/files/1689946/ddcutil-detect.txt)
[ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/1689947/ddcutil-environment.txt)

rockowitz, 2018-02-03:
Thank you for the detailed report. Unfortunately, I don't have very good 
news.   The I2C communication with the monitor is garbled.  Some 
monitors are just dirty.   I'll add the HP 27es to the rogues gallery at 
www.ddcutil.com/monitor_notes.

Have you tried plugging in a different monitor for testing? 
Alternatively, have you tried running ddcutil with this monitor on 
another system?

It doesn't appear that your laptop is docked.  But if it is, try using a 
non-dock connection.  There are problems with the i915 driver when using 
a docking station.

Try the undocumented option "--sleep-strategy 2", aka "-y 2". This 
doubles the time that ddcutil sleeps over that required by the DDC/CI 
spec.  Occasionally this helps.

Lastly, when submitting reports in the future, please use the 
"--verbose" and "--stats" options.

Sanford

monkeydri, 2018-02-05:
Thank you very much for your answer.

I tried the option `--sleep-strategy 2` but no success. I also plugged it to another computer with the same result. I have no other HDMI monitor to try.

I plan to plug the monitor to a windows computer through a HDMI breakout board with an I2C USB sniffer (I happen to own a saleae logic analyzer), because there is a HP tool on windows which is able to control brightness and other settings, so there is certainly a way to do it.

I will let you know once its done, thanks again !

rockowitz, 2018-02-06:
Do let me know what you find with the logic analyzer.   I would not be 
surprised to find that the tolerances are marginal - good enough for the 
Windows driver, which appears to be more sophisticated than what is 
available on Linux.   It's also likely that monitor testing occurs on 
Windows, not on Linux.

You might want to install EnTech's softMCCS utility for Windows. EnTech 
is a make of DDC chipsets, and softMCCS is their utility for exercising 
the full protocol.

Short of writing my own I2C drivers, there's not much that ddcutil can 
do to tweak the I2C layer.  And therein lies maintenance madness.

Sanford

closed 2025-02-09 by rockowitz