rockowitz/ddcutil #47: Some output with ddcutil interrogate for Dell U3818DW

1kevin335200, 2018-02-02:
[output.txt](https://github.com/rockowitz/ddcutil/files/1691101/output.txt)

rockowitz, 2018-02-03:
Thank you for report.  It appears that the U3818DW is one of those 
monitors that never responds to a feature (getvcp) request with an 
indication that the feature is unsupported.  I've seen this before on 
inexpensive monitors, but never on a high-end monitor such as an 
Ultrasharp.    To confirm, can you run the following command and send me 
the output:

   ddcutil getvcp 59 --trace ddc

(I chose feature 59 because it's not in the listed capabilities and 
appears to have a garbage response.)

Sanford

1kevin335200, 2018-02-03:
Here it is:
[output.txt](https://github.com/rockowitz/ddcutil/files/1691698/output.txt)

rockowitz, 2018-02-03:
Looking at the DDC packet confirms that the "Unsupported VCP Code" flag 
is not set even though the feature is unsupported.

Does this misinformation affect your use of ddcutil?

Sanford

rockowitz, 2018-02-03:
> output.txt <https://github.com/rockowitz/ddcutil/files/1691705/output.txt>
> This is the output of command "ddcutil getvcp known"

VCP feature group KNOWN includes all VCP codes for which ddcutil has a 
definition, i.e. all codes defined in the MCCS (Monitor Control Command 
Set) specs.
In your case, "ddcutil getvcp known" tries to read each of the known 
features, and the monitor replies that is has a valid answer, even when 
the feature code is unsupported.

VCP feature group SCAN, on the the other hand, includes all codes x01 to 
xff.

1kevin335200, 2018-02-03:
No. I only try using ddcutil to adjust brightness and volume and it works.
Thanks for the information, now i have learned a little more about it

closed 2019-03-05 by rockowitz