rockowitz/ddcutil #48: Macro for ddcutil version

FedeDP, 2018-02-18:
Hi,
is there a macro to check ddcutil major.minor.patch version? 
I see there are 2 runtime functions, but i cannot see any macro for it.

Thanks!

rockowitz, 2018-02-19:
There's no special macro, but what would it do that the API calls do not?

Sanford

FedeDP, 2018-02-19:
Dealing with (possible) API breaks?
Eg, i now use it this way:
<pre>#if DDCUTIL_VERSION_MAJ == 0 && DDCUTIL_VERSION_MIN == 8 && DDCUTIL_VERSION_PATCH < 7
    #define ddca_free_any_vcp_value(x) (void)x
#endif</pre>

For now, i'm creating these macros from pkg-config output:
<pre>DDCUTIL_VERSION_MAJ=$(shell pkg-config --modversion ddcutil | cut -d '.' -f1)
DDCUTIL_VERSION_MIN=$(shell pkg-config --modversion ddcutil | cut -d '.' -f2)
DDCUTIL_VERSION_PATCH=$(shell pkg-config --modversion ddcutil | cut -d '.' -f3)</pre>

But it would surely be  better to use a macro shipped by you (as i always fear that pkg-config file is not synced).

rockowitz, 2018-02-25:
Federico,

First, my apologies for taking to so long to reply.  I've been 
traveling, and the answer required a bit of thought.

The only viable way to create macros that work at build time is to 
generate them from the Autotools configuration, based on the version 
values in configure.ac.   This in turn leads to some ugly dependency 
issues.

I'd suggest creating wrapper functions, or macro equivalents, e.g.

void my_ddca_free_any_vcp_value(DDCA_Any_Vcp_Value * valrec) {
    DDCA_Ddcutil_Version_Spec vspec = ddca_ddcutil_version();
    if (vspec.major == 0 && vspec.minor == 8 && vspec.micro < 7)
      ddca_free_any_vcp_value(valrec);
}

This makes the function call slightly more expensive, but the overhead 
entails no IO so is trivial. ddcutil spends 90% of its elapsed time in 
sleeps mandated by the DDC/CI protocol.  Even if ddcutil's non-IO time 
were reduced to 0, there would be no perceptible decrease in elapsed time.

As a heads up, the next point release will have extensive changes to the 
API, base on my experience building a GUI UI.

Sanford

FedeDP, 2018-02-26:
That would give "undefined reference" on a 0.8.6 version of ddcutil, as that version would not provide ddca_free_any_vcp_value() that is still referenced in code, even if not used.
Did i misunderstand anything?

By the way is there an ETA for 0.8.7?
Thanks!

rockowitz, 2018-02-26:
You're absolutely right.  I'll give the problem some more thought.

0.8.7 is several weeks away.  The bug fixes and enhancements that have 
accumulated for the command line version are minor, and I expect more 
API changes.

Sanford

FedeDP, 2018-02-26:
> 0.8.7 is several weeks away
I see, thanks for the info!

rockowitz, 2018-05-14:
Federico, 

The new 0.9.0 release includes file ddcutil_macros.h, which contains macro definitions DDCUTIL_VMAJOR, DDCUTIL_VMINOR, and DDCUTIL_VMICRO specifying the version number.  

Note that the C API has been thoroughly reworked, and hopefully regularized.   See [C API Changes for Release 0.9.0](http://www.ddcutil.com/c_api_90/).

Sanford

FedeDP, 2018-05-14:
Thank you very much!  
I'll let you know if I encounter any issue ;)

closed 2018-05-14 by FedeDP