rockowitz/ddcutil #50: LG 22MP48D-P Invalid display, DDC communication failed

linxlan, 2018-03-02:
ddcutil interrogate output: https://pastebin.com/88QT5M3n

rockowitz, 2018-03-02:
Thank you for sending the report. Here is what I see:

- It's a desktop system, with integrated Intel graphics (i915). The monitor
   is connected to an HDMI output.   (Could be DVI, sometimes mis-labeled.)

- There's an Nvidia card in the system, but the Nvidia driver does not 
load.

- The I2C communication is "dirty", with duplicate bytes in the stream.
It may be that the monitor is out of tolerance with the I2C/DDC specs,
it may be that the i915 driver is unforgiving, or it may be an issue in the
Linux I2C layer.

Have you been able to test any permutations:
- different monitor
- this monitor on a Windows system using softMCCS from enTech
- use the Nvidia driver

Sanford

linxlan, 2018-03-02:
> - There's an Nvidia card in the system, but the Nvidia driver does not
load.
I use nvidia for gpu passthrough

> Have you been able to test any permutations:
>- different monitor
>- this monitor on a Windows system using softMCCS from enTech
>- use the Nvidia driver
Another monitor (BenQ G925HDA) seems work. I had no errors when executed "ddcutil detect" and successfully changed display brightness
softMCCS work with both monitors
ddcutil interrogate output for second monitor
[log2.txt](https://github.com/rockowitz/ddcutil/files/1775405/log2.txt)

linxlan, 2018-03-03:
Just tested with nvidia driver. Everything works fine with xorg setting:
`Option "RegistryDwords" "RMUseSwI2c=0x01; RMI2cSpeed=100"`
I think there should be some equivalent of this setting for intel driver.

rockowitz, 2018-03-04:
Hate to say it, but it seems that the Windows drivers are just more 
robust in handling monitors with marginal I2C implementations.  For the 
open source drivers, I2C handling is shared by the general I2C layer and 
the specific video  driver (.e.g. i915, nouveau).  Unfortunately, the 
userspace i2c-dev driver has
no mechanism for tweaking things like bus speed.

I realize it's a bit of work, but it would be interesting to see if 
communication works with the nouveau driver instead of the proprietary 
video driver.   That would indicate whether the issue is specific to the 
i915 driver.  (The open-source drivers use dev-i2c, the proprietary 
Nvidia driver does not.)

Sanford

linxlan, 2018-03-05:
Nouveau driver also works.
ddcutil interrogate output attached
[log4.txt](https://github.com/rockowitz/ddcutil/files/1779275/log4.txt)