rockowitz/ddcutil #51: LG-27UD49 DisplayPort failed to detect

evansan, 2018-03-25:
[interrogate.log](https://github.com/rockowitz/ddcutil/files/1845011/interrogate.log)

evansan, 2018-03-25:
I have RX 470
Kernel is 4.15.12-1-zen
driver is amdgpu

rockowitz, 2018-03-25:
1) Module i2c-dev is required but not loaded.   The I2C devices exist in
/sys/bus/i2c/devices but not /dev/i2c-* are not present.   

If loading i2c-dev does not resolve the problem, read on ...

2) The output of "ddcutil environment" is one part of "ddcutil
interrogate", and its output can be hard to spot.  It provides
suggestions for resolving problems.  Run it standalone and see what it
suggests.

3) Install i2c-tools.  Program i2cdetect can be helpful.

4) I see you're running a custom Zen kernel.  What is the relation of
ddcutil to the virtualization?
(The config files are not in the expected location in /boot, but I doubt
there's much to be found there.)

5) I see you're running XWayland.  Unfortunately, I've not been able to
get it running on any of my systems, so no experience in this regard.

6) ddcci is listed in DKMS, but it's not clear if it's loaded.  May
present a conflict.

Sanford

evansan, 2018-04-02:
solved by loading i2c-dev

closed 2018-04-02 by evansan

rockowitz, 2018-04-02:
Thanks for posting the resolution.