rockowitz/ddcutil #52: HDVI-DVI adapter breaking DDC communication?

hfcheng, 2018-05-07:
Disclaimer: This is probably not a ddcutil issue, but posted for those people having the same problem like me.

Running into a issue where DDC communication failed + 0x30 (EDID block#) present=false, but the situation is a bit odd. My environment has a years old Dell U2410H with 2 DVI inputs, I have one HDMI-DVI adapter purchased years ago looks like [this](https://www.amazon.com/Cable-Matters-Gold-Plated-Female-Adapter/dp/B007NLEF3W) and ddcutil works perfectly.

Recently I purchased another adapter and things stopped working. The display has no problem at all but only the DDC functionality is broken. Tried some other similar adapters, the situation remained the same. Simply swap the old adapter back and everything works like magic again. It also works if I use a DVI cable + Apple HDMI/DVI adapter.

Looks like something inside the adapter is breaking the DDC communication, but that is just my speculation. Log attached for reference.
[log.txt](https://github.com/rockowitz/ddcutil/files/1979484/log.txt)

rockowitz, 2018-05-10:
Hello David,

Thanks for sharing your experience.  It may be helpful to others.

First, a couple general points.  ddcutil environment/interrogate shows 
the output of i2cdetect, which is a separate program from ddcutil.  The 
output shows slave address x50 (EDID) present, but not slave address x37 
(DDC).  So the problem is clearly happening at a level below ddcutil.   
Also, the fact that slave address x30 (EDID block number) is not present 
is purely informational.  It's a mechanism for accessing secondary EDID 
blocks on newer monitors.  ddcutil uses only the primary EDID block.

I agree that what you're seeing is very odd.   From a signaling point 
perspective, the common features of DVI and HDMI are interchangeable.  
If fact, it's common for monitors/drivers to misidentify HDMI 
connections as DVI connections, to no ill effect.  Per the pinout specs 
(see wikipedia), the DDC signal lines on a DVI-HDMI adapter should be 
straight passthru, though it may be that the ground for the lines is 
shared differently.  It's also the case that address x50 (EDID) is 
working, but not address (x37).  I have no explanation for why an 
adapter should affect only  one slave address and not the other.

One thought:  Are you using the same physical connection on the video 
card and monitor for all the tests?

I'll continue to give the problem some thought.

Regards,
Sanford

rockowitz, 2018-05-13:
> Sanford,
>
> Firstly appreciated for your time looking into this.
>
> Yes, during the tests I tried to make the adapter as the independent
> variable.
> I also made lots of different tests trying to isolate the root cause:
>
>   * Different HDMI cable
>   * Different DVI input (the Dell monitor has two)
>   * Different hosts (from RPi3, Intel NUC and MBP)
>   * Different monitor (two both from Dell and old though, newer
>     monitors no longer have DVI inputs)
>   * Direct HDMI link to monitor
>     All results are pointing to the same conclusion (the adapter).
>
> Just did yet another test where I have chained two HDMI-DVI adapters
> in the middle of HDMI link:
>
> Like this:
> Monitor HDMI input -- HDMI-DVI-Bad -- Apple HDMI/DVI Adapter -- HDMI
> source
>
> And the result remained the same, as long as I replace the adapter
> back to the old one I was using (even they look extremely similar),
> everything is back to normal.
>
> HDMI-DVI signals are interchangeable is also what I learnt and
> everyone is saying the adapter should be bi-directional. So I am
> totally lost why the vendors could possibly get this wrong.
>
> But anyway, I think it does not worth the effort investigating further
> or trying more adapters.
> I should live with the current working configuration and still enjoy
> the great tool you created. :)
>
> David

David,

Thank you for taking the effort to thoroughly test and document this 
failure case.  I'm sure it will help others trying to puzzle why DDC 
communication is not working.   It certainly will help me in responding 
to users with less precise complaints of ddcutil not working.

Regards,
Sanford

closed 2018-05-14 by hfcheng