rockowitz/ddcutil #53: Does anyone have ddcutil or any other  DDC utility working in Fedora/Nvidia?

Futureknows, 2018-06-19:
Tried almost everything on multiple displays. Seems like Nvidia/Pascal just doesn't support DDC although once a couple years ago it worked for a day. Since then I've spent hours and nothing.

rockowitz, 2018-06-20:
I assume you're referring to the proprietary Nvidia driver.  Yes, its 
I2C support is problematic. I assume you've seen the Nvidia page 
<http://www.ddcutil.com/nvidia/>  on the website.   Note the recent 
addition courtesy of user arcnnmx about the need to put settings in 
/etc/modprobe.conf or /etc/modprobe.d/something.conf.  The only thing I 
would add is that I seem to be having better luck with DisplayPort 
connected devices, which emulate I2C over the Aux channel.

Sanford

Futureknows, 2018-06-20:
Thanks for responding. After addind modprobe I actually have ddcutil working on HDMI. 
For some reason gddccontrol won't work although the i2c is clearly working. 
For now I will stick with ddcutil, thanks!

closed 2025-02-09 by rockowitz