rockowitz/ddcutil #56: Viewsonic XG2703-GS not playing nice

lhw, 2018-07-19:
The usual maximum retries exceeded information. Note that I was able to use the proprietary driver before over DVI without any issues with an older monitor. But the new monitor requires HDMI and even with the module options I am not able to make it play nice.
I have a trace output here: https://gist.github.com/lhw/bdb6a2d3ee9cad1898e45a6398cc1a75 plus some interrogation information.
Maybe you have some insight which could get this to work.  But seeing this:

`(create_ddc_response_packet) Starting. i2c_response_bytes=6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e` 

doesn't get my hopes up.
Otherwise it might be the first entry from Viewsonic for your rogues gallery.

rockowitz, 2018-07-19:
Lennart,

Thanks for the report.

Yes, the Viewsonic XG2703-GS is a very confused monitor.  When queried 
for an unsupported VCP feature, it does not reply properly.  Sometimes 
(e.g. feature x00 which never exists and is queried during monitor 
examination) it returns a well-formed response packet, but with the 
invalid opcode flag not set.   Other times, e.g. feature x60 which you 
traced, it just returns garbage, which causes ddcutil to retry.  (Note 
that feature x60 is not listed in the capabilities string.)  On the 
other hand, examining the probe output at the end of interrogate, I see 
that supported features do return normally.

I note also that VCP feature xC8 returns xFF for the display controller 
manufacturer, indicating a manufacturer designed controller.

So to paraphrase on old joke:
   Patient:  It hurts when I lift my hand above my head.
   Doctor:  Then don't lift your hand above your head. (i.e. don't use 
unsupported features)

As a side note, a Qt based GUI for ddcutil is in development.   It works 
for well implemented monitors, but gracefully handling poor monitor DDC 
implementations in a GUI environment is challenging.

Regards,
Sanford

lhw, 2018-07-19:
Alright. That was about what I expected. The other features don't really interest me for my use case. Thanks for the read-through.

closed 2018-07-19 by lhw