rockowitz/ddcutil #57: Fails to build with GCC-8

sten0, 2018-07-21:
Hi,

I'm not the Debian maintainer of this package, but just a user who has been struggling for years to have reliable DDC control of his monitor's backlight.  Would you please resolve this issue so that ddcutil remain part of the next Debian (and derivatives such as Ubuntu) release?

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=897737

Thanks!
Nicholas

sten0, 2018-07-22:
Hi Sanford,

On Sat, Jul 21, 2018 at 04:25:55PM -0700, rockowitz wrote:
>    Nicholas,
> 
>    The GCC 8 issues were addressed in ddcutil 0.9.0. The current version
>    in Debian test is 0.9.1. I'm learning what I need to do to close out
>    the Debian bug report.

Good to hear!  :-) For Bug #897737 I set the "forwarded" URL to this
issue.  Here is the relevant documentation: https://www.debian.org/Bugs/server-control

It is strongly preferred to reply to the bug's email address and use
"Control: foo bar" headers rather than emailing the control server.

I find email faster than webpage copy&pasting, and if you do to then
"bts show --mbox 897737", which on my system opens up the thread in
mutt; other MUAs ought to work fine though.  Find the most recent
email, and reply to that one.  At the top of the email, add BTS
control headers.  eg:
```
Control: fixed -1 sourcepackage/version-in-Debian-archive
```

`-1` in this case refers to the bug being replied to.  0.9.0 wasn't
uploaded to Debian, so you'll want to use `0.9.1-1`.  Always consult
the changelog and/or tracker.debian.org to make sure that the version
that contains the fix was successfully accepted into the archive.
This isn't as big of an issue if you're just marked the affected and
nonaffected versions, but in this case we're closing a bug.

Now change the To: address from 897737@bugs.debian.org to what it
needs to be to close the bug

https://www.debian.org/Bugs/Developer#closing

This is the preferred way to close bugs :-)

Happy hacking!
Nicholas

rockowitz, 2018-07-22:
Debian bug 897737 has been closed.

closed 2018-07-22 by rockowitz