rockowitz/ddcutil #59: execution_stats.c:686:22: error: statement will never be executed [-Werror=switch-unreachable]

dracwyrm, 2018-08-05:
Gentoo Linux has strict code checking, and this error came up. Please see this bug report for all logs as text files can't be uploaded here (I don't think) and some of them are really long: https://bugs.gentoo.org/662804

In file included from execution_stats.c:31:0:
execution_stats.c: In function ‘call_tuned_sleep’:
execution_stats.c:686:22: error: statement will never be executed [-Werror=switch-unreachable]
    assert(event_type != SE_DDC_NULL);
           ~~~~~~~~~~~^~~~
cc1: all warnings being treated as errors

Thanks.

rockowitz, 2018-08-05:
Jonathan,

Thanks for reporting the error.  The bug has been fixed and the change 
will be included in the next point release.    For now, the offending 
assert() statement can safely be deleted.

Sanford

dracwyrm, 2018-08-05:
Thank you for the very quick response! Glad this helped.

closed 2018-09-18 by rockowitz