rockowitz/ddcutil #60: Ddcutil C api freezes display

FedeDP, 2018-08-28:
Hi!

I am using ddcutil C api (0.9.1) in my project, https://github.com/FedeDP/Clightd.
I use it to change backlight level on external monitors.

Recently an issue was opened (https://github.com/FedeDP/Clight/issues/35), complaining about "part of screen" freezed when using my software. Digging into the issue revealed that the issue comes from my src/backlight.c DDCUTIL_LOOP macro, used at L184. 
When commenting that loop, brightness setting works fine.
The fact is the loop is copied and adapted from your examples.

Note that issue opener has only an internal laptop monitor (dell xps13), and "ddcutil detect" or "ddcutil setvcp 0x10 5" both return "Display not found"; my guess is that DDCUTIL_LOOP shouldn't do anything.

Systemd logs contain an error:
`Aug 27 21:24:09 tiny kernel: i2c_designware i2c_designware.0: controller timed out
Aug 27 21:24:08 tiny kernel: i2c_designware i2c_designware.0: controller timed out
Aug 27 21:24:07 tiny kernel: i2c_designware i2c_designware.0: controller timed out
Aug 27 21:24:06 tiny systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
Aug 27 21:24:06 tiny dbus-daemon[399]: [system] Successfully activated service 'org.clightd.backlight'
Aug 27 21:24:06 tiny systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
Aug 27 21:24:06 tiny dbus-daemon[399]: [system] Activating via systemd: service name='org.clightd.backlight' unit='clightd.service' requested by ':1.82' (uid=1000 pid=3910 comm="busctl call org.clightd.backlight >`

Have you any idea?
Thanks!

FedeDP, 2018-09-17:
I'm sorry to bump this thread...is there some more info that i can provide you?

rockowitz, 2018-09-17:
Federico, 

I had sent replies, but apparently they did not get connected to the thread.

Sanford

From the first reply:
I'm also not sure what to say about the i2c_designware controller timed out issue.  I do note that there's a lot of activity re designware on the linux i2c mailing list. 

From the second reply: 

I have found file brightness.c in your project CLightd.   I had been looking in CLight.  I would suggest the following:

1) Build a custom version of CLightd that reports the contents of struct DDCA_Display_Info.   API functions ddca_report_display_info() and ddca_report_display_info_list() may be helpful.

2) Ask your user to run "ddcutil interrogate".  The output may identify issues.  You can send this to me for review.

FedeDP, 2018-09-18:
Thanks for the reply!

> I had sent replies, but apparently they did not get connected to the thread.

Np.

> I'm also not sure what to say about the i2c_designware controller timed out issue. I do note that there's a lot of activity re designware on the linux i2c mailing list.

Yeah i saw that too.

Btw will suggest to run "ddcutil interrogate" to the user experiencing this issue and i'll try to get back here with more (useful) info.
Thanks!

FedeDP, 2018-09-18:
Here is `ddcutil interrogate` output: http://dpaste.com/0MEDHWW. Let me know if you find anything that could relate to the issue.
Thanks!

rockowitz, 2018-09-18:
Federico,

1) If the system has not been configured to give the current user RW 
access to all /dev/i2c* devices, command "ddcutil interrogate" needs to 
be run as root.   Otherwise, the /dev/i2c devices cannot be probed.

2) There are no external displays connected - only the internal laptop 
monitor (eDP connection).  Internal displays do not support DDC.

3) The system journal has lots the following entry:

     Sep 11 09:52:02 tiny root[3650]: ACPI group/action undefined: 
video/brightnessup / BRTUP

I assume these are errors from your program controlling the brightness 
of the internal monitor.

4) Module dev-i2c, needed by ddcutil, has not been loaded.

Sanford

FedeDP, 2018-09-18:
I see; my program installs an /usr/lib/modules-load.d/ conf file to automatically load i2c-dev, but the user giving this output had installed clightd without ddcutil support (because otherwise he experiences freezes). I'll come back with new output as soon as he deliver :)

FedeDP, 2018-09-19:
Wonderful news: running it again, with i2c-dev module loaded and by root, froze the screen.
Here is the new output: http://dpaste.com/1M6MWFX.

rockowitz, 2018-09-19:
Federico,

First, an observation. The internal display (eDP1) is QHD+ (3200x1800).  
My understanding is that many of high resolution displays internally are 
multiple displays "stitched together", which may relate to why only part 
of the screen freezes.

In your DDCUTIL_LOOP macro, replace the "if (dlist)" block with 
"ddca_report_display_info_list(dlist, 1);"  and have your user re-run 
the program.  This will tell us whether the screen corruption occurs 
during display detection or later and whether the 
ddca_get_display_info_list2() call is in fact filtering out the internal 
display which does not support DDC.

Note that the "if (dlist)" test is unnecessary (though caution when 
using an API is a good thing).  ddca_get_display_info_list2() always 
succeeds.  The returned status code is always 0, and a 
DDCA_Display_Info_List struct is always allocated.

Are you building ddcutil from the tarball or are you using one of the 
pre-built packages?    There's lots of diagnostic output that can be 
controlled by setting switches within the code.

Finally, a comment that doesn't apply to the problem at hand.  I see 
that you're using the monitor serial number as a key.  This is a bit 
dangerous, as it's not always set, even for external monitors.  See 
http://www.ddcutil.com/monitor_notes for examples, including several LG 
and Samsung monitors.

Regards,
Sanford

FedeDP, 2018-09-19:
> In your DDCUTIL_LOOP macro, replace the "if (dlist)" block with "ddca_report_display_info_list(dlist, 1);"  and have your user re-run the program.  This will tell us whether the screen corruption [...]

I will and report back.
> Note that the "if (dlist)" test is unnecessary (though caution when using an API is a good thing).  ddca_get_display_info_list2() always succeeds.  The returned status code is always 0, and a DDCA_Display_Info_List struct is always allocated.

I see, thanks!
> Are you building ddcutil from the tarball or are you using one of the pre-built packages?    There's lots of diagnostic output that can be controlled by setting switches within the code.

I am building from aur, and the user affected by the issue too. So he can rebuild it if needed.

> Finally, a comment that doesn't apply to the problem at hand.  I see that you're using the monitor serial number as a key.  This is a bit dangerous, as it's not always set, even for external monitors.  See http://www.ddcutil.com/monitor_notes for examples, including several LG and Samsung monitors.

Nice catch, thanks! So, is there any universal/unique ID that can be used to identify a monitor here?

Btw really thank you for the time you're spending on the issue.

rockowitz, 2018-09-19:
> In your DDCUTIL_LOOP macro, replace the "if (dlist)" block with
>     "ddca_report_display_info_list(dlist, 1);" and have your user
>     re-run the program. This will tell us whether the screen [...]

Some possibilities, none great:

1) Use field path of type DDCA_IO_Path in struct DDCA_Display_Info. This 
identifies a monitor by its physical access path.
2) Hash the EDID, but beware that monitors can return different EDIDs 
depending on the connector - a VGA EDID will have different information 
from a DVI EDID.
3) Use field dref of type DDCA_Display_Ref (opaque pointer) in struct 
DDCA_Display_Info.  The API enumerates the displays once, when first 
needed, and maintains the data in a permanent data structure.

But note that solutions 1 and 3 raise the larger question which the API 
does not yet address, namely what happens if a monitor is connected to 
or disconnected from the computer after libddcutil is initialized.
>
> Btw really thank you for the time you're spending on the issue.
There's something going on here.   I suspect it's a corner case of a 
corner case.  Trying to diagnose these problems without having the 
hardware in front of me is a challenge, but between you, your user, and 
me I'm sure we'll figure it out.

FedeDP, 2018-09-26:
> Thanks for the patch.
After running the resulting clightd and then entering:
busctl call org.clightd.backlight /org/clightd/backlight org.clightd.backlight setallbrightness "d(bdu)s" 0.07 0 0 0 "" in another terminal, the resulting output was Found 0 displays

It returns `Found 0 displays`.

> Some possibilities, none great: 1) Use field path of type DDCA_IO_Path in struct DDCA_Display_Info. This identifies a monitor by its physical access path. 2) Hash the EDID, but beware that monitors [...]

I don't know whether to implement such a change: monitor serial number as a key is only used to set possibly different brightness levels for each screen (thus bus interface must have a key to access correct screen). But it is not used by clight and can (and probably will) be dropped. 
Really thanks anyway!
BTW I was wondering whether there was a method in ddcutil to monitor DDC changes, something like  udev_monitor API.

rockowitz, 2018-09-26:
> Thanks for the patch.
>     After running the resulting clightd and then entering:
>     busctl call org.clightd.backlight /org/clightd/backlight
>     org.clightd.backlight setallbrightness "d(bdu)s" [...]
Does the screen get corrupted in this case?  If it does, then the 
problem occurs in ddca_get_display_info_list2(), and we can drill down 
into that function.  If it doesn't, well that makes no sense as the "if 
(dlist)" block has been eliminated.

> Some possibilities, none great: 1) Use field path of type
>     DDCA_IO_Path in struct DDCA_Display_Info. This identifies a
>     monitor by its physical access path. 2) Hash the EDID, but beware [...]
Experimental command "ddcutil watch" reports changes made using the 
OSD.  It uses feature codes x02 (New control value) and x52 (Active 
control).   Unfortunately, the defined behavior of these feature codes 
varies with MCCS (VCP) version, and it's possible that the behavior 
implemented by the monitor does not reflect the declared version.  More 
importantly, feature codes x02/x52 report only changes made using the OSD.

FedeDP, 2018-09-26:
> Does the screen get corrupted in this case?  If it does, then the problem occurs in ddca_get_display_info_list2(), and we can drill down into that function.  If it doesn't, well that makes no sense as the "if (dlist)" block has been eliminated.

It doesn't. 
So the issue is somewhere here:
<pre>
        for (int ndx = 0; ndx < dlist->ct; ndx++) {
            DDCA_Display_Info *dinfo = &dlist->info[ndx];
            DDCA_Display_Ref dref = dinfo->dref;
            DDCA_Display_Handle dh = NULL;
            if (ddca_open_display(dref, &dh)) {
                continue;
            }
            DDCA_Any_Vcp_Value *valrec;
            if (!ddca_get_any_vcp_value_using_explicit_type(dh, br_code, DDCA_NON_TABLE_VCP_VALUE, &valrec)) {
                func;
                ddca_free_any_vcp_value(valrec);
            }
            ddca_close_display(dh);
        }
        ddca_free_display_info_list(dlist);
</pre>
Do you see any issue with this code?

FedeDP, 2018-09-26:
<pre>
void
ddca_free_display_info_list(DDCA_Display_Info_List * dlist) {
   if (dlist) {
      for (int ndx = 0; ndx < dlist->ct; ndx++) {
         ddca_free_display_info(&dlist->info[ndx]);
      }
      free(dlist);
   }
}
</pre>
I'd say that ddca_free_display_info_list looks quite clean...I don't understand what's going on.

Please note that another user is experiencing same issue, and he owns same laptop (Dell XPS13 9630).

rockowitz, 2018-09-26:
Per your earlier post, ddca_report_display_info_list() is reporting 
"Found 0 displays". So we know that dlist->ct = 0, and the body of your 
for loop should never execute.

The DDCUTIL_LOOP macro looks reasonable, but at least for debugging 
purposes I suggest that you replace it with with actual functions that 
perform the behavior you want on each of the detected displays.  (I note 
that there are only 2 different functions passed to the macro.)

Sanford

rockowitz, 2018-09-26:
> void
> ddca_free_display_info_list(DDCA_Display_Info_List * dlist) {
>     if (dlist) {
>        for (int ndx = 0; ndx < dlist->ct; ndx++) {
>           ddca_free_display_info(&dlist->info[ndx]);
>        }
>        free(dlist);
>     }
> }
>
> I'd say that ddca_free_display_info_list looks quite clean...I don't 
> understand what's going on.
>
> Please note that another user is experiencing same issue, and he owns 
> same laptop (Dell XPS13 9630).
Was that a typo?  Should it be Dell XPS 13 9360?

FedeDP, 2018-09-27:
> Was that a typo?  Should it be Dell XPS 13 9360?

Yes it was (i fixed it immediately after)!

> The DDCUTIL_LOOP macro looks reasonable, but at least for debugging purposes I suggest that you replace it with with actual functions that perform the behavior you want on each of the detected displays.  (I note that there are only 2 different functions passed to the macro.)

I will, and as soon as testers try it out, i will report back.

FedeDP, 2018-09-30:
i'm back with more news. 
In the end it seems like `ddca_get_display_info_list2` is the culprit.
My tester did forget to modprobe i2c-dev in latest tests. After modprobing it, it now freezes the screen and prints "Found 0 displays".

rockowitz, 2018-09-30:
Ok.  At least that makes sense.  The challenge is going to be how to 
find the point in the code where the corruption happens.   I have 
hunches, but they're just that. Does your user have the ability to set 
breakpoints?

Also, since the problem is not seen on the standalone ddcutil, but only 
when using the API, I suspect there's some interaction between clightd's 
access to the display and ddcutil's probing. What happens if you call 
ddca_get_display_info_list2() at the start of clightd initialization, 
before any of your own display access?

Lastly, the dpaste links on previous messages no longer work.  In the 
future, please include output as message attachments so they are permanent.

Sanford

FedeDP, 2018-09-30:
> Ok.  At least that makes sense.  The challenge is going to be how to
find the point in the code where the corruption happens.   I have
hunches, but they're just that. Does your user have the ability to set
breakpoints?

I don't think so, but i will ask. I will probably tell them (we currently have 2 users with same issue on same laptop) to start posting here if you don't mind.

> Also, since the problem is not seen on the standalone ddcutil, but only
when using the API, I suspect there's some interaction between clightd's
access to the display and ddcutil's probing.

Well it happens with `ddcutil interrogate` too. So i don't think clightd interaction with display is relevant; what do you think?

> Lastly, the dpaste links on previous messages no longer work.  In the
future, please include output as message attachments so they are permanent.

Ok.

rockowitz, 2018-09-30:
> Ok. At least that makes sense. The challenge is going to be how to
>     find the point in the code where the corruption happens. I have
>     hunches, but they're just that. Does your user have the [...]
Makes sense.

> Also, since the problem is not seen on the standalone ddcutil, but
>     only
>     when using the API, I suspect there's some interaction between
>     clightd's
>     access to the display and ddcutil's [...]

If the problem occurs on interrogate, I agree it's not an issue of 
clightd interaction.  It should also make it easier for me to create 
modified versions of the code that exclude certain actions. interrogate 
just packages up the detect, environment, and probe commands at maximal 
verbosity, along with a few other tests.  So before trying to modify the 
code, which (if any) of the following commands
trigger the problem:
   ddcutil detect
   ddcutil detect --verbose
   ddcutil environment
   ddcutil environment --verbose

jghauser, 2018-10-01:
I'm on of the two people having issues with this.

I just tried the commands and both `ddcutil detect` and `ddcutil environment` lead to freezes (I didn't the verbose flag as I thought it would probably cause the same issues and restarting my computer all the time is a hassle). If you want, I can still run those commands, too.

FedeDP, 2018-10-01:
@jghauser 
Just in case, are you using some powersave option for i915 kernel module? 
I see that archwiki XPS13 (9360) page has a couple of interesting points:
https://wiki.archlinux.org/index.php/Dell_XPS_13_(9360)#Module-based_Powersaving_Options
https://wiki.archlinux.org/index.php/Dell_XPS_13_(9360)#Freezing_after_waking_from_suspend

Dr-Ongo, 2018-10-02:
both `ddcutil detect` and `ddcutil environment` freeze the screen for me too whether I use the verbose flag or not. I've attached the output I get from both commands.

[ddcutil-detect.txt](https://github.com/rockowitz/ddcutil/files/2436115/ddcutil-detect.txt)
[ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/2436116/ddcutil-environment.txt)

Dr-Ongo, 2018-10-02:
Still freezes with `i915.disable_power_well=0` set

FedeDP, 2018-10-02:
@Dr-Ongo  Thanks for the test with i915.disable_power_well=0!

Dr-Ongo, 2018-10-03:
The last light before the freeze happens with `ddcutil detect --trace all` is:
```
(invoke_i2c_reader) reader=read_reader, bytect=1
```

rockowitz, 2018-10-03:
That's most helpful!  However,  it doesn't indicate the context in which 
the call occurs,  e.g. the I2C slave address being read from.  Can you 
report the several trace lines prior to this line?  If 
invoke_i2c_reader() is being called where I think it is being called, I 
would expect to see calls to i2c_check_bus(), 
i2c_detect_ddc_addrs_by_fd(), and i2c_set_addr().

Thank you!

Sanford

Dr-Ongo, 2018-10-03:
I've attached a photo of my screen after the glitch occurs. It seemed like the easiest way to capture several lines because screenshots don't work (the screenshot just looks normal)
[image]

rockowitz, 2018-10-03:
Thank you.  The screen corruption occurs where I suspected.  
i2c_check_bus() makes 3 trivial writes or reads to slave addresses x30 
(EDID block number), x50 (EDID), and x37 (DDC) to see if these addresses 
are responsive.   It's on the DDC probe that the corruption occurs.  
What doesn't make sense to me is that the point of corruption is on 
busno=2 (/dev/i2c-2). Your earlier interrogate output shows that 
/dev/i2c-3 is your laptop panel.

The DDC test at this layer for slave address x37 is vestigial.  A higher 
layer of display detection actually attempts a proper DDC exchange.   
I'll push out the current development version of ddcutil with the test 
disabled.  Can you pull from a branch on github, or do you need a tarball?

Sanford

rockowitz, 2018-10-03:
Branch next15 on github disables the test for I2C slave address x37.  It may resolve the screen corruption problem, or it may just push it to another point in the code.

Dr-Ongo, 2018-10-04:
the next15 branch froze my screen but it seems like it was at a different place. I've attached a photo. I thought it might be helpful to give you the full text of the trace so that is attached too, as well as some error messages that appeared in my terminal window when I redirected the output to a file.
[image]
[ddcutil-trace-all.txt](https://github.com/rockowitz/ddcutil/files/2444482/ddcutil-trace-all.txt)
[ddcutil-trace-all-terminal-output.txt](https://github.com/rockowitz/ddcutil/files/2444483/ddcutil-trace-all-terminal-output.txt)

rockowitz, 2018-10-04:
> 20181004_101148

I suspect that there's some missing terminal output, i.e. output that 
hasn't yet been flushed when the screen freezes.  I've modified the 
trace routines to call fflush() after every write, and uploaded the 
changes to branch next15.  Can you try running "ddcutil detect" with the 
new code and send a picture of the hung screen?

FYI the error messages in the terminal window when redirecting are from 
stderr.  It would be helpful to know when the stderr output occurs in 
relation to stdout, e.g. by redirecting using "&> filename".

Regards,
Sanford

Dr-Ongo, 2018-10-05:
[image]
[ddcutil-next15-trace-all.txt](https://github.com/rockowitz/ddcutil/files/2448716/ddcutil-next15-trace-all.txt)

rockowitz, 2018-10-05:
> 20181005_120846

Well, I'm baffled.  If the screenshot is to believed, the corruption 
occurs on the probe of /dev/i2c-2. But the full trace output shows that 
there's nothing there, the display is connected to /dev/i2c-3.

In the spirit of "Hail, Mary", I've made 2 changes in the current next15 
branch:
- disable the slave address x30 probe, which is where the screenshot 
seems to indicate the corruption occurs.
   This test is purely informational.
- do not probe a /dev/i2c device if it is associated with an eDP 
connected display.

The next step, I'm afraid, will be to step through ddcutil execution in 
a debugger. I'm still looking for a local Dell XPS 13 to test with.

Thanks again for all your help.

Sanford

Dr-Ongo, 2018-10-06:
Latest next15 did not freeze the screen. Output attached. Thank you for all this work you're doing to fix an issue that seems to only affect 1 model of laptop.
[ddcutil.txt](https://github.com/rockowitz/ddcutil/files/2452666/ddcutil.txt)

rockowitz, 2018-10-06:
Whew!

It appears that the test triggering the corruption is the probe of I2C 
slave address x30, EDID block number.  This can safely be left disabled, 
as it is currently purely informational. ddcutil uses only EDID block 0.

There's still a bunch of cleanup ahead, including re-enabling some 
probes that might themselves be problematic, but we're on the downhill 
slope.

A couple questions:

What happens if you execute "ddcutil environment --verbose"?  Does that 
now work as well?

The interrogate output indicated that you don't have i2cdetect 
installed.  Please install it and execute command "i2cdetect 3" to probe 
/dev/i2c-3.  Does it corrupt the screen?  If not, what is the output.

Thanks again for all your help.

Regards,
Sanford

rockowitz, 2018-10-08:
> No freezes with these commands
> full output of |ddcutil environment --verbose| attached
> ddcutil-environment-verbose.txt
>
> output of |i2cdetect 3| was
>
> |WARNING! This program can confuse your I2C bus, cause data loss and
> worse! I will probe file /dev/i2c-3. I will probe address range
> 0x03-0x77. Continue? [Y/n] y 0 1 2 3 4 5 6 7 8 9 a b c d e f 00: -- --
> -- -- -- -- -- -- -- -- -- -- -- 10: -- -- -- -- -- -- -- -- -- -- --
> -- -- -- -- -- 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 30:
> 30 31 32 33 34 35 36 37 -- -- -- -- -- -- -- -- 40: -- -- -- -- -- --
> -- -- -- -- -- -- -- -- -- -- 50: 50 51 52 53 54 55 56 57 58 59 5a 5b
> 5c 5d 5e 5f 60: 60 61 62 63 64 65 66 67 -- -- -- -- -- -- -- -- 70: --
> -- -- -- -- -- -- -- |

The current next15 branch on github contains the following changes of 
relevance to this thread:
- Re-enables the probe of slave address x37 which had been disabled 
trying to find the cause
   of screen corruption.  We'll see if this is also a problem.
- Made the test for eDP devices more noisy for debugging purposes.  No 
use is currently made of the result.
- Add Synopsys Designware devices to the list of I2C devices that are 
not probed.  Needlessly probing the
   Synopsys devices may have been the cause the Designware errors seen 
in the logs.

Let me know how it goes.

Regards,
Sanford

Dr-Ongo, 2018-10-08:
running `ddcutil detect --trace all` with the current next15 doesn't freeze the screen, but does lead to some temporary display corruption. By the time ddcutil detect is finished running, everything looks normal again. Photo and trace output attached.
[image]
[ddcutil-detect-trace.txt](https://github.com/rockowitz/ddcutil/files/2456062/ddcutil-detect-trace.txt)

Dr-Ongo, 2018-10-08:
No display corruption at all this time
[ddcutil-detect.txt](https://github.com/rockowitz/ddcutil/files/2456397/ddcutil-detect.txt)
[ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/2456398/ddcutil-environment.txt)

Dr-Ongo, 2018-10-08:
[ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/2456699/ddcutil-environment.txt)

rockowitz, 2018-10-08:
My oops here.  Latest next15 changes had not been pushed to github.  
Please download again and rerun "ddcutil environment --verbose".  Thanks.

Sanford

Dr-Ongo, 2018-10-08:
Haha no worries.
[ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/2456996/ddcutil-environment.txt)

rockowitz, 2018-10-11:
I've pushed out a new set of updates to branch next15.  Most 
significantly the display detection code has been reworked and 
simplified reflecting the changes made for the Dell XPS 13.  Please run 
the following commands and send me the output.  Thanks.

   ddcutil detect --verbose
   ddcutil environment --verbose

Dr-Ongo, 2018-10-12:
No corruption or anything unusual to report.
[ddcutil-detect.txt](https://github.com/rockowitz/ddcutil/files/2471106/ddcutil-detect.txt)
[ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/2471107/ddcutil-environment.txt)

rockowitz, 2018-10-12:
Branch next15 contains the following changes:
- additional timestamp messages to better correlate any relation between 
actions
   of the environment command and messages seen in log scans.
- suppress a debug message related to eDP device detection on the detect 
command

Please run the usual commands at your convenience:
   ddcutil detect --verbose
   ddcutil environment --verbose

Thanks you,
Sanford

Dr-Ongo, 2018-10-14:
Sorry about the delay in running the latest test.
[ddcutil-detect.txt](https://github.com/rockowitz/ddcutil/files/2476869/ddcutil-detect.txt)
[ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/2476870/ddcutil-environment.txt)

rockowitz, 2018-10-14:
No problem about the delay. We're just in cleanup mode now.  Can you 
rerun with the latest push of branch next15?  Thanks.

Sanford

Dr-Ongo, 2018-10-15:
[ddcutil-detect.txt](https://github.com/rockowitz/ddcutil/files/2476989/ddcutil-detect.txt)
[ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/2476990/ddcutil-environment.txt)

rockowitz, 2018-10-15:
I think we're wrapped on this.  The detect output is clean, and there 
don't seem to be any Synopsys DesignWare errors in the logs that might 
be associated with ddcutil activity, though I do note there are a lot of 
callback suppressed messages from drm_dp_i2c_do_msg about the time the 
environment command queries udev (hard to say for certain because the 
timestamps aren't granular enough).

Thanks again for all your help.

Sanford

FedeDP, 2018-10-15:
After having thanked both @Dr-Ongo and @jghauser, i'd like to thank @rockowitz too that had the patient and the willing to collaborate with us to fix this bug.

Thank you very much for all the time you spent looking into the issue searching and fixing the bug.

Dr-Ongo, 2018-10-15:
Thank you both for all your work solving the problem!

rockowitz, 2018-10-16:
Again, thanks to all of you, particularly @Dr-Ongo, who quickly tested 
build after build.

How urgent is it that these changes get into a ddcutil point release?  
Can you use the next15 development branch of ddcutil for now?   Putting 
out a formal release, with documentation, platform-specific testing, and 
distribution-specific builds takes a fair amount of effort.   Given 
everything else that's going on with ddcutil (user supplied feature 
definitions for manufacturer-specific features, ddcui GUI, and related 
API changes), the next point release is naturally a couple months away.

Sanford

rockowitz, 2018-11-04:
Changes to display detection will be part of release 0.9.3.

closed 2018-11-04 by rockowitz