rockowitz/ddcutil #61: Second monitor in MST chain not supported

AndreasBackx, 2018-09-03:
I've got 2 of my monitors setup using MST of which only the first in the chain is detected and works with ddcutil. Both monitors are Dell U2515H monitors that work when not using MST. If you need any more information, feel free to ask. I will try not to use MST, but I'll have to find myself a longer cable. The Acer monitor mentioned in the information is working correctly btw and is not having any problems atm.

[Here's a gist of all the information I could get, (I am part of the group for i2c devices, but I thought I'd run the environment command as root so it might contain more information).](https://gist.github.com/AndreasBackx/d7ab65f74c494aebf0edb4f32a7c24f3)

rockowitz, 2018-09-03:
Andreas,

Interesting report.  I know there are problems with MST support in the 
i915 driver.  This is the first I've seen for AMD.  Some observations:

1) Both the radeon and amdgpu drivers are loaded.  Is there a reason for 
this?

2) Looking at the filtered log files, it appears there's program failure 
at line 88 of amdgpu driver file amdgpu_dm_mst_types.c, when doing an 
i2c transfer.  Can you send the full dmesg output for a failure, along 
with file Xorg.0.log?

Regards,
Sanford

AndreasBackx, 2018-09-04:
> 1) Both the radeon and amdgpu drivers are loaded.  Is there a reason for
this?
Ah my bad, the wiki page on AMD is a bit confusing. I'll remove the radeon one.

> 2) Looking at the filtered log files, it appears there's program failure
at line 88 of amdgpu driver file amdgpu_dm_mst_types.c, when doing an
i2c transfer.  Can you send the full dmesg output for a failure, along
with file Xorg.0.log?
I'll report back to you when I get home tonight.

rockowitz, 2018-09-04:
Also, please re-run "ddcutil env -v" without the radeon driver, on the 
off-chance that's triggering the problem.  Thanks.

AndreasBackx, 2018-09-04:
Alright, I've removed radeon from the modules, and updated/added the files you asked for in the gist. It doesn't seem to have made a difference on the surface as I assume by placing amdgpu first in the modules list, it used that (which is what the wiki says).

The `Xorg.0.log` file was first removed after having already booted with only the amdgpu module, then I rebooted, and finally logged into i3. I hope that gives you some concise logs.

AndreasBackx, 2018-11-12:
Any update on this?

rockowitz, 2018-11-12:
As noted previously, there's a failure in the amdgpu driver near 
function dm_dp_aux_transfer() at line 88 of file 
amdgpu_dm_mst_types.c.   This is a failure in the amdgpu driver.  Not 
being able to replicate your hardware configuration I can't say more, 
beyond the general comment that I2C MST support in Linux video drivers 
is still problematic.  I suggest you report this to the amdgpu driver 
folks.

Sanford

AndreasBackx, 2019-04-14:
I've been changing my setup a bit and I moved to Wayland. I realised that I completely forgot about this issue. I was ready to create a bug report on the issue tracker for amdgpu, though when I recreated the logs, I no longer saw the warning you were referring to. [Here are the new logs.](https://gist.github.com/AndreasBackx/c8e295da821e4df9f3daa7a04768f84d) I'm now running [Sway](https://github.com/swaywm/sway). Has anything else changed? Do you need some more information perhaps?

rockowitz, 2019-04-23:
Andreas,

I apologize for taking so long to respond.

Even though the amdgpu warnings are no longer there, the issue is the 
lack of MST I2C support in the amdgpu driver, and should be directed to 
the amdgpu folks. There is no /dev/i2c device associated with the second 
Dell monitor associated with DP-5.

I suggest you install program i2cdetect (normally found in package 
i2c-tools).  "ddcutil environment" uses this program as an additional 
check if it's available.  What I expect it will confirm is that there's 
no /dev/i2c device for the second Dell monitor.

Sanford

closed 2025-02-09 by rockowitz