rockowitz/ddcutil #63: setvcp --display wildcard, range or "all" option

louisrousseau, 2018-10-21:
Use case here is to set brightness on a trio of identical monitors with a single command.
(Current workaround is to script three consecutive calls with `--display` 1 to 3.)

rockowitz, 2018-10-23:
Louis, 

I think you're suggesting that ddcutil add the ability to set identical values for a feature on multiple monitors, or perhaps to set multiple feature values on multiple monitors in one invocation. It seems to me that a script is the appropriate place to orchestrate such behaviour.  Why is it important to add the ability to ddcutil itself?  The script, even with bells and whistles, can't be longer than a dozen or so lines.  

Sanford

altV, 2018-10-24:
One thought that comes to mind is - two ddcutil commands doesn't seem to work simultaneosly. I can guess this might be intrinsic limitation of videocard or something else.
Other than that I just scripted the two lines of code for my two monitors.

louisrousseau, 2018-10-25:
I ended up scripting as well; as reported by @altV parallel calls using xargs -P have their commands run in sequence.

rockowitz, 2018-10-25:
The issue is whether the feature warrants the complexity, especially 
when the behavior described here can easily be scripted.  How frequently 
would one want to set the same exact value on multiple monitors?  Even 
in the case of brightness (VCP feature x10) two different monitors would 
likely require different values for the same brightness level.  Nor 
would adding, say, 5 to the value for feature x10 necessarily increase 
the brightness for the same amount in different monitors.  So now we're 
into modifying the command line syntax to apply different feature value 
changes to different monitors. The number of command line options is 
already frighteningly large.  It's a rabbit hole not worth going down 
when, as I've said, the facility can easily be scripted by someone to 
handle their specific situation.

Sanford

rockowitz, 2018-10-25:
> I actually just wanted to ask why it doesn't work running two commands
> at once in bash in parallel, I do not see much value of having it in
> linux world as well

Ahh.  That's a different question from the thread subject (--display 
wildcard, range, or "all" option). The library version has the ability 
to access multiple monitors in parallel in a single instance of the 
library, and doesn't allow multiple threads to access the same monitor 
simultaneously. However, there's no ability to coordinate access across 
library instances or ddcutil command line instances. That would require 
turning ddcutil into a service.

I would imagine that two parallel instances could well conflict during 
the detection phase of command execution, where each /dev/i2c device is 
opened.  It's all a matter of timing.  Using the --nodetect and --bus 
option instead of --display avoids opening every /dev/i2c device for 
examination and might keep the ddcutil instances from clobbering each 
other.  Send me your script (as an attachment) and I'll see how it behaves.

Sanford

altV, 2018-10-25:
yay it works!! thank you :)
```
❯ cat bin/brightness
#!/bin/bash
# sudo ddcutil setvcp 10 "$@" -d 1
# sudo ddcutil setvcp 10 "$@" -d 2

sudo ddcutil setvcp 10 "$@" --nodetect --bus 0 &
sudo ddcutil setvcp 10 "$@" --nodetect --bus 8 &

```

closed 2018-11-06 by louisrousseau

Ryanf55, 2020-04-14:
I have at most four monitors connected but frequently re-arrange my system. I would like use `setvcp` on all monitors with manufacturer code `--mfg` "ACR", however the behavior, as specified in documentation is that "The first monitor to satisfy all the specified criteria is selected". Can anyone assist in writing a script that sets the brightness for all monitors matching a criteria like manufacturer code?

rockowitz, 2020-04-15:
Take a look at the output of "ddcutil detect --terse".   It is easily 
parsable and will give you the information you require to select 
display(s) to operate on.   Keep in mind that there's no expectation 
that the same brightness value for multiple monitors, even from the same 
manufacturer, will have identical luminescence for the same setvcp 
value.  Even monitors of the same model can age differently.

-- Sanford

Ryanf55, 2020-04-17:
Thanks. For anyone wanting a full script to accomplish the goal of this ticket with existing ddcutil functions, see below. 

_monitor_brightness.sh_
```
#!/bin/bash

Help()
{
   # Display Help
   echo "This script sets the brightness of all connected monitors"
   echo
   echo "Syntax: monitor_brightness [-h] brightness"
   echo "options:"
   echo "h     Print this Help."
   echo
}

while getopts ":h" option; do
   case $option in
      h) # display Help
         Help
         exit;;
   esac
done

brightness=$1
brightness_c="0x10"

if [ -z "$brightness" ]
then 
    echo "Error. No brightness specified"
    exit 1
fi

# Detect all monitor serial numbers
# arr=( $(`ddcutil detect --terse | grep Monitor: | sed -e 's/.*:.*:.*:/\n/'`))
monitor_serials=`ddcutil detect --terse | grep Monitor: | sed -e 's/.*:.*:.*://'`

for serial_num in $monitor_serials;do 
    ddcutil --sn $serial_num setvcp ${brightness_c} ${brightness}
done
```

To parse "ddcutil detect --terse" for only the serial numbers:
`ddcutil detect --terse | grep Monitor: | sed -e 's/.*:.*:.*://'`
Note: You can put grep before the sed to filter by brand or model number if needed.

If I get around to adding functionality to have a calibration curve for each monitor, I will add the script here, however the above satisfies the OP's original request.

rockowitz, 2020-04-17:
This will work most of the time for most users.  However:

- Some monitors may not support DDC.  Those will be reported as "Invalid 
display" rather than "Display N".

- Some monitors, e.g. those by LG, do not have serial numbers.

cymen, 2021-07-23:
If you end up here looking for how to alter the brightness of all your monitors at once and you happen to have them all work with the typical command, you could try using xargs with the parallel option. In my case, I have 3 monitors attached so to count from 1 to 3 I'm using `seq 3` and then passing it to xargs like so:

```sh
seq 3 | xargs -n 1 -P 3 sh -c 'ddcctl -d $1 -b 10' sh
```

I added these handy aliases to my shell rc file:

```sh
alias day="seq 3 | xargs -n 1 -P 3 sh -c 'ddcctl -d \$1 -b 80' sh"
alias night="seq 3 | xargs -n 1 -P 3 sh -c 'ddcctl -d \$1 -b 10' sh"
```

Maybe a better way to do it but this works great in that all three monitors change brightness at the same time.

rockowitz, 2021-07-24:
@cymen Your post describes, as far as I can tell, use of MacOS program ddcctl (https://github.com/kfix/ddcctl).  This Github repository is for a different program, ddcutil.  Simply changing "ddcctl" to "ddcutil" in your examples will not work.  The commands have different syntax.

cymen, 2021-07-24:
@rockowitz Whoops, I was indeed referring to the MacOS program. Thanks for the correction.

Nikola-Milovic, 2023-01-21:
I think this is the `ddcutil` equivalent 

```
#! /bin/bash
if ! [ -x "$(command -v ddcutil)" ]; then
	echo 'Error: ddcutil is not installed.' >&2
	exit 1
fi

export b_val=$1
seq 2 | xargs -n 1 -P 2 sh -c 'sudo -u root ddcutil -d $1 setvcp 10 $b_val >/dev/null' sh
```

Some notes, I added 2 entries to the sudoers, (not sure they are necessary)

```
USER ALL=(ALL) NOPASSWD: /path/to/script
USER ALL=(ALL) NOPASSWD: /bin/ddcutil
```

NOTE: This doesn't work as is, if ran as it is, it will only set 1 display and it will say that it cannot find the other one
`Display not found`, but if you don't use `seq` and `xargs` and just place it on 2 lines

```
sudo -u root ddcutil -d 1 setvcp 10 $b_val >/dev/null
sudo -u root ddcutil -d 2 setvcp 10 $b_val >/dev/null
```

It works fine.

I am open to improving this, any suggestions are welcome

acerspyro, 2023-09-13:
> Thanks. For anyone wanting a full script to accomplish the goal of this ticket with existing ddcutil functions, see below.
> 
> **_monitor_brightness.sh_**
> 
> ```
> #!/bin/bash
> 
> Help()
> {
>    # Display Help
>    echo "This script sets the brightness of all connected monitors"
>    echo
>    echo "Syntax: monitor_brightness [-h] brightness"
>    echo "options:"
>    echo "h     Print this Help."
>    echo
> }
> 
> 
> while getopts ":h" option; do
>    case $option in
>       h) # display Help
>          Help
>          exit;;
>    esac
> done
> 
> brightness=$1
> brightness_c="0x10"
> 
> if [ -z "$brightness" ]
> then 
>     echo "Error. No brightness specified"
>     exit 1
> fi
> 
> # Detect all monitor serial numbers
> monitor_serials=`ddcutil detect --terse | grep Monitor: | sed -e 's/.*:.*:.*://'`
> 
> for serial_num in $monitor_serials;do 
>     ddcutil --sn $serial_num setvcp ${brightness_c} ${brightness}
> done
> ```
> 
> To parse "ddcutil detect --terse" for only the serial numbers: `ddcutil detect --terse | grep Monitor: | sed -e 's/.*:.*:.*://'` Note: You can put grep before the sed to filter by brand or model number if needed. For me I have a mixed brand and model number setup but like to change brightness frequently depending on the application I am using, which set of inputs, and other things.
> 
> If I get around to adding functionality to have a calibration curve for each monitor, I will add the script here, however the above satisfies the OP's original request.

This only seems to affect one monitor for some reason. Both my monitors are exactly the same model of Dell S2721DGF.

It would be much easier if it'd be built into the tool rather than having to fiddle with scripts. Besides, as noted, I don't think you can affect two monitors at the same time as each instance of the tool locks the i2c channel for itself, so the brightness has to step for every monitor, with a delay in-between. The only solution I can see for this is to allow the tool to affect multiple monitors at once.