rockowitz/ddcutil #68: eizo EV2785

yannick, 2018-11-30:
seems to not work.
here is a interrogate
https://gist.github.com/yannick/cfb5440a8ecc5bfb6cec8c7fae2f5036

also i did some usb sniffing in windows, this one first shows the internal eizo identification menu and then does some gamma/brightness calibration stuff
https://gist.github.com/yannick/c61294e474598d849a46aa996c41a9d1

yannick, 2019-01-03:
hi @rockowitz  
no worries, and thanks for prividing a great tool!
here is the output:
https://gist.github.com/d3bc454d90de8a5e39f55869b9a4d961

rockowitz, 2019-01-03:
The output confirms that the Eizo is using a private "usage page" for 
its USB reports, instead of those defined in the spec.  It might be 
possible to reverse engineer the Eizo reports using usblyzer on Windows, 
but that would be a lot of work and might possibly run afoul of 
licensing issues.

Regards,
Sanford

closed 2019-05-30 by rockowitz