rockowitz/ddcutil #69: capabilities command fails with "get_vcp_version_by_display_ref: Assertion `psc == 0' failed"

sdomin, 2018-12-07:
The capbilities command fails:
> $ sudo ddcutil capabilities
> MCCS version: not present
> (lock_distinct_display) Attempting to lock display already locked by current thread
> ddcutil: ddc_vcp_version.c:151: get_vcp_version_by_display_ref: Assertion `psc == 0' failed.
> Aborted

But actually getting and setting parameters works fine:
> $ sudo ddcutil setvcp 10 20
>
> $ sudo ddcutil getvcp 10
> VCP code 0x10 (Brightness                    ): current value =    20, max value =   100

I'm running arch linux, ddcutil version is:
> $ pacman -Qi ddcutil
> Name            : ddcutil
> Version         : 0.9.3-1

Monitor is DELL U2713H, graphics is intel embedded i5-7260U
In case it's useful, the capabilities command used to work fine some time around April 2018 on the exact same system.

Thanks!

rockowitz, 2018-12-09:
Sergiy,

For personal reasons, I will not be able to address this problem for a 
couple weeks.  However, it certainly seems like a regression.

So that I can explore the problem when I return to work, please send the 
output of the following commands as attachments:

sudo ddcutil capabilities --trace ddc --trace ddcio

sudo interrogate

Thanks,
Sanford

rockowitz, 2018-12-10:
> Hi Sanford,
>
> Thank you for checking this.
> Oddly enough it works fine now (and it wasn't a major issue for me
> because I only rely on setvcp).
> I suspect it's my monitor playing up. I did reboot before reporting
> the bug, but I didn't turn the monitor off/on, which is the only thing
> I can think of that happened between the time I reported the issue and
> now.
>
> Sorry for the trouble and thank you for your work!
> Enjoy the coming holidays :)

The symptoms you describe are unlikely to the the result of a poor 
monitor DDC implementation.  If you can replicate the problem please 
send the output I requested and I'll take a closer look in a couple weeks.

Sanford

sdomin, 2018-12-10:
I didn't mean poor DDC implementation. There have been couple times when the monitor stopped reacting to menu buttons - including the power button. The only solution I found at the time was pulling the power cord.
Perhaps it was something like that - but, sure, it's just a random guess.

I will report the trace details if I reproduce it again.

rockowitz, 2018-12-20:
Sergiy,

I believe I see a path through the code that could trigger the abort you 
reported.  It would occur if the capabilities string returned by the 
monitor does not include a "vcp()" string, perhaps because it was 
ill-formed.

Do I understand correctly that the "ddcutil capabilities" command now 
works ok all/some of the time.  If some of the time, it may be that some 
flakiness in the monitor sometimes returns an ill-formed capabilities 
string, which then triggers the bug.

Regards,
Sanford

rockowitz, 2018-12-21:
I have updated development branch 0.9.4 to handle the double open 
problem that you triggered.  Thanks for reporting it.

Sanford

sdomin, 2018-12-21:
Thank you for checking this and fixing!

(No, I wasn't able to reproduce it since I reported the issue. I did try to run the command couple times over last 2 weeks. Also tried unloading and then reloading i2c_dev module, unplugging and replugging monitor and running the command while selecting unconnected input source on the monitor - all worked as expected. Which is a bit odd given that at the time of reporting I was able to reproduce it many times in a row - both before and after a PC restart).