rockowitz/ddcutil #70: Add value name "USB-C" for VCP code 0x60 (input source) value 0x1b (currently shown as "(null")

aryoda, 2018-12-29:
My Dell U3818DW supports DisplayPort over USB-C as input source value 0x1b but the input source is not yet known to ddcutil:

```
   Feature:  0x60 (Input Source)
      Values:
         0x0f: DisplayPort-1
         0x11: HDMI-1
         0x12: HDMI-2
         0x1b: (null)
```

I am not sure if this is DELL specific...

Can users of other display please confirm (or reject) my observation that `0x1b` is the standard value for (DisplayPort over) `USB-C`?

Does anybody know an official name for this value?

Next step would be to add a name for this value to `ddcutil`, "USB-C" or "DP over USB-C" or "DP over USB"...

rockowitz, 2018-12-29:
It will indeed be interesting to see if this is a common usage.  It may 
well be Dell specific.  I will be surprised if there is 
cross-manufacturer agreement on the use of this code.

This may be a good use for user supplied feature definitions. Release 
0.9.3 reworked the ddcutil internals to allow for them. It also 
incorporated required API changes, so that hopefully when the facility 
is fleshed out any additional changes will be upward compatible.   I had 
thought of the facility as a way to provide complete definitions for 
features in the manufacturer specific range.  At this point it's not 
possible to add a single value definition, the entire description of 
feature x60 would have to be replaced.   If you want to play with it I 
can send you details.

aryoda, 2018-12-29:
Using the user-supplied feature definitions of v0.9.3 or higher is a very good use case for that.

I am currently doing my "homework" (example implementation of a C++ ddcutil API...) but two weeks of school holidays are melting like snow with C++ ;-)

I will examine v0.9.3 after that (currently I am using a "pinned" git submodule with v0.9.2).

prototux, 2019-09-07:
On my U2419HC (so, one generation newer) it's also using 0x1B for the USB-C input source, so, i don't know if it's consistent across OEMs, but at least, it seems to be consistent between dell monitor generations.

rockowitz, 2019-09-08:
Thanks for taking the time to let me know.

Sanford

SumDum, 2020-04-18:
Hi Sanford,
On a ViewSonic VP3881 USB-C is 0x17
```
   Feature: 60 (Input Source)
      Values:
         0f: DisplayPort-1
         11: HDMI-1`
         12: HDMI-2
         17: Unrecognized value
```
So unfortunately no, not common usage.

rockowitz, 2020-04-20:
Thanks for posting this information.

madmalkav, 2020-07-25:
On a Lenovo Thinkvision P44W-10 , 0x14 is for the port labeled USB Type-C 10G (DP 1.4 Alternate Mode compatible) and 0x13 for the port labeled USB Type-C 5G (DP 1.2 Alternate Mode compatible).

attah, 2022-03-13:
My Dell P3222QE also uses 0x1b for USB-C. But my Asus MB16AC calls its single UCB-C input 0x11, aka HDMI-1 :(
Thanks for this great tool!

rockowitz, 2022-03-14:
For those unaware of it, the [User Defined Features](https://www.ddcutil.com/udf/) facility can be used to address variations like these.

aryoda, 2022-03-14:
I'd suggest to close this issue (perhaps incl. an attached example user-defined features file to "solve" the issue)...

bbkr, 2024-06-12:
On a ViewSonic VP2785-4K USB-C is 0x17.

rockowitz, 2024-12-13:
That's what the wiki is for. Add information to [page](https://github.com/rockowitz/ddcutil/wiki/Dell), or create a page for input switching on Dell monitors, similar to that for LG monitors.

closed 2025-02-09 by rockowitz