rockowitz/ddcutil #71: Nvidia w/ Maximum retries exceeded after proposed wiki fix

KillerKelvUK, 2018-12-30:
Hi, long time user of ddcutil and recently completely a full hardware upgrade and now encountering new issues.

Clean Ubuntu 18.04LTS install using rockowitz ppa for binaries.  AMD TR4 system with a GTX 1060 and using an Asus ROG PG279 Monitor with the Nvidia 390.87 drivers.

Out of the box ddcutil didn't work with my environment and I needed to apply then wiki advised fix of `options nvidia NVreg_RegistryDwords=RMUseSwI2c=0x01;RMI2cSpeed=100` via modprobe.d which allowed my monitor to be detected...

> $ ddcutil detect
> Display 1
>    I2C bus:             /dev/i2c-8
>    EDID synopsis:
>       Mfg id:           ACI
>       Model:            ROG PG279Q
>       Serial number:    blah blah
>       Manufacture year: 2018
>       EDID version:     1.3
>    VCP version:         2.2

However when I try a getvcp 60 I hit a max reties error even when I push the retry limit to the max 15,15,15.  Can you recommend any further debugging?

rockowitz, 2018-12-30:
Kelvin,

My initial reaction is that you're just out of luck.  The monitor 
probably can't keep up with the nominal 100mhz speed of the I2C bus, and 
the Nvidia drivers seem to be particularly unforgiving.

However, I do have some questions and suggestions:

1) Does the problem only occur with feature x60?   What about feature 
x10 (brightness)?

2) Have you tried using a different monitor with the system? Using the 
Asus monitor on a different system?

3) Do you see the same problem with the nouveau driver?

4) The following command options may shed light on what is happening:

   --ddc    report DDC protocol and data errors
   --stats  report error statistics
   --excp  report freed exceptions, another what to see the underlying 
errors causing the retry failures
   --y 2     use experimental sleep strategy feature, double the amount 
of time ddcutil sleeps between DDC operations

Please run the interrogate command and send the output as an 
attachment.  Perhaps something will pop out at me.

Sanford

KillerKelvUK, 2019-01-01:
Hey, thanks for the suggestions...

1)  Feature 10 works but plenty others don't.  I can also 'setvcp 10'.
2)  Can't test this monitor on a different rig but will dig out an old monitor to test on this rig however reading around it looks likely this Asus monitor has limited DDC/CI support, it certainly isn't a published feature from the manufacturer so I appreciate this may not go any further.
3)  Again will get to this test shortly, a bit of config to back out to get this working.
4)  Tried all those additional parameters, have attached the outputs here [ddcutil_tests.txt](https://github.com/rockowitz/ddcutil/files/2719561/ddcutil_tests.txt)

And heres the output from [ddcutil_interrogate.txt](https://github.com/rockowitz/ddcutil/files/2719552/ddcutil_interrogate.txt)

Appreciate you looking into this Sanford, thanks.

closed 2019-03-05 by rockowitz